MarlinFirmware/Marlin #8971: Fystec TMC2130 SPI config

kevinmce, 2017-12-30:
### Bug Report
- Description: I have been unable to get the TMC2130 to move while in SPI mode. 
As i have read in another issues  #8480 & [https://github.com/teemuatlut/TMC2130Stepper/issues/13](https://github.com/teemuatlut/TMC2130Stepper/issues/13)), the  #Fysetc drivers come with the standalone bridge connected. It took me some time to find that issue so if anything this issue may help others who run into this first. As a test I removed all SPI cables and tested the TMC2130 in standalone mode, Everything works as expected. 
When i removed the bridge to enable SPI the driver fails to energies the motor at all. I did manage to load Marlin 1.1.8 and run command M112, the out put is below. Any help in reading this output would be great, i dont know what it should look like. 

For now i have put the drivers back in standalone mode. But it would be great if we can figure out how to make these Fysetc drivers work on SPI. 

> 13:55:04.717 : N17 M122*20
> 13:55:04.717 : Y
> 13:55:04.717 : Enabled		false
> 13:55:04.717 : Set current	800
> 13:55:04.717 : RMS current	1436
> 13:55:04.717 : MAX current	2025
> 13:55:04.717 : Run current	25/31
> 13:55:04.717 : Hold current	12/31
> 13:55:04.717 : CS actual		0/31
> 13:55:04.717 : PWM scale	0
> 13:55:04.717 : vsense		0=.325
> 13:55:04.717 : stealthChop	false
> 13:55:04.717 : msteps		256
> 13:55:04.717 : tstep		0
> 13:55:04.717 : pwm
> 13:55:04.717 : threshold		0
> 13:55:04.717 : [mm/s]		-
> 13:55:04.717 : OT prewarn	false
> 13:55:04.717 : OT prewarn has
> 13:55:04.717 : been triggered	false
> 13:55:04.717 : off time		0
> 13:55:04.717 : blank time	16
> 13:55:04.717 : hysterisis
> 13:55:04.717 : -end		-3
> 13:55:04.733 : -start		1
> 13:55:04.733 : Stallguard thrs	0
> 13:55:04.733 : DRVSTATUS	Y
> 13:55:04.733 : stallguard
> 13:55:04.733 : sg_result		0
> 13:55:04.733 : fsactive
> 13:55:04.733 : stst
> 13:55:04.733 : olb
> 13:55:04.733 : ola
> 13:55:04.733 : s2gb
> 13:55:04.733 : s2ga
> 13:55:04.733 : otpw
> 13:55:04.733 : ot
> 13:55:04.733 : Driver registers:
> 13:55:04.733 : Y = 0x00:00:00:00

franji, 2017-12-30:
I'm using Fystec TMC2130 drivers and they work perfectly in SPI mode.
To mount them use these tutorials:
https://toms3d.org/2017/12/09/tmc2130-guide/
and 
http://www.3dcampy.com/2017/12/20/tutorial-instalacion-y-configuracion-driver-trinamic-tmc2130-en-ramps-y-mks-gen-1-4/
the second is in Spanish but it is very descriptive.

kevinmce, 2017-12-31:
Thanks for the links, the 2nd tutorial in Spanish is the best I've seen yet, but it do not see anything beyond what i already knew...

What exactly did you do to make the FYSETC ready for SPI? Only remove the jumper?

franji, 2017-12-31:
In my case I have also removed the pin I indicated in the photo as indicated by the tuto
[image]

kevinmce, 2017-12-31:
I figured it out. 

Seems the MKS Gen board has a rotated AUX 3 board compared to MKS Gen V1.4. 
Simple mistake, but after inspecting the campy tutorial closer i realized the pinout was different between these boards. 
After connecting to the correct SPI pins on the MKS Gen V1.0 everything works properly. 
[image]

kevinmce, 2017-12-31:
Thanks franji for the support, without someone to discuss this with i may have given up earlier.

frahlg, 2018-01-03:
I have the problem with my Fystec only moving in one direction. I am using a Rumba board, and have de-soldered the SPI-bridge. I have gotten the driver to work as expected in Stealtchop using the example sketch from the TMC2130 library, it then moves back and forth really smooth. But when using Marlin it only goes in the minus-direction, regardless of what I am doing. I have tried two different drivers, as well as two different axis. Tried Marlin bugfix-branch as well. Everything works in stand-alone mode.

lemmingDev, 2018-01-03:
Only moving in one direction sounds like an endstop config issue...

oakdesign, 2018-01-03:
@lemmingDev  np ot has been checked the issue is the CFG4 and 5 configuration

frahlg, 2018-01-03:
I managed to solder the miniature soldering points (which I suppose is CFG4 and 5?) shown in this thread: https://github.com/MarlinFirmware/Marlin/issues/8480

When I soldered these according to the picture it works as intended, even with Stallguard endstop. Glad I bought a couple of extra fystec-drivers as I have about 50% success rate in soldering these super small points. Will never recommend anyone buying the Fystec V1.0.

closed 2018-02-11 by thinkyhead

MaxwellXia, 2018-11-08:
good job!

robustini, 2018-12-08:
> 
> 
> In my case I have also removed the pin I indicated in the photo as indicated by the tuto
> [image]

This pin is not connected on the driver board (NC), so there's no difference.