MarlinFirmware/Marlin #8973: Nozzle doesn't exit bed margins when probing [FR]

Skybound420, 2017-12-30:
I'm using [Jim Brown's Easy Config fork of Marlin](https://github.com/JimBrown/MarlinTarantula) and have been successfully for months with my Tevo Tarantula.  However, I build a new mount for my BLTouch and relocated the probe to a new position, though my issue also occurred previously, just not as much of a problem.

My nozzle is centered and the probe offset is also stored to EEROM and I have bed margin set to 10 and currently use LINEAR bed leveling and grid points = 3.  When the G29 executes, it seems my machine won't allow the nozzle(s) to exit the bed margin and because of the probe's orientation, I'm losing about 20mm from the left side of the bed and roughly 40-50mm from the rear.  I noticed that the probe adheres to the margin, but so does the nozzle during probing.

IDK if this constitutes as a firmware issue, or just a possible conflict in my sketch.  Any insight on how to remedy this is most appreciated.

This is my first post to git, so apologies if I'm not following protocols correctly.  I tried to upload pics, but I guess this is not possible.  I will gladly provide any other info upon request.

R3play210, 2017-12-31:
This specifies where the probe can reach on your bed. 

Example:
>   // Set the boundaries for probing (where the probe can reach).
>   #define LEFT_PROBE_BED_POSITION 15
>   #define RIGHT_PROBE_BED_POSITION 185
>   #define FRONT_PROBE_BED_POSITION 20
>   #define BACK_PROBE_BED_POSITION 170

Also make sure your offsets are set correctly.

Example:
> #define X_PROBE_OFFSET_FROM_EXTRUDER -26 // X offset: -left  +right  [of the nozzle]
> #define Y_PROBE_OFFSET_FROM_EXTRUDER -40 // Y offset: -front +behind [the nozzle]
> #define Z_PROBE_OFFSET_FROM_EXTRUDER -2.40 // Z offset: -below +above  [the nozzle]

These settings can be found in `configuration.h`

Skybound420, 2017-12-31:
My offsets are good as the G28 probes dead center of the bed, however in Easy Config, the BED_POSITION lines are configured much differently to include the inclusion of other variables.  The maker said he'd help me through the bugs after the new year, but I was hoping it could be ironed out sooner.  Thanks for the input.

[Line 1166](https://github.com/JimBrown/MarlinTarantula/blob/EasyConfig-1.1.x/Marlin/example_configurations/Tevo%20Tarantula/Configuration.h)

tcm0116, 2017-12-31:
If you could zip up your configuration files and upload them to this issue, that would help a lot in allowing up to help you. For example, you didn't mention which leveling scheme you're using, which is very pertinent to this issue. 

I don't know much about the fork your referring to, so I'm going to assume that it's very current (i.e. up to version 1.1.8). 

You might want to play around with `X_BED_SIZE`, `Y_BED_SIZE`, `[X|Y|Z]_MIN_POS`, and `[X|Y|Z]_MAX_POS`. You ultimately want these set to the actual travel limits of the printer and not the actual bed size if you want the nozzle to be able to travel outside of the bed. 

When you say that it's probing dead center of the bed, are you referring to the nozzle or the probe being in the center of the bed?

Skybound420, 2018-01-01:
Easy Config consists of grouped variables for those new to IDE and printing in general.  That said, most configurable variables are bunched together in about 150 lines in the beginning of Configuration.h.  This approach was made for quite a few different printers, but mine is the Tevo Tarantula.

Because I custom designed my own printer parts, I was able to greatly extend X travel, so Jim made a set of variables to deal with this during probing, though they don't seem to be functioning as intended.  I have some experience with Arduino sketches, though I've never used anything as large as Marlin and I don't feel comfortable altering any of the values because I don't know which is used in what way(s), or by what other functions/methods.  That said, I simply try to stay within the confines of Easy Config until I gain more experience.

When I said dead center, I meant that my probe was probing the exact center of the bed, but only during G28.  It seems that G29 somehow changes what is considered the center of the bed.  I tried investigating the sketch to see if I could figure anything out, but I couldn't.  Initially, I had some values assigned for the last variables called XTRA_BED_LEFT, XTRA_BED_RIGHT, XTRA_BED_FRONT and XTRA_BED_BACK, but I've since zeroed those out as one of my attempts.  Still the issue persists.

For reasons unknown to me, I am unable to upload the zip folder with the config files, so I will just upload them to my git and link back to this post.

JimBrown, 2018-01-02:
@Skybound420 Actually, the XTRA_BED_* settings are what you need to set to allow the carriage/bed to physically move the probe to the edges of the print surface (with the nozzle potentially going outside the print surface). This assumes you have set the other offsets (nozzle and probe) correctly. You will need to physically measure how far the carriage/bed can actually move past the configured bed size.

For anyone else wondering about my EasyConfig fork, I did it specifically for the Tevo Tarantula, as that is the printer I started with, and still use. There are two branches both kept up to date with the bugfix branches here. It moves/copies the most common settings to the top of Configuration.h (with appropriate comments), simplifies the setting of offsets, probe points, and bed size (it does some math later on in the config file) and adds some sanity checking to those settings.

Skybound420, 2018-01-02:
My issue is now resolved.  To encroach more into the -X axis XTRA_BED_LEFT is needed.  To encroach more into +Y (rear) axis, I had to apply that value to XTRA_BED_FRONT.  Also, like your probe centering scheme, I find I only needed positive values in either of the paired categories.

[image]

closed 2018-02-11 by thinkyhead