MarlinFirmware/Marlin #8974: Delta Printer manual X & Y move strange/odd behavior, Bug?

Ordenkeim93499, 2017-12-30:
I wanted to update my current Marlin Firmware (1.1.5) to Marlin 1.1.8 cause i bought new TMC2130 stepsticks. So i wanted to first get the Firmware working with my old Stepsticks (just wanted to do it in little steps) but this is where this strange thing occured. 
It's just.... i have no clue there the heck this thing comes from.

- ### I found an old issue with the exact same consequences :
   https://github.com/MarlinFirmware/Marlin/issues/7320
  ------------------------------------------------------------
### Bug Report
- **Description:** Delta homes all axis fine and manually travels on the Z axis just as fine,
 if X or Y get moved manually one slider goes crazy and drives itself half way down stops and then goes up again to it's start position 
------------------------------------------------------------
- **Expected behaviour:** X or Y axis should move the wanted way and the expected distance (10mm)
- **Actual behaviour:** if you want to move X by 10mm two of the motors just wobble and the third one drives the slider half way down the linear rail and than back up again to it's start position. The same thing on the Y axis not on Z.
------------------------------------------------------------
- **Steps to reproduce:**
   -    Home Delta
   -    Select X or Y axis to move on LCD
   -    Input any number doesnt matter if it's 10mm 1mm or 0.1mm
   -    See how the axis go crazy :)

[Config Delta.zip](https://github.com/MarlinFirmware/Marlin/files/1594855/Config.Delta.zip)

------------------------------------------------------------
**A few stats to my Printer:** 

- Printer: Anycubic Kossel Delta Plus
- Mainboard: Trigorilla (RAMPS 1.4)
- Firmware: Currently Marlin 1.1.5 (working)
- Stepsticks: Polulu a4988

mylife4aiurr, 2017-12-31:
Try this, Report back if doesnt work
https://github.com/MarlinFirmware/Marlin/issues/8901

thinkyhead, 2017-12-31:
This is now patched in `bugfix-1.1.x` and has been backported to 1.1.8 as well.

closed 2017-12-31 by thinkyhead