MarlinFirmware/Marlin #8975: [2.0.x] Due thread

Bob-the-Kuhn, 2017-12-30:
There's been a LOT of work being done on the Due lately (thanks @ejtagle ) and some of the discussions have gotten spread out over multiple PRs so are hard to follow.  I'm hoping this issue will serve as a common place to discuss all things Due.

Bob-the-Kuhn, 2017-12-31:
Here's one item for the wish list.

While the LPC1768 users are wanting to be able to assign a PWM function to any pin (definitely more pins than the current 6), the Due appears to not have **ANY** useable PWMs.

While working on PINS_DEBUGGING for the Due I noticed that the Arduino library for SAM only implemented 4 hardware PWMs and they're running at 1KHz
```
Due Pin  RAMPS-FD signal
  6      SERVO1_PIN        
  7      SERVO0_PIN        
  8      HEATER_BED_PIN    
  9      HEATER_0_PIN 
```
The two servo pins are already serviced by Marlin/HAL routines that run at 50Hz so they don't (can't) use the hardware PWM function.

Heater pins don't normally use fast PWMs.

I don't see a nice way of adding more PWMs.  The hooks just aren't there for a multi-user friendly system..  As best I can tell we'll have to be like the LPC1768 & write our own **analogWrite** function.  That means we wouldn't be able to use the standard library.

teemuatlut, 2017-12-31:
Due compiling is broken when `SDSUPPORT` is disabled. Unless this has been fixed recently.

@Bob-the-Kuhn I'd like to see us rather use inheritance to modify what we need or even better submit a patch to upstream. The problem with making yet another solution just for Marlin is that it won't be accessible for any outside resources. Similar to how it is with LPC & SPI & TMC.

dembach, 2017-12-31:
For the community i highly recommend to do a fusion with MK4Duo. (also working without SDSUPPORT)

the only reason for this fork was unsolved feature requests by this version of Marlin was made to accommodate some requests made by the community RepRap Italy ...

For now, it is the only complete working 32-bit implementation for such a widespread variant of boards and Hardware.

Both Versions are good, but the best for the community would be , the best of both. 

best regards

Markus

byebyedizzy, 2018-01-03:
For RuRAMPS4Due board there are 6 PWMs (if I'm not wrong):
----
#define HEATER_0_PIN       13
#define HEATER_1_PIN       12
#define HEATER_2_PIN       11
#define HEATER_BED_PIN      7  // BED H1

#define FAN_PIN             9
#define FAN1_PIN            8
----

fiveangle, 2018-01-05:
Somewhat Due related, is anyone looking into why the Due changes in #8853 broke teensy20pp targets ?

-=dave

Roxy-3D, 2018-01-05:
> For the community i highly recommend to do a fusion with MK4Duo. (also working without SDSUPPORT)

We appreciate having Pull Requests submitted for feature that MK4Duo has that we don't have....

Bob-the-Kuhn, 2018-01-05:
@dembach - could you provide a list of the features that could be migrated?

Bob-the-Kuhn, 2018-01-05:
@fiveangle - can you provide more info on the teensy20 issue(s)?

fiveangle, 2018-01-06:
Bob- I have no idea how this is causing the failures in compiling the avr teensy libraries because I don't see how the HAL stuff for Due could be getting picked up, but perhaps you can make heads or tails of it.

I bisected all the changes until I found it's specifically #8853 that did it:

https://github.com/MarlinFirmware/Marlin/issues/8898

-=dave

Bob-the-Kuhn, 2018-01-06:
Thanks - I'll look at 8898 & see what I can do.

Bob-the-Kuhn, 2018-01-06:
Turns out we're both partially correct.

There's actually 12 PWMs assigned to logical pins 2-13.

6, 7, 8 & 9 come from the PWM controller.  The others come from the timers.

OneDayFlie, 2018-01-09:
Hi, 
So i am hooking up a ramps-fd v1... 
But after every boot... my heater starts heating on itself? 
What can i do?
Thanks in advance!

Roxy-3D, 2018-01-09:
That sounds wrong!   I guess you complain about it!!!   

@Bob-the-Kuhn  Due is designed with MOSFET's that don't turn on at reset, right?  If so...  This shouldn't be happening.    Or maybe that was one of the cautions we heard about why Due was a bad 'Reference Platform'.   I can't remember...

Bob-the-Kuhn, 2018-01-09:
The ramps-fd v1 outputs for D8-D11 are inverted from the usual.  

The PINS file has the following
```
#define INVERTED_HEATER_PINS
#define INVERTED_BED_PINS
#define INVERTED_FAN_PINS
```

There's nothing in the Marlin code base that uses these.  Must have been the MK4DUE code base where they were functional.

Looks like another item for the DUE to do list.

alfredanil, 2018-01-09:
FYI ramps-fd v1 to v2.1 are not meant for practical use, it was already cautioned by the developer several times. Although Ramps-Fd V2.2 seems more production ready but unfortuately nobody produces this boardas far as I know.  I use RADDS V1.5 with Repetier, MK4Due and G2Core firmwares without any major issues.

Bob-the-Kuhn, 2018-01-09:
I'm not sure what the V1 issues are.  I haven't actually done anything more than test moves and get a BIG solid state relay to offload the bed.

RADDS V1.5 is a moderately priced board.  How do you connect a RepRap Discount Full Graphics LCD to it?

alfredanil, 2018-01-10:
@Bob-the-Kuhn
As for Ramps-FD V1 to V2.0a, they have some serious problem with heaters going out of control, it is dangerous(*****cautioned by developer). I use Nextion Touch display but I think it is also possible to connect any LCD display including RepRap Discount Full Graphics LCD. It has pins specifically allotted for LCD, encoder and beeper.

HakanBastedt, 2018-01-10:
Tried on my Due Kossel a few weeks ago, ramps-fd v1. The issue is that D8-D11 has signal levels inverted, so mosfets are active low. When booting and when downloading new firmware the mosfets are on and heat up the bed and nozzle and what not. It can be fixed in a few macros by inverting the logic there and make the pwm got from 255-0 instead of 0-255.

I can dig it up if there is interest in it.

HakanBastedt, 2018-01-10:
I had a look. Unfortunately I didn't have the actual changes saved, so this is from memory.
In Conditionals_post.h at line 799:
```
/**
 * Heated bed requires settings
 */
#if HAS_HEATER_BED
  #ifndef MAX_BED_POWER
    #define MAX_BED_POWER 255
  #endif
  #ifndef HEATER_BED_INVERTING
    #define HEATER_BED_INVERTING false
  #endif
  #define WRITE_HEATER_BED(v) WRITE(HEATER_BED_PIN, (v) ^ HEATER_BED_INVERTING)
#endif
```
So there is already prepared for inverting the bed heater pin. in pins_RAMPS_FD_V1.h, the inverting define should be renamed to HEATER_BED_INVERTING.
A few lines above in the same file there is this chunk
```
/**
 * Helper Macros for heaters and extruder fan
 */
#define WRITE_HEATER_0P(v) WRITE(HEATER_0_PIN, v)
#if HOTENDS > 1 || ENABLED(HEATERS_PARALLEL)
  #define WRITE_HEATER_1(v) WRITE(HEATER_1_PIN, v)
  #if HOTENDS > 2
    #define WRITE_HEATER_2(v) WRITE(HEATER_2_PIN, v)
    #if HOTENDS > 3
      #define WRITE_HEATER_3(v) WRITE(HEATER_3_PIN, v)
      #if HOTENDS > 4
        #define WRITE_HEATER_4(v) WRITE(HEATER_4_PIN, v)
      #endif // HOTENDS > 4
    #endif // HOTENDS > 3
  #endif // HOTENDS > 2
#endif // HOTENDS > 1
#if ENABLED(HEATERS_PARALLEL)
  #define WRITE_HEATER_0(v) { WRITE_HEATER_0P(v); WRITE_HEATER_1(v); }
#else
  #define WRITE_HEATER_0(v) WRITE_HEATER_0P(v)
#endif
```
This needs to have the same treatment as the bed macro, and the define should be something like HEATER_0_INVERTING.
Then there is the FAN pin. In module/planner.cpp at line 350 the pwm value needs to be decreasing from 255-0 or 255-FAN_MIN_PWM instead of increasing, when the fan pin is inverted.
```
    #ifdef FAN_MIN_PWM
      #define CALC_FAN_SPEED(f) (tail_fan_speed[f] ? ( FAN_MIN_PWM + (tail_fan_speed[f] * (255 - FAN_MIN_PWM)) / 255 ) : 0)
    #else
      #define CALC_FAN_SPEED(f) tail_fan_speed[f]
    #endif
```
Also this code at inc/Conditionals_post.h at line 834 needs to have inverting code (from first snippet) inserted
```
#if HAS_FAN0
  #define WRITE_FAN(v) WRITE(FAN_PIN, v)
  #define WRITE_FAN0(v) WRITE_FAN(v)
#endif
#if HAS_FAN1
  #define WRITE_FAN1(v) WRITE(FAN1_PIN, v)
#endif
#if HAS_FAN2
  #define WRITE_FAN2(v) WRITE(FAN2_PIN, v)
#endif
#define WRITE_FAN_N(n, v) WRITE_FAN##n(v)
```

AnHardt, 2018-01-10:
- * - = + // doubled negation!
Either change the PWM or the polarity of the pin - not both.
Inverting the pin makes more sense to me.

dembach, 2018-01-10:
@Bob-the-Kuhn : 
since MK4Duo was made for the italian RepRap community , there are some features, @MagoKimbra  could describe best...
just some lines of the Intro:

- System Management MKr4 for 4 extruders width just two drivers and 8 relay.

- System Management MKr6 for 6 extruders width just two drivers and 8 relay.

- System Management MKr12 for 12 extruders width just four drivers and 16 relay.

- Management Multyextruder NPr2, 4/6 extruders with only two engines.

- Management Dual Extruder DONDOLO.

- Added DHT Sensor Temperature and Humidity (DHT11, DHT21 and DHT22)

- Added Extruder Encoder for control filament movement

- Added Cooler and Hot Chamber

- Added support NEXTION lcd touch

it is up to you, what should become available in Marlin too... My Opinion is, if we can support the italian RepRap Community with Marlin and the contributors of MK4Duo settle over to Marlin-Team, it would be the best.

thanks

Markus from Germany

MagoKimbra, 2018-01-10:
```
System Management MKr4 for 4 extruders width just two drivers and 8 relay.
System Management MKr6 for 6 extruders width just two drivers and 8 relay.
System Management MKr12 for 12 extruders width just four drivers and 16 relay.
```
This system was born more than 4 years ago to be able to use the ramps with 4 extruders. Today it is also used by Prusa for its MK3, Marlin already has the switch system used by Prusa to have 4 extruders with a driver. It would be missing 6 on the 12th, but it is easy to implement. This year at the Maker Faire in Rome we presented the 12 colors with 12 extruders and 4 drivers.

`Management Multyextruder NPr2, 4/6 extruders with only two engines.`
This system is not widely used, it is not worth implementing it.

Management Dual Extruder DONDOLO.
Also this system was born a long time ago before it was also used by Marlin with the name of switching extruder, therefore already implemented.

Added DHT Sensor Temperature and Humidity (DHT11, DHT21 and DHT22)
Support for DHT11 or DHT22 sensor for temperature and Humidity, Great for chamber. Only for SAM.

Added Extruder Encoder for control filament movement
This could also be good to improve it compared to what was done in MK.

Added Cooler and Hot Chamber
Support Cooler for Laser CO2 and Hot Chamber

Nextion
This part also needs graphics and maybe even more software cleaning.

MagoKimbra, 2018-01-10:
I need information about how many bytes is the eeprom for UBL. I asked this to the developer.

Roxy-3D, 2018-01-10:
UBL only uses a few bytes of EEPROM storage in the main configuration area.  It keeps track of whether UBL is activated (at save time so it can be restored at power up).  It also keeps track of what the last used mesh number is (at save time so that mesh can be restored).

The empty space from the end of the EEPROM configuration area to the end of the EEPROM is then determined to be 'free' and can be used for mesh storage.     Each mesh point is stored as a float.   So a 10 x 10 mesh will consume 400 bytes of the empty EEPROM area.     A 15 x 15 mesh will consume 900 bytes of the space.    UBL will allow as many different mesh patterns to be saved as there is space.

The amount of storage the mesh data takes can easily be cut in half, but there has been no reason to do that.    Simply storing the mesh points as an int would allow 3 digits of precision with a +/- 32 mm difference from 0.000 mm.

MagoKimbra, 2018-01-10:
No sorry @Roxy-3D I was wrong it was not for this threat... Sorry...

dembach, 2018-01-11:
@MagoKimbra , it was meat, you can decide, what part/Feature it worthful to be ported to Marlin of your Repository MK4Duo.

HakanBastedt, 2018-01-11:
@AnHardt: It depends, for softpwm the signal can be inverted, when AnalogWrite is used the pwm value must be backwards. It was from memory though, and there is not really the need look at the fan. I now have a pretty clean version here which only inverts the heaters and the bed on the large screw connectors D8-D11. Fan is not considered, it works correctly when connected to D2 and D12, the small connector. I think this is a reasonable approach.

end of pins_RAMPS_FD_V1.h
```
#if defined(HEATER_0_PIN)
  #if (HEATER_0_PIN >= 8 && HEATER_0_PIN <= 11)
    #define HEATER_0_INVERTING
  #endif
#endif

#if defined(HEATER_1_PIN)
  #if (HEATER_1_PIN >= 8 && HEATER_1_PIN <= 11)
    #define HEATER_1_INVERTING
  #endif
#endif

#if defined(HEATER_2_PIN)
  #if (HEATER_2_PIN >= 8 && HEATER_2_PIN <= 11)
    #define HEATER_2_INVERTING
  #endif
#endif

#if defined(HEATER_3_PIN)
  #if (HEATER_3_PIN >= 8 && HEATER_3_PIN <= 11)
    #define HEATER_3_INVERTING
  #endif
#endif

#if defined(HEATER_BED_PIN)
  #if (HEATER_BED_PIN >= 8 && HEATER_BED_PIN <= 11)
    #define HEATER_BED_INVERTING
  #endif
#endif
```
And changes in Conditionals_post.h 

```

/**
 * Helper Macros for heaters and extruder fan
 */
#ifndef HEATER_0_INVERTING
  #define WRITE_HEATER_0P(v) WRITE(HEATER_0_PIN, v)
#else
  #define WRITE_HEATER_0P(v) WRITE(HEATER_0_PIN, !(v))
#endif
#if HOTENDS > 1 || ENABLED(HEATERS_PARALLEL)
  #ifndef HEATER_1_INVERTING
    #define WRITE_HEATER_1(v) WRITE(HEATER_1_PIN, v)
  #else
    #define WRITE_HEATER_1(v) WRITE(HEATER_1_PIN, !(v))
  #endif
  #if HOTENDS > 2
    #ifndef HEATER_2_INVERTING
      #define WRITE_HEATER_2(v) WRITE(HEATER_2_PIN, v)
    #else
      #define WRITE_HEATER_2(v) WRITE(HEATER_2_PIN, !(v))
    #endif
    #if HOTENDS > 3
      #ifndef HEATER_3_INVERTING
        #define WRITE_HEATER_3(v) WRITE(HEATER_3_PIN, v)
      #else
        #define WRITE_HEATER_3(v) WRITE(HEATER_3_PIN, !(v))
      #endif
      #if HOTENDS > 4
        #ifndef HEATER_4_INVERTING
          #define WRITE_HEATER_4(v) WRITE(HEATER_4_PIN, v)
        #else
          #define WRITE_HEATER_4(v) WRITE(HEATER_4_PIN, !(v))
        #endif
      #endif // HOTENDS > 4
    #endif // HOTENDS > 3
  #endif // HOTENDS > 2
#endif // HOTENDS > 1
#if ENABLED(HEATERS_PARALLEL)
  #define WRITE_HEATER_0(v) { WRITE_HEATER_0P(v); WRITE_HEATER_1(v); }
#else
  #define WRITE_HEATER_0(v) WRITE_HEATER_0P(v)
#endif

/**
 * Heated bed requires settings
 */
#if HAS_HEATER_BED
  #ifndef MAX_BED_POWER
    #define MAX_BED_POWER 255
  #endif
  #ifndef HEATER_BED_INVERTING
    #define WRITE_HEATER_BED(v) WRITE(HEATER_BED_PIN, v)
  #else
    #define WRITE_HEATER_BED(v) WRITE(HEATER_BED_PIN, !(v))
  #endif
#endif
```

A question: Should 2.x actually work on Due? I have it set up on a delta, and the X axis need like five times more steps than the X and Y and the step motors sounds like full of gravel. I can barely complete the homing. Also fans behaves wrong way, only soft_pwm works correct.

maukcc, 2018-01-12:
We have a fully working MK4duo on one of our test machines (incl. SD and graphic lcd support)
The things we would really miss on marlin are:

Added DHT Sensor Temperature and Humidity (DHT11, DHT21 and DHT22)
We use this in our humidity controlled filament storage

Added Extruder Encoder for control filament movement

Added Hot Chamber

Added support NEXTION lcd touch

maukcc, 2018-01-26:
Also the DOOR_PIN functionality would be a must for any safety regulations to control the laser

closed 2019-02-21 by Bob-the-Kuhn

TwatBurglar, 2019-03-07:
@Bob-the-Kuhn 

> @byebyedizzy - turns out we're both partially correct.
> 
> There's actually 12 PWMs assigned to logical pins 2-13.
> 
> 6, 7, 8 & 9 come from the PWM controller. The others come from the timers.

@Bob-the-Kuhn  Do you know if any 25khz PWMs (for laser power) for DUE were implemented?   I had a look through the code, it looks like no :(

Bob-the-Kuhn, 2019-03-07:
I doubt it.  I also didn't see any hint of 25KHz PWMs.