MarlinFirmware/Marlin #8976: PRINTRBOARD_F6_HEATER_FAN_PIN in pins_PRINTRBOARD_REVF.h is not defined

cookew, 2017-12-31:
```PRINTRBOARD_F6_HEATER_FAN_PIN``` as mentioned in the comments after ```define DISABLE_JTAG``` in the pins_PRINTRBOARD_REVF.h file does not exist and should probably be changed to say ```44```.  It is the pin on the REVF6 board where the fan should probably be connected to, right next to the other fan.

cookew, 2017-12-31:
Might be a good idea to mention to disable HEATERS_PARALLEL in Configuration_adv.h for the auto fan to work properly. The fan will only turn on when heating if HEATERS_PARALLEL is enabled.

I am not using an Extrudrboard, so I don't know how changing the pins to work with all of that will change the required settings for this to all work properly.

fiveangle, 2018-01-02:
@cscott can you comment ?  These were your changes in #7286

thinkyhead, 2018-02-11:
@cscott

closed 2018-02-11 by thinkyhead