MarlinFirmware/Marlin #8985: G29 P1 off center

black4hawk, 2017-12-31:
- Description: 
Printer: 
Anycubic Kossel Linear (180mm printable, 200mm bed)
Marlin 1.1.8, 
- Expected behaviour:  G29 P1 probing center, right height output
- Actual behaviour: g29 P1 Probes off center (manual input) (edge of bed)
  "Set Delta Height" in  delta calibration menu checks middle, spits out wrong delta height (4.4mm off)
   g33 probes as with previous version but spits out a wrong delta height
   "https://github.com/jam3sward/Marlin/tree/1.1.5-Anycubic-Linear" 

- Steps to reproduce:
 manual input g29 P1 / g33 into interface

FIRMWARE:
-still changing (steps per unit etc not compiling (allready fixed),. not finally configured
https://drive.google.com/open?id=1UUMGfIDIJqZnV84dZ7M66R6BPWYRNF3A

AnHardt, 2018-01-01:
@black4hawk 
The math calculating the carriage positions is unable to handle more than horizontal/vertical 'diagonal rods'.
Despite of that i see a more than horizontal position in your video.
-> Configure correct values for your machine!

thinkyhead, 2018-02-11:
Presuming this is solved.

closed 2018-02-11 by thinkyhead