MarlinFirmware/Marlin #8987: Strange delta during manual movement of axis

filipgd, 2017-12-31:
Hi 
I discovered some strange behavior of my firmware during delta movement configuration.
It took me few hours to check code but I can't get what is wrong.
When I move axis slowly 0,1mm it works just fine - all three axis moves little bit.
When I move axis faster or send G-code (G0X1) one axis goes about 100mm down and back.
Don't know why its moving like this.

https://youtu.be/O6UU3tznyw4

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1595402/Marlin.zip)

nmblufc, 2018-01-01:
I have exactly the same issue with my delta and Marlin 1.1.8 do please post an update if you find a solution.
I have quickly reviewed your configuration and it is very similar to my file.

Cheers

filipgd, 2018-01-01:
I downgrade to 1.1.5 and it is Ok
But 1.1.5 have poot delta menu so i would like to get new one

nmblufc, 2018-01-01:
Thanks for reply.

Whilst looking for an older firmware version I noticed that V2 marlin is available so loaded that and instantly solved the problem.
Just thought I would let you know as you need better menu options.

p3p, 2018-01-01:
1.1.8 was re-released to fix a serious delta movement issue are you sure this was in the re-release, 2.0.x should be functionally the same as 1.1.x for 8bit AVR.

nmblufc, 2018-01-01:
I will check out 1.1.8 again with latest release but for now V2 is working and I can get on with calibration etc.
Thanks

filipgd, 2018-01-02:
p3p - Yes it is 1.1.8 with AVR mega
Think its bug because if i move slow it looks ok. 

Already checked 2.0 and there is also issue with autocalibration will post it in another thread.
Still working on 1.1.5 :]

Ordenkeim93499, 2018-01-02:
I discovered the same problem and also opened and issue but it got resolved.
https://github.com/MarlinFirmware/Marlin/issues/8974
This is probably gonna help you, also download Marlin 1.1.8 again this issue was fixed but in the same version of Marlin

closed 2018-02-11 by thinkyhead