MarlinFirmware/Marlin #8992: Turn off servo signal pin to 0v via marlin?

Erfquake, 2018-01-01:
I have a tevo tornado using MKS base 1.4 ramps board:
M42 P4,5,6,7,11 S0 does not turn turn off the signal pin on the servo connection.  It always hover around 300 mV and will vary depending on the M280 command?

Is there a way via Marlin to detach that pin so that it is completely 0?

thinkyhead, 2018-02-11:
If you disable all servos and probes in your configuration do you still see voltage on the pin?

boelle, 2019-02-19:
@Erfquake  Please post your question either on discord: https://discord.gg/n5NJ59y or on facebook: https://www.facebook.com/groups/2080308602206119/
The issue list is for bugs and feature requests only
Please close this issue once you have posted it on one of the 2 links
Thanks :-D

AnHardt, 2019-03-03:
> // Only power servos during movement, otherwise leave off to prevent jitter
//#define DEACTIVATE_SERVOS_AFTER_MOVE

closed 2019-03-06 by thinkyhead