MarlinFirmware/Marlin #8993: lcd_temp_menu_e0_filament_change

gucio1200, 2018-01-01:
------------------------------------------------------------
### Compile Error

sketch/ultralcd.cpp: In function 'void lcd_tune_menu()':
ultralcd.cpp:1396: error: 'lcd_temp_menu_e0_filament_change' was not declared in this scope
           MENU_ITEM(submenu, MSG_FILAMENTCHANGE, lcd_temp_menu_e0_filament_change);
                                                  ^
sketch/ultralcd.cpp:301:30: note: in definition of macro '_MENU_ITEM_PART_2'
         menu_action_ ## TYPE(__VA_ARGS__); \
                              ^
sketch/ultralcd.cpp:315:39: note: in expansion of macro 'MENU_ITEM_P'
   #define MENU_ITEM(TYPE, LABEL, ...) MENU_ITEM_P(TYPE, PSTR(LABEL), ## __VA_ARGS__)
                                       ^
sketch/ultralcd.cpp:1396:11: note: in expansion of macro 'MENU_ITEM'
           MENU_ITEM(submenu, MSG_FILAMENTCHANGE, lcd_temp_menu_e0_filament_change);
           ^
sketch/ultralcd.cpp: In function 'void _lcd_ubl_validate_custom_mesh()':
sketch/ultralcd.cpp:2102:37: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
       lcd_enqueue_command_sram("G28");
                                     ^
sketch/ultralcd.cpp: In function 'void lcd_prepare_menu()':
ultralcd.cpp:2595: error: 'lcd_temp_menu_e0_filament_change' was not declared in this scope
             MENU_ITEM(submenu, MSG_FILAMENTCHANGE, lcd_temp_menu_e0_filament_change);
                                                    ^
sketch/ultralcd.cpp:301:30: note: in definition of macro '_MENU_ITEM_PART_2'
         menu_action_ ## TYPE(__VA_ARGS__); \
                              ^
sketch/ultralcd.cpp:315:39: note: in expansion of macro 'MENU_ITEM_P'
   #define MENU_ITEM(TYPE, LABEL, ...) MENU_ITEM_P(TYPE, PSTR(LABEL), ## __VA_ARGS__)
                                       ^
sketch/ultralcd.cpp:2595:13: note: in expansion of macro 'MENU_ITEM'
             MENU_ITEM(submenu, MSG_FILAMENTCHANGE, lcd_temp_menu_e0_filament_change);
             ^

tsmaa, 2018-01-02:
compilation error on file ultralcd.cpp line 2592, ramps 1.4 EFB, 1.1.8 bugfix
'lcd_temp_menu_e0_filament_change' was not declared in this scope

'lcd_temp_menu_e0_filament_change' called from line 2592 before defined

marcio-ao, 2018-01-03:
Could you let me know which branch this is in? I can't find "lcd_temp_menu_e0_filament_change" in the source. I made an initial PR for multiple extruder filament changes, but it did not have that particular function name in it, so it must be a later addition. If you tell me which branch you are using and what Marlin version, maybe I can help track down the problem.

tsmaa, 2018-01-03:
> 1.1.8 bugfix
downloaded yesterday then compiled

marcio-ao, 2018-01-03:
@tsmaa , @gucio1200 : Thank you, I've identified the pull request that introduced this bug and linked this bug report to it so that hopefully it can get addressed.

tsmaa, 2018-01-03:
Glad to see you.
I'm happy to help Marlin community

gucio1200, 2018-01-03:
My error disappear in bugfix last commit.

marcio-ao, 2018-01-03:
@gucio1200: I can confirm that it compiles under the Arduino IDE, but I believe it may still have issues if compiled using the Makefile and avr-gcc.

tsmaa, 2018-01-03:
Yes, Compilation completed without error under Arduino 1.8.5, bugfix version committed 5 hours ago and downloaded today

closed 2018-01-04 by thinkyhead