MarlinFirmware/Marlin #8995: Blank screen in subsequent 3-point bed leveling runs

Physicator, 2018-01-01:
### Bug Report
Hi, this is a minor but somewhat annoying issue which apparently is related to #6790.
I configured Marlin for 3-point manual bed leveling. When the procedure is invoked initially (after powering up the printer) everything works perfectly. But during subsequent bed leveling runs the display essentially remains empty. That is, the messages "Homing XYZ", "Click to Begin", "Next Point...", "Move Z:...", "Leveling Done!" are not displayed. Some of them do appear when rotating the encoder, though. This issue seems to be independent of intermediate actions such as returning back to the info screen or disabling bed leveling in the menu. I tried to pinpoint the issue in the code (G29 and ultralcd) but I am not familiar enough with the concepts and structure, so I eventually gave up.
Please let me know if you need further information.
Thanks!

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1596089/Configuration.zip)

Physicator, 2018-01-03:
Little Update: Actually, after printing a model the display is working properly again during bed leveling, at least during the first leveling run. The problem still occurs in subsequent leveling runs then.

thinkyhead, 2018-02-11:
This has affected all forms of LCD-based leveling. I spent a long time trying to diagnose, but don't recall if any of my attempted patches worked.

Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see if we fixed this issue. If the problem has been solved then we can close it. If you still see the bad behavior we should investigate further.

Physicator, 2018-02-12:
Hm, I am totally new to GitHub (my first post), and the files in the bugfix-1.1.x branch just look like a mess for me. Could you indicate which filkes I need to update? Thanks a lot!

Roxy-3D, 2018-02-13:
Look in the https://github.com/MarlinFirmware/MarlinDocumentation area.   There will be documentation to help you in which ever direction you are going.    If you want a more organized file layout, use the bugfix-2.0.0 branch.   It is almost feature compatible with bugfix-1.1.x.   But it has the files laid out in a more organized way.

thinkyhead, 2018-02-20:
@Physicator — I've confirmed this is still an issue. So far no solution but we'll keep trying things.

closed 2019-05-16 by thinkyhead