MarlinFirmware/Marlin #8997: Different behaviour of babystepping when BABYSTEP_ZPROBE_OFFSET btw lcd and m290

GMagician, 2018-01-01:
z babystepping is always done (lcd and m290) when BABYSTEP_ZPROBE_OFFSET is disabled but when enabled lcd babystepping is blocked if not `planner.leveling_active`.

Is there a reason for such different behaviour?

Roxy-3D, 2018-01-01:
I don't think that makes sense...    At least for me, if I'm Baby Stepping, I'm not sure I want the machine second guessing me.   Baby Stepping moves slow enough (and I'm in control of the movement), I don't think we need to worry about any damage.

My opinion is we need to pull out the 'if' part and just do the  'then' clause unconditionally...
```
        if (planner.leveling_active)
              thermalManager.babystep_axis(Z_AXIS, babystep_increment);
```

GMagician, 2018-01-01:
@Roxy-3D this is my idea too but since I don't have enough experience and historical knowledge maybe there was a reason to do that

Roxy-3D, 2018-01-01:
I think this is safe to change...   I'll change it in both BugFix branches...

closed 2018-01-02 by GMagician