MarlinFirmware/Marlin #8998: Help programming  :  classroom

studiodyne, 2018-01-01:
Works fine
`MENU_ITEM(function, MSG_FILAMENTCHANGE " " MSG_E1, lcd_enqueue_filament_change_e0);`
but
when i make a function like this
```
void  lalalalalla () {
MENU_ITEM(function, MSG_FILAMENTCHANGE " " MSG_E1, lcd_enqueue_filament_change_e0);
}
```

Roxy-3D, 2018-01-01:
Declare your lcd_enqueue_filament_change_e0() function immediately above the function where you are trying to put the statement:
```
MENU_ITEM(function, MSG_FILAMENTCHANGE " " MSG_E1, lcd_enqueue_filament_change_e0);
```
That probably will resolve the scope problem.  (But it might not be the 'clean' way to get things defined.)

studiodyne, 2018-01-02:
Hello @Roxy-3D 
Thks for your help , but finally it was a school old problem
Scope is a unit , some {} , but scope is also ' #if  + #endif ' 
And i have made a function with some inside function that have been created under ' #if  + #endif '  , and just for this , the compilator EXPLODE WITHOUT SAYING ' not under compilator scope '

I'm beginner 😄

closed 2018-01-02 by studiodyne