MarlinFirmware/Marlin #9005: [BR] Bed PID Autotune kills Heater Immediately

houseofbugs, 2018-01-01:
Running 1.1.8 release with PIDTEMPBED enabled. When going to tune the bed it immediately kills the printer. Tried on a Melzi board and MKS Gen 1.4. Same result on both.

> Send: M303 E-1 C8 S60
> Recv: PID Autotune start
> Recv:  T:23.19 /0.00 B:23.41 /0.00 @:0 B@:127
> Recv: Error:Heating failed, system stopped! Heater_ID: bed
> Changing monitoring state from 'Operational' to 'Error: Heating failed, system stopped! Heater_ID: bed
> '
> Recv: Error:Printer halted. kill() called!
> Changing monitoring state from 'Error: Heating failed, system stopped! Heater_ID: bed
> ' to 'Offline'

Ordenkeim93499, 2018-01-02:
Maybe you can upload your Configuration.h and Configuration_adv.h just to see if there is something wrong

revilor, 2018-01-04:
PR  #9037 fixes the issue, or setting WATCH_BED_TEMP_PERIOD in Configuration_adv.h to no more than 127 seconds

closed 2018-01-06 by houseofbugs