MarlinFirmware/Marlin #9011: [FR] ? Why bad auto level when bad Z homing ?

spilz87, 2018-01-02:
Hello, not sure it's a feature request or if the way other people want...

Relase 1.1.8
I'm using safe Z homing (in the middle of the bed).
I use AUTO_BED_LEVELING_BILINEAR on the front left quarter of the bed (Z homing is outside of the auto bed leveling) 

When the bed is not clean where the Z homing is done, so Z homing not perfect, it affects the auto level.

Is it normal ??? 
Why auto level is not "autonomous" ??

Thanks for your help
Regards

Tannoo, 2018-01-03:
Leveling does care about the homing positions. Homing the machine is to put the machine in a known position.
Leveling is to find the bed...even if it's past the Z homing position.

spilz87, 2018-01-03:
I'm ok with that, but after leveling, as we know where is the bed, moving to Z = 0 should put the nozzle exactly on the bed (if everything else is ok, not depending the Z homing reference.

Actually I'm doing the same test with release 1.1.6 : the bed leveling is not affected by the Z homing
So I think it's a bug of the 1.1.8

Mechotronic, 2018-01-08:
Are you using M420 S1 after your G28 to enable ABL?

spilz87, 2018-01-08:
I don't use it for the test, but I plan to use it in my start gcode
but my first layer is not good, even with G29 command

Mechotronic, 2018-01-08:
If you run G28 after G29 it turns off ABL. You have to use M420 S1 after G28 for any prints or testing.

(G29 > M500 (this only needs to be done occasionally, not before each print))  > G28 > M420 S1

spilz87, 2018-01-08:
I do not run G28 after G29.
actually the ABL seems to work as there is Z mouvement when I move in X and Y 
But there is an offset depending on the position off the Z homing (done before the G29), and it's not normal I think

spilz87, 2018-01-08:
just try the small gcode I posted

one time normally and check the nozzle is close to the bed when you run G1 Z0
second time with something between the bed and the Z probe during the G28 then remove it before the G29. You will not get the nozzle on the bed when you run G1 Z0 (on 1.1.8 )

you will not have this issue with the 1.1.6 release

thinkyhead, 2018-01-09:
See if you get better results using a single `G28`. Two of them is not needed. And note that the `X`, `Y`, `Z` parameters don't take values. They are just flags.

thinkyhead, 2018-01-09:
Before proceeding further, please test `bugfix-1.1.x` to see if we already patched your issue.

spilz87, 2018-01-13:
I just test it with the last 1.1.8 bug fix code, the issue is still there :(
Actually I have to go back to 1.1.6 to don't have it

spilz87, 2018-01-13:
Biggest is the part you put on the bed to create an offset on Z homing
More the nozzle try to go under the bed
Like if there is a bad calcul between Z home and bed leveling

spilz87, 2018-01-13:
After other test, my issue is maybe from somewhere else :
Before (until 1.1.6 I guess ) Z offset probe was not clear in EEPROM 
But in 1.1.8 the value is replace by the one in configuration.h file 
My issue is maybe from here : during tests I switch so much time from one to an other release that I didn't check the value ( which didn't change before)

spilz87, 2018-03-06:
Actually I did other test and I still have the issue.
I think it comes from Leveling fade Z height (M420 Zx ) : 
if fade Z height = 0 (so disabled), Z homming doesn't affect autolevel, 
if fade Z height = 1, Z homming does affect autolevel.

if you do a Z homming in the middle of the print, you doesn't see it, but if not, there is an offset, the Z_values medium is not null and this is the issue because Leveling fade Z height over compensate for this offset.

For me at the end of the autolevel, Z values medium should be calculated and deducted to Z values, and this medium is added (or deducted) to actual_position(Z_axis). like this Z=0 is really the medium bed level

spilz87, 2018-03-06:
actually I realy see this issue when I change my glass on the bed (they have not the same mesh) or when I change the position of the print.
If you stay always at the same position for print, with the same bed config, without knowing it, you have compensate the offset by changing your Z_offset value. But it's not mathematically correct.

Hope it can help
Sorry for my bad English, I can explain differently if you don't understand

spilz87, 2018-03-06:
how can I add label ENABLE_LEVELING_FADE_HEIGHT and AUTO_BED_LEVELING_BILINEAR ?

thinkyhead, 2018-03-07:
Labels are for administrators only.

boelle, 2019-02-19:
@spilz87 please try with latest bugfix 2.0 if your post is a question then Please post your question either on discord: https://discord.gg/n5NJ59y or on facebook: https://www.facebook.com/groups/2080308602206119/
The issue list is for bugs and feature requests only
Please close this issue once you have posted it on one of the 2 links
Thanks :-D

spilz87, 2019-02-20:
Actually it’s a retorical question, I guess the answer is “no it’s not normal”. 
I tried it with bugfix 2.0 the issue was still their.

spilz87, 2019-02-20:
No
In French we call it “rhetorical question”. There is a “?” But you are not waiting for a response. 
Just delete the “?” In the title if it’s a problem for you.

boelle, 2019-03-17:
try with this start code

```
G21
G90
G28
G1 Z15
G29
G1 X100 Y100
G1 Z0
```
still the same problem?

boelle, 2019-07-21:
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle

InsanityAutomation, 2019-07-23:
To explain this a bit better for @spilz87 G28 and G29 are totally separate routines. In some cases, users have different home switches from the probe. A variable exists as the difference between them. When G28 sets 0, that step position is locked in. There may be a change to a different switch, probe moved manually or by solenoid, who knows. Then G29 comes in and checks the bed for position difference from that known 0.

If we could know for certain nothing was moved at all, we could in theory sanity check that if you probe with G29 at the same point as G28 homes and get a big deviation something is wrong, however that is reaching pretty far into preventing users from making simple mistakes. 

Quite simply they are isolated functions as they may be done with independent hardware or with physical adjustments to the same hardware.

spilz87, 2019-07-23:
Thanks for your explanation @InsanityAutomation 
Does that explain an other issue I get : 
I turn on 
G28 
Then G29 
Then G1 Z0
-> nozzle perfect on the bed
Do it again without restart : 
G28
Then G29
-> nozzle distance from bed is equal to the delta found during G29 :(

??

InsanityAutomation, 2019-07-23:
@spilz87 yes, because first command has no bilinear compensation, second has some compensation.

spilz87, 2019-07-23:
@InsanityAutomation thank
So what is the solution to run a second print without restart the printer ?
What start gcode can I had to cancel this compensation on the start of the second print ?

InsanityAutomation, 2019-07-23:
@spilz87 Simply dont rerun G29 every time. Run it once, store the mesh with M500, and activate it with M420 S1 in youre start gcode.

spilz87, 2019-07-23:
I tried it, but as I change the glass on the bed often, I need to do a G29 on every start gcode

InsanityAutomation, 2019-07-23:
If it fits on youre board, UBL may be more suitable. You can store multiple mesh's in different slots and select between them.

spilz87, 2019-07-23:
Something I don’t understand, G28 should disable bed leveling according to documentation, G29 will enable it again before probing ??

spilz87, 2019-07-23:
Actually I don’t want to check which glass I put and in which way and if I didn’t change the bed high when I remove the last glass etc. 
Bed leveling is to simplify this for me

InsanityAutomation, 2019-07-23:
G28 disables it before, and can re-enable it after if you turned on that configuration option. Typically the Z offset will repeat if you set it after G29, so even after subsequent G29 runs it should be ok. 

For a commissioning setup, I would heat the bed and nozzle, let the bed soak at temp for 5min or so, Run G28; G29; G1 (x/y to mid position) G1 Z0 then babystep the z offset (make sure M851 is merged with the z offset so it stores it) then M500. From there it should work as expected with different plates.

spilz87, 2019-07-23:
Z offset is ok for first print, but on second it is no more ok :( that’s the point ...

InsanityAutomation, 2019-07-23:
With M851 combined with babystepping or without?

spilz87, 2019-07-23:
I don’t use M851
I directly use the Z offset through lcd configuration menu and save it in eeprom.

InsanityAutomation, 2019-07-23:
Since youre using a probe for homing and want that to persist, you must use 

#define BABYSTEP_ZPROBE_OFFSET          // Combine M851 Z and Babystepping

spilz87, 2019-07-23:
I Don’t really understand what you suggest : 
M851 is to define offset between probe and nozzle, mine is good after a turn on, and it doesn’t dépend of the glass I use. So for me I don’t need to use M851 which is for people who need to change this offset, which I don’t need. 
But if I run a second print, the G29 is like it modify the value 2 times and the offset is no more good.  :(

spilz87, 2019-07-24:
I don’t need to babystep anymore for the first print after a restart, 
But I do need it for second print without a restart

spilz87, 2019-07-24:
When my print will finish I will test a gcode that show the issue and post it here

InsanityAutomation, 2019-07-25:
In that case its likely something isnt storing to eeprom, has it been initialized since flashing with M502; M500; ? Can you post a terminal log from when the machine is powered up?

spilz87, 2019-07-25:
I initialized the eeprom with M502 and save with M500
 you can try this which is OK : 

> G28
> G29
> G1 X100 Y100
> G1 Z0
> M0 S5 
> ; Nozzle is on the bed
> G28
> G29
> G1 X100 Y100
> G1 Z0
> ; Nozzle is on the bed

and this which is not OK : 

> G28
> G29
> M420 S1;Enable compensation using current grid/mesh
> M420 Z1;Gradually reduce compensation until Z=1
> G1 X100 Y100
> G1 Z0
> M0 S5 Premier Test
> ; Nozzle is on the bed
> G28
> G29
> M420 S1;Enable compensation using current grid/mesh
> M420 Z1;Gradually reduce compensation until Z=1
> G1 X100 Y100
> G1 Z0
> ; Nozzle is NOT on the bed

this one neither : 

> M420 S0;Disable compensation using current grid/mesh
> G28
> G29
> M420 S1;Enable compensation using current grid/mesh
> M420 Z1;Gradually reduce compensation until Z=1
> G1 X100 Y100
> G1 Z0
> M0 S5 Premier Test
> ; Nozzle is on the bed
> M420 S0;Disable compensation using current grid/mesh
> G28
> G29
> M420 S1;Enable compensation using current grid/mesh
> M420 Z1;Gradually reduce compensation until Z=1
> G1 X100 Y100
> G1 Z0
> ; Nozzle is NOT on the bed

spilz87, 2019-07-25:
Looks like the issue is from M420 but don’t really understand how to solve it, M420 S0 should disable it

lrpirlet, 2019-07-26:
@spilz87 
Hopefully you realize that 1.1.9 is frozen software that will not be patched anymore… at fortiori 1.1.8... Today bugfix 2.0 only will receive patches.

Now, if you do use G10-G11 (firmware-based retract and recover) you may benefit from going to 1.1.9... [See  the release 1.1.9](https://github.com/MarlinFirmware/Marlin/releases/tag/1.1.9) and in particular the [Fix the Z un-hop on G11](https://github.com/MarlinFirmware/Marlin/commit/2b49f8ee653843617d9251cb031854ac653ce18d) … I had Z 'incoherences' under 1.1.8 more than one year ago because of that (my solution was to zapp the firmware based retract)

Hope that helps

lrpirlet, 2019-07-26:
Humm
Thinking a little more, the difference between the first G28 and the second G28 is the fact the the motors may have not been released. this could make the printer to believe that position is still known …

What about issuing [M18 ](http://marlinfw.org/docs/gcode/M018.html) to Disable all steppers immediately... but move the nozzle to a safe position first.

spilz87, 2019-07-26:
@lrpirlet 
Actually now all my boards are on bugfix-2.0.x

Even if motors have not been released, it shouldn’t affect G28 and G29

boelle, 2019-07-26:
do you have 

//#define RESTORE_LEVELING_AFTER_G28

or 

#define RESTORE_LEVELING_AFTER_G28

?

spilz87, 2019-07-26:
I have 
#define RESTORE_LEVELING_AFTER_G28

That can explain it 
I  ever see this option before,
I will test it

But why G29 doesn’t erase it ?

boelle, 2019-07-26:
you also have to save probing when you do it

so G29
then M500 to save it

and then after each G28 it will fetch level data from eeprom, you dont have to do G29 before each print

spilz87, 2019-07-26:
Actually I change the glass on the bed for each print and I don’t want to care about which one I use neither which way I put it, so I prefer to spend some time with an automatic leveling than save several mesh and select the good one for each print

boelle, 2019-07-26:
then its best to have //#define RESTORE_LEVELING_AFTER_G28 you dont want to restore anything

then do the G28 first and G29 last

spilz87, 2019-07-26:
I just compile it after comment the line
I test the last gcode exemple 

Same issue :(

But I think the issue is more about M420

Without M420 lines run the gcode 2 times give the same result
With M420 lines the result is different

boelle, 2019-07-26:
you are doing a G29 before each print, then M420 is not needed

boelle, 2019-07-26:
but that is just me, you might have another opinion and that is fine.

spilz87, 2019-07-26:
I need M420 Z1 to define the gradually reduce 

Or I miss something ?

spilz87, 2019-07-26:
But I still think it’s not normal and it’s a bug somewhere

lrpirlet, 2019-07-28:
Humm
Not too sure here… (I am NOT really a programmer)

It looks to me that an ``` #endif ``` is missing in the M420 code at around line 195... It is very visible when using code contraction in VSCode.

A way to verify would be issue M420 S1 V … This would enable compensation and the argument V should print the mesh, in case there is really a missing endif, it would print the "Invalid mesh." message.

```C++   
 if (leveling_is_valid()) {

      // Subtract the given value or the mean from all mesh values
      if (parser.seen('C')) {
        const float cval = parser.value_float();
        #if ENABLED(AUTO_BED_LEVELING_UBL)

          set_bed_leveling_enabled(false);
          ubl.adjust_mesh_to_mean(true, cval);

        #else

          #if ENABLED(M420_C_USE_MEAN)

            // Get the sum and average of all mesh values
            float mesh_sum = 0;
            for (uint8_t x = GRID_MAX_POINTS_X; x--;)
              for (uint8_t y = GRID_MAX_POINTS_Y; y--;)
                mesh_sum += Z_VALUES(x, y);
            const float zmean = mesh_sum / float(GRID_MAX_POINTS);

          #else

            // Find the low and high mesh values
            float lo_val = 100, hi_val = -100;
            for (uint8_t x = GRID_MAX_POINTS_X; x--;)
              for (uint8_t y = GRID_MAX_POINTS_Y; y--;) {
                const float z = Z_VALUES(x, y);
                NOMORE(lo_val, z);
                NOLESS(hi_val, z);
              }
            // Take the mean of the lowest and highest
            const float zmean = (lo_val + hi_val) / 2.0 + cval;

          #endif

          // If not very close to 0, adjust the mesh
          if (!NEAR_ZERO(zmean)) {
            set_bed_leveling_enabled(false);
            // Subtract the mean from all values
            for (uint8_t x = GRID_MAX_POINTS_X; x--;)
              for (uint8_t y = GRID_MAX_POINTS_Y; y--;) {
                Z_VALUES(x, y) -= zmean;
                #if ENABLED(EXTENSIBLE_UI)
                  ExtUI::onMeshUpdate(x, y, Z_VALUES(x, y));
                #endif
              }
            #if ENABLED(ABL_BILINEAR_SUBDIVISION)
              bed_level_virt_interpolate();
            #endif
          }

        #endif
      }
```

**missing endif here??**

```C++
    }
    else if (to_enable || seenV) {
      SERIAL_ECHO_MSG("Invalid mesh.");
      goto EXIT_M420;
    }
```

lrpirlet, 2019-07-29:
Forget about previous intervention… I just goofed…
I mixed conditional for pre-processor and conditional inside the C++ code

spilz87, 2019-08-01:
ok I think I found an explaination about where is the problem : 
it is when : 
- bed leveling is activated
- fade Z height is activated (fade Z>0) 
- if printer do move at Z > fade Z (generally in start Gcode)

explaination : 
in my start gcode after G28 and G29 the hotend is at Z18 and I move to X100 Y100 then it goes down to Z0, but during the XY move, as fade Z is lower than 18mm (here it's 1mm), there is no compensation. so when it goes down to Z0 the hotend is not on the bed.

if I do this : 
fade Z = 1mm
after G28 and G29, hotend is at Z18, it goes down to Z0.5, then it moves to X100 Y100, with compensation during this move, then it goes down to Z0 -> the hotend is on the bed.

I think the issue is due to fade Z and Moving "too much" on Z, and it's only seen at start as during print we don't go back to lower Z 

I'm not sure to be clare 
let me know if you Don't understand me ^^