MarlinFirmware/Marlin #9013: No Init. SD in menu

R3play210, 2018-01-02:
Heya!

So after upgrading to RAMPS1.4 and Marlin 1.1.8 I noticed that I don't have the option to Initialize the SD card anymore.

Is this now something you need to enable somewhere?

Ordenkeim93499, 2018-01-02:
Yes in the Configuration.h approximately in line 1461 `//#define SDSUPPORT` you have to uncommend it by deleting the two` //` if you mean this.

R3play210, 2018-01-02:
No.
I miss the function to initialize the SD card

Tannoo, 2018-01-03:
It looks as though only if the card is not readable.

```cpp
   #if ENABLED(SDSUPPORT)
      if (card.cardOK) {
        if (card.isFileOpen()) {
          if (card.sdprinting)
            MENU_ITEM(function, MSG_PAUSE_PRINT, lcd_sdcard_pause);
          else
            MENU_ITEM(function, MSG_RESUME_PRINT, lcd_sdcard_resume);
          MENU_ITEM(function, MSG_STOP_PRINT, lcd_sdcard_stop);
        }
        else {
          MENU_ITEM(submenu, MSG_CARD_MENU, lcd_sdcard_menu);
          #if !PIN_EXISTS(SD_DETECT)
            MENU_ITEM(gcode, MSG_CNG_SDCARD, PSTR("M21"));  // SD-card changed by user
          #endif
        }
      }
      else {
        MENU_ITEM(submenu, MSG_NO_CARD, lcd_sdcard_menu);
        #if !PIN_EXISTS(SD_DETECT)
          MENU_ITEM(gcode, MSG_INIT_SDCARD, PSTR("M21")); // Manually initialize the SD-card via user interface
        #endif
      }
    #endif // SDSUPPORT
```

tcm0116, 2018-01-03:
>It looks as though only if the card is not readable

It's not that it's not readable, it's if the presence of the SD card can't be detected. If the control board has a `SD_DETECT` pin, then the SD card will be automatically initialized when it is inserted. Without a `SD_DETECT` pin, Marlin has no way of knowing that a card was installed, so the user has to manually initialize the SD card.

thinkyhead, 2018-02-11:
Presuming this is resolved.

closed 2018-02-11 by thinkyhead