MarlinFirmware/Marlin #9016: filament change/runout routine triggers minutes after the sensor is triggered during print

v1ru3, 2018-01-02:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: if during a print i trigger the sensor for filament run out the printer pauses for filament change minutes after the event.
- Expected behaviour: to trigger immediately after the sensor triggers.
- Actual behaviour: long delay between the sensor being triggered and the printer to prompt for filament change.
- Steps to reproduce:
start printing and trigger the sensor.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1598743/Marlin.zip)

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

AnHardt, 2018-01-02:
Likely the test is at the wrong end of the queue.

v1ru3, 2018-01-02:
seem to be

v1ru3, 2018-01-02:
hope is an easy fix.

revilor, 2018-01-07:
As there is no FIL_RUNOUT_PIN defined in your configuration: how is the sensor connected to the board?

thinkyhead, 2018-01-07:
> As there is no FIL_RUNOUT_PIN defined in your configuration: how is the sensor connected to the board?

@v1ru3 — I assume it's defined in your `pins_MELZI_CREALITY.h` file, yes?

thinkyhead, 2018-01-07:
The filament runout event immediately adds an `M600` to the end of the command buffer. Since the `M600` command includes movement, it must wait for the planner buffer to empty before it starts. The planner buffer may contain up to 16 linear movements. On a Cartesian these may be long movements, but on Delta (or when using `SEGMENT_LEVELED_MOVES`) they should complete pretty quickly.

If the runout sensor isn't triggering right away, it could be due to a floating pin. Are you using an NC switch with your sensor?

v1ru3, 2018-01-07:
pins_MELZI_CREALITY.h is where the pin is defined, yes.
I know it adds M600 at the end of the queue and it's not an immediate trigger. But the way it behaves now, it triggers the pause after a very long time... more than 45 seconds.

No NC switch. The filament run-out switch is grounded on the beeper pin. I'll try the ENDSTOPPULLUP_FIL_RUNOUT, but it should notmake a difference in my case. Plus, it use to work as it was supposed to in 1.1.7...

AnHardt, 2018-01-07:
Not that queue.
The one where the measurements are put in at the one side and used on the other.
So not 45s later but MEASUREMENT_DELAY_CM of filament later.

AnHardt, 2018-01-07:
Sorry. For some reason i assumed a relation to #8617.

v1ru3, 2018-01-07:
changin ENDSTOPPULLUP_FIL_RUNOUT, made no difference.

v1ru3, 2018-01-07:
M119 reports correct operation for the switch, by the way.

v1ru3, 2018-01-17:
Any updates on this one?

thinkyhead, 2018-01-22:
Unable to reproduce. My runout sensor works reliably.

Can you test with some simple small object like a hollow cube (or just pull the filament out and insert it loosely into the runout sensor during a test print, then pull it out) and see what goes on? The filament runout sensor should trigger a response as soon as the switch is seen to be triggered.

At the start of testing send `M111 S3` to get extra logging, so we can see if `M600` is being enqueued in the host output log.

v1ru3, 2018-01-26:
you Sir are correct. Turns out my sensor is bad.

closed 2018-01-26 by v1ru3

thinkyhead, 2018-01-26:
I keep telling 'em, give us sensors that sense! Sensors that sense are sensible!

gordo3di, 2019-01-04:
@thinkyhead 

> The filament runout event immediately adds an `M600` to the end of the command buffer. Since the `M600` command includes movement, it must wait for the planner buffer to empty before it starts. The planner buffer may contain up to 16 linear movements. On a Cartesian these may be long movements, but on Delta (or when using `SEGMENT_LEVELED_MOVES`) they should complete pretty quickly.
> 
> If the runout sensor isn't triggering right away, it could be due to a floating pin. Are you using an NC switch with your sensor? Try changing your `ENDSTOPPULLUP_FIL_RUNOUT` setting and see if it helps.

I just ran a test print of a 300mm box and I triggered the filament runout sensor. You are correct, the printer had to run through 16 commands in the buffer. I noticed you mentioned about SEGMENT_LEVELED_MOVES for delta. I have it turned on for Cartesian but it doesn't seem to add more movements to the buffer (even though it seems to work properly). The printer still completed the long movements (ie. 300mm) as a single command in the buffer. Below are essentially the moves that finished from when I triggered the sensor until it actually paused.

```
echo:N20 G1 X59.231 Y350.769 E49.6003*106
echo:N21 G1 X59.231 Y59.581 E66.1139*87
echo:N22 G1 X59.231 Y59.231 F1872*115
echo:N23 G92 E0.0000*88
echo:N24 G1 E-0.7500 F4200*42
echo:N25 M117 Layer 2, Z=0.450*38
echo:N26 G1 Z0.450 F3000*44
echo:N27 G1 X59.175 Y59.175 F7800*117
echo:N28 G1 E0.0000 F4200*9
echo:N29 G92 E0.0000*82
echo:N30 G1 X350.825 Y59.175 E10.5855 F3120*13
echo:N31 G1 X350.825 Y350.825 E21.1710*84
echo:N32 G1 X59.175 Y350.825 E31.7564*103
echo:N33 G1 X59.175 Y59.525 E42.3292*95
echo:N34 G1 X59.175 Y59.175 F3120*120
echo:M105
```

If the segments are added to the buffer the filament runout would occur much faster for something like a box and theoretically if we have the segments set to 5mm then the max it could go is 80mm of movement.

We're trying to install our new filament runout sensor inside the extruder but unfortunately this means we need to have immediate or near immediate runout detection. 

Not really sure where I was going with this but what are your thoughts on saving the buffered commands so the detection is immediate? Is it possible to inject a command into the buffer on the fly? Is this a bigger task to tackle?

Thanks

AnHardt, 2019-01-05:
> what are your thoughts on saving the buffered commands so the detection is immediate? Is it possible to inject a command into the buffer on the fly? Is this a bigger task to tackle?

The detection is already close to imediat - it's the reaction on the detection what is delayed.
It's impossible to inject a movement command 'on the fly'. It needs a good amount of preparation to do so.
It's not impossible to improve the reaction time, but it's a job for the 'big boys'.

Currently we have two ways to stop a print.
* Stop planing new planner buffer lines and wait until the buffer is empty. This is what is currently used with filament runout and can last for a while.
* Stop planing new planer buffer lines, stop stepping immediately and delete the planer buffer content. This is used as a emergency stop and with power failure detection. This is not recoverable without homing because the sudden stop may lose steps because it is not deaccelerated. The content of the buffer is lost forever. (In the example above the layer change would be lost!)

What Marlin is missing is a coordinated deaccelerated fast stop. A concept for that could be:
* When detecting 'filament runout' or 'power failure' or any other situation we want to stop fast (endstop ?) - set a flag as fast as possible (maybe in a interrupt).
* If the flag is set
** in the stepper IRQ go into deacceleration mode. In the planner buffer we already have all needed information. We have the maximum allowed deacceleration for every line and we have a point from where in the line we have to deaccelerate. If the complete line as accelerating or at target speed this point is the end of that line. However for the complete buffer content its warranted the final speed is zero.  If we ignore the precalculated point from where on to deaccelerate but do it when the 'flag' is set we can reach the stop much faster (at least in the cases when the buffer is full of long lasting lines). 
We can stop stepping when the speed is at or below jerk-speed, or at zero.
** Meanwhile the planner stopped accepting new moves, finalised planing the last accepted move and optimized the buffer for highest possible speed (as always). So planner buffer state is consistent. The stepper ISR did not rewrite the planner buffer, just ignored the instructions from where to deaccelerate.
We can now save the machine state outside the planer/stepper.

* Now, when the planer does not plan anymore, the speed is zero and no further steps are done, we can save the stepperISR state (position and where in the last line we stopped) and ether save the entire planer buffer, reset it and use it for injecting new moves (for now to bring the nozzle away from the part), or if we just want to inject some lines we can simply use an other (maybe much shorter) line-buffer. Now reset the flag and resume stepping.

* Now we can do the filament change or whatever

* When we want to resume
** at first we have to restore the machine physical state (position, nozzles priming state, temperatures, etc)
** than the machines outer logical state
** than the blanner buffer state. For that we stop the stepper IRQ again and copy in the saved contend (or switch to the original buffer). Now we have to **replan the buffer content**. Parts of the first move are already done and current speed is zero. Most of the work is already done, we just have to shift the points for end of acceleration and begin of deaccelerations, what usually the 'optimizer' does. Now we can enable stepping an accepting new moves from the original queue. 
Done.

Remark:
Deaccelerated coordinated fast break could have less 'break-way' than just stopping to step. When steps are skipped the movement is accelerated again on the 'backside' of the 'dents'. Moving 'the field' 'right' means keeping the 'magnet' on the frontside of the 'dent'.