MarlinFirmware/Marlin #9020: Inductive Probe bug report

rjcuomo1, 2018-01-03:
[configuration and adv.zip](https://github.com/MarlinFirmware/Marlin/files/1599472/configuration.and.adv.zip)

So, if I home through the smart controller display in the "prepare" menu, everything works normally and correctly. Everything homes then I can bed run the "bed leveling" and that works normally as well. The problem comes in when I try to print. X and Y home fine (they use optical end stops), then Z does it's first bump, the sensor triggers then the plate moves downwards. The LED on the probe stays turned on and plate doesn't return for the second homing bump. Then the printer just continues on with the printing. The probe LED indicates triggered even though it's 10 mm from the bed though the z-axis says it's 0 on the display. This happens when I try to print by SD or USB. Once the probe gets locked up like that, it won't even home through the prepare menu anymore. I have to reset the printer or power cycle to get it to home correctly again through the menu. Has anyone else had this issue?

Version: Marlin 1.1.7
Printer: Homemade Corexy printer
Probe: 61mm-M12-DC-5V-NPN-NO-LJ12A3-4-Z-BX-5V-300mA-Inductive-Proximity-Sensor-Switch
Code:
//===========================================================================
//============================== Endstop Settings ===========================
//===========================================================================

// @section homing

// Specify here all the endstop connectors that are connected to any endstop or probe.
// Almost all printers will be using one per axis. Probes will use one or more of the
// extra connectors. Leave undefined any used for non-endstop and non-probe purposes.
#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG
//#define USE_XMAX_PLUG
//#define USE_YMAX_PLUG
//#define USE_ZMAX_PLUG

// coarse Endstop Settings
//#define ENDSTOPPULLUPS // Comment this out (using // at the start of the line) to disable the endstop pullup resistors

#if DISABLED(ENDSTOPPULLUPS)
// fine endstop settings: Individual pullups. will be ignored if ENDSTOPPULLUPS is defined
//#define ENDSTOPPULLUP_XMAX
//#define ENDSTOPPULLUP_YMAX
//#define ENDSTOPPULLUP_ZMAX
//#define ENDSTOPPULLUP_XMIN
//#define ENDSTOPPULLUP_YMIN
//#define ENDSTOPPULLUP_ZMIN
//#define ENDSTOPPULLUP_ZMIN_PROBE
#endif

// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
#define X_MIN_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define X_MAX_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define Z_MIN_PROBE_ENDSTOP_INVERTING true // set to true to invert the logic of the probe.

// Enable this feature if all enabled endstop pins are interrupt-capable.
// This will remove the need to poll the interrupt pins, saving many CPU cycles.
//#define ENDSTOP_INTERRUPTS_FEATURE

rjcuomo1, 2018-01-13:
Anyone have any thoughts?

Bob-the-Kuhn, 2018-01-13:
The probe stays in the triggered state.

My first reaction is that it's a hardware issue.  
- Power?  If you can, measure the voltage at the sensor when it's locked up.  If it's under 4.5V try running heavier power and ground wires to the sensor.
- Wiring problem?  Does anything like a heater or a fan turn on just before it locks up?

How repeatable is this?  Does it ever happen when homing manually?  Does it happen when leveling the bed?  Does it happen if you just mannually home and then start the print?

rjcuomo1, 2018-01-13:
Repeatable like clockwork. Manual leveling and homing works perfectly through the firmware and through commands via USB. It only has this problem when I start a print (through SD, usb directly from Cura, and through Octoprint) and it doesn't home correctly, it just locks up.

The only issue at this point is the sensor, all the other inputs and outputs work flawlessly manually or printing. I'll test the power going to the sensor and paste in the gcode when I get a chance.

rjcuomo1, 2018-01-14:
So the voltage is 4.87v at the connector to the board. 

and here is the beginning GCODE
;FLAVOR:Marlin
;TIME:1217
;Filament used: 1.36219m
;Layer height: 0.2
;Generated with Cura_SteamEngine 3.1.0
M190 S60
M104 S200
M109 S200
M82 ; absolute extrusion mode
G28 ;Home
G1 Z15.0 F6000 ;Move the platform down 15mm
;Prime the extruder
G92 E0
G1 F200 E3
G92 E0
;LAYER_COUNT:101
;LAYER:0
M106 S255
M204 S3000
M205 X20 Y20
G0 F4500 X165.157 Y165.478 Z0.2
;TYPE:SKIRT
G1 F1800 X165.552 Y165.094 E0.01832
G1 X165.998 Y164.77 E0.03666

rjcuomo1, 2018-01-14:
The probe has a 10k resistor built-in between VCC and signal

revilor, 2018-01-15:
Sometimes the heated bed interferes with inductive probes. You could try

#define PROBING_HEATERS_OFF

to see if this helps.

rjcuomo1, 2018-01-17:
That didn't have any effect :/

This is getting frustrating! But thank you guys for your help

revilor, 2018-01-17:
I can give you one last list of things you could try (if you haven't already) before I'm out of options:

- increase Z_HOME_BUMP_MM to 5 or 10, maybe the 2mm are just the border line where the probe starts to trigger
- try G28 as the very first command in your gcode, maybe even home each axis separately
- if nothing else helps try bugfix-1.1.x for the version with all the latest bug fixes

rjcuomo1, 2018-01-18:
Here's the bed leveling information, perhaps I may have missed something:

//===========================================================================
//=============================== Bed Leveling ==============================
//===========================================================================
// @section calibrate

/**
 * Choose one of the options below to enable G29 Bed Leveling. The parameters
 * and behavior of G29 will change depending on your selection.
 *
 *  If using a Probe for Z Homing, enable Z_SAFE_HOMING also!
 *
 * - AUTO_BED_LEVELING_3POINT
 *   Probe 3 arbitrary points on the bed (that aren't collinear)
 *   You specify the XY coordinates of all 3 points.
 *   The result is a single tilted plane. Best for a flat bed.
 *
 * - AUTO_BED_LEVELING_LINEAR
 *   Probe several points in a grid.
 *   You specify the rectangle and the density of sample points.
 *   The result is a single tilted plane. Best for a flat bed.
 *
 * - AUTO_BED_LEVELING_BILINEAR
 *   Probe several points in a grid.
 *   You specify the rectangle and the density of sample points.
 *   The result is a mesh, best for large or uneven beds.
 *
 * - AUTO_BED_LEVELING_UBL (Unified Bed Leveling)
 *   A comprehensive bed leveling system combining the features and benefits
 *   of other systems. UBL also includes integrated Mesh Generation, Mesh
 *   Validation and Mesh Editing systems.
 *
 * - MESH_BED_LEVELING
 *   Probe a grid manually
 *   The result is a mesh, suitable for large or uneven beds. (See BILINEAR.)
 *   For machines without a probe, Mesh Bed Leveling provides a method to perform
 *   leveling in steps so you can manually adjust the Z height at each grid-point.
 *   With an LCD controller the process is guided step-by-step.
 */
//#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
#define AUTO_BED_LEVELING_BILINEAR
//#define AUTO_BED_LEVELING_UBL
//#define MESH_BED_LEVELING

/**
 * Enable detailed logging of G28, G29, M48, etc.
 * Turn on with the command 'M111 S32'.
 * NOTE: Requires a lot of PROGMEM!
 */
//#define DEBUG_LEVELING_FEATURE

#if ENABLED(MESH_BED_LEVELING) || ENABLED(AUTO_BED_LEVELING_BILINEAR) || ENABLED(AUTO_BED_LEVELING_UBL)
  // Gradually reduce leveling correction until a set height is reached,
  // at which point movement will be level to the machine's XY plane.
  // The height can be set with M420 Z<height>
  #define ENABLE_LEVELING_FADE_HEIGHT

  // For Cartesian machines, instead of dividing moves on mesh boundaries,
  // split up moves into short segments like a Delta. This follows the
  // contours of the bed more closely than edge-to-edge straight moves.
  #define SEGMENT_LEVELED_MOVES
  #define LEVELED_SEGMENT_LENGTH 5.0 // (mm) Length of all segments (except the last one)

  /**
   * Enable the G26 Mesh Validation Pattern tool.
   */
  #define G26_MESH_VALIDATION   // Enable G26 mesh validation
  #if ENABLED(G26_MESH_VALIDATION)
[... 173 lines cut ...]

#if ENABLED(SKEW_CORRECTION)
  // Input all length measurements here:
  #define XY_DIAG_AC 282.8427124746
  #define XY_DIAG_BD 282.8427124746
  #define XY_SIDE_AD 200

  // Or, set the default skew factors directly here
  // to override the above measurements:
  #define XY_SKEW_FACTOR 0.0

  //#define SKEW_CORRECTION_FOR_Z
  #if ENABLED(SKEW_CORRECTION_FOR_Z)
    #define XZ_DIAG_AC 282.8427124746
    #define XZ_DIAG_BD 282.8427124746
    #define YZ_DIAG_AC 282.8427124746
    #define YZ_DIAG_BD 282.8427124746
    #define YZ_SIDE_AD 200
    #define XZ_SKEW_FACTOR 0.0
    #define YZ_SKEW_FACTOR 0.0
  #endif

  // Enable this option for M852 to set skew at runtime
  //#define SKEW_CORRECTION_GCODE
#endif

myoung008, 2018-01-18:
I suggest wiring.  Inductive probes don't tend to work well on 5v, so use 12v for power.  There are lots of suggestions about how to wire them with resistor bridges on the signal line, but those gave me inconsistent results.

The wiring method that works for me is a simple diode on the signal pin, cathode toward the board, anode toward the sensor.  The odd part is that the sensor LED appears triggered at all times, maybe 80% brightness, then goes to full brightness when actually triggered.  Marlin can see the correct trigger state regardless of the LED however.

M48 repeatablilty is essentially perfect with this wiring method.

rjcuomo1, 2018-01-18:
My next step was probe replacement since it seems like I'm running out of steps, lol.

benlye, 2018-01-18:
If you're thinking about replacing the probe anyway, you could get one which is designed for 5V, and avoid the wiring complications of a 6-36V probe altogether.

I have this one and it works very well. 
https://m.aliexpress.com/s/item/32719316657.html?trace=wwwdetail2mobilesitedetail&productId=32719316657&productSubject=M18-8mm-sensing-DC-5V-NPN-NO-LJ18A3-8-Z-BX-5V-cylinder-inductive-proximity-sensor

Bob-the-Kuhn, 2018-01-20:
Here's the NC version.  It was just added tonight.

https://www.aliexpress.com/store/product/M18-8mm-sensing-DC-5V-NPN-NC-LJ18A3-8-Z-AX-5V-cylinder-inductive-proximity-sensor/1162433_32851947449.html

rjcuomo1, 2018-01-21:
I have the one that works on 5v currently. It's a NPN and NO

rjcuomo1, 2018-02-05:
Thank you all again for your help! The problem was the 5v inductive probe. I went with a 12v with a voltage divider and it works.

closed 2018-02-05 by rjcuomo1