MarlinFirmware/Marlin #9021: http://marlinfw.org/tools/lin_advance/k-factor.html please support origin in center of bed

Lenbok, 2018-01-03:
The web based g-code generator for configuring LIN_ADVANCE looks awesome, but my printer is configured so that the center of the bed is 0,0 and the current generator therefore puts all the test lines off one corner of the bed.

Can we please have an option for the pattern to be centered on 0,0 rather than width/2,height/2?

Sineos, 2018-01-03:
Please provide me with your settings and I will provide you with a gcode file to test.

Lenbok, 2018-01-03:
Here are the settings I would like to try:

````
; ### Marlin K-Factor Calibration Pattern ###
; -------------------------------------------
;
; Created: Thu Jan 04 2018 08:26:41 GMT+1300 (NZDT)
; Settings:
; Print Size X = 90 mm
; Print Size Y = 125 mm
; Print Rotation = 0 degree
; Print Pattern = Standard
; Print Frame = true
; Prime Nozzle = true
; Prime Extrusion Multiplier = 2.5
; Dwell Time = 2 s
; Filament Diameter = 2.86 mm
; Nozzle Diameter = 0.4 mm
; Nozzle Temperature = 197 °C
; Nozzle / Line Ratio = 1.2
; Bed Temperature = 85 °C
; Slow Printing Speed = 1200 mm/min
; Fast Printing Speed = 4200 mm/min
; Movement Speed = 7200 mm/min
; Use BL = false
; Retraction Distance = 2 mm
; Bed Size X = 200 mm
; Bed Size Y = 200 mm
; Layer Height = 0.2 mm
; Extrusion Multiplier = 1
; Starting Value K-Factor = 0
; Ending value K-Factor = 100
; K-Factor Stepping = 5
; Test Line Spacing = 5 mm
; Test Line Length Slow = 20 mm
; Test Line Length Fast = 40 mm
````

Sineos, 2018-01-03:
Please try attached file:
[kfactor_center.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/1601854/kfactor_center.gcode.txt)

Lenbok, 2018-01-04:
I haven't run it on the printer yet, as I noticed something bad in a gcode preview... The following line goes outside of my Y range of [-100, 100]:
````
G1 Z30 Y200 F7200 ; move away from the print
````

(And a minor note, in the settings section at the top is the new line `; Home Center = true` -- this current text may just be a placeholder, but it would be better to word this as "Origin Center" rather than "Home Center", since it is the origin we are talking about, not the home position.)

Sineos, 2018-01-04:
> I haven't run it on the printer yet, as I noticed something bad in a gcode preview... The following line goes outside of my Y range of [-100, 100]:
True, missed to adapt the final move to the new origin. New try:
[kfactor.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/1604280/kfactor.gcode.txt)

Lenbok, 2018-01-04:
Thanks, that worked well. The only problem was that since my mendel90 homes at XMAX,YMAX,ZMAX, and I have clips at the corners of my bed, I had to remove one of the clips to prevent a head crash since the homing is followed by Z-only move to 0.2 before any X and Y movement. Would it be safer to do that first move as a combined move to the initial X, Y, and Z?

Sineos, 2018-01-05:
https://github.com/MarlinFirmware/MarlinDocumentation/pull/124

Will be available when @thinkyhead finds the time to merge and publish it. Also addresses your valid input on the first move after homing.

closed 2018-02-11 by thinkyhead