MarlinFirmware/Marlin #9022: 1.1.8 Z-offset with M851 but not M500 automaticly (babystepping)

JupaCreations, 2018-01-03:
Feature Request? Bug?

Z-offset is at -1.0 mm
When babystepping is finished the new value is stored in the RAM for example -1.2 mm.
Without power recycle of Marlin all works fine.
When power is recycled the old value of -1.0 mm is restored from EEprom.
Is this specs?
As it would be nice when exiting babystepping/confirming new Z-offset value, the value is stored with M500 immediatly.

Thanks in advance,

Paco Raap

tcm0116, 2018-01-03:
No changes are stored in Marlin automatically. If you adjust a value, you'll need to run M500 to save the change to the EEPROM.

JupaCreations, 2018-01-04:
tcm1106 thanks for the answer.
It would obvious the next step is to integrate the M500 if that is possible or is this code wise impossible?
If it can where coud I add the code for a custom version?

thanks Paco Raap

tcm0116, 2018-01-04:
Saving the EEPROM is a very expensive operation which will disrupt
printing, so it's not advised to use M500 during printing. If you really
want to save your settings automatically, you could put an M500 at the end
of your end GCode script in your slicer. This would cause any changes
you've made during a print to be saved when the print completes.

JupaCreations, 2018-01-04:
Thanks for the answer.
I will instruct users to add M500 to the End - Gcode.

Thanks, Paco Raap

closed 2018-01-04 by JupaCreations