MarlinFirmware/Marlin #9023: [2.0.x] Compile for Due stops in usb_task.c

byebyedizzy, 2018-01-03:
### Compile Error
When I compile with for Arduino Due I get an error:
------------
sketch\src\HAL\HAL_DUE\usb\usb_task.c: In function 'HAL_init':

sketch\src\HAL\HAL_DUE\usb\usb_task.c:275:7: warning: implicit declaration of function 'sprintf_P' [-Wimplicit-function-declaration]

       sprintf_P(diam, PSTR("%d"), (int)((DEFAULT_NOMINAL_FILAMENT_DIA) * 1000.0));

       ^

sketch\src\HAL\HAL_DUE\usb\usb_task.c:275:7: warning: implicit declaration of function 'PSTR' [-Wimplicit-function-declaration]

sketch\src\HAL\HAL_DUE\usb\usb_task.c.o: In function `HAL_init':

sketch\src\HAL\HAL_DUE\usb/usb_task.c:275: undefined reference to `PSTR'

sketch\src\HAL\HAL_DUE\usb/usb_task.c:275: undefined reference to `sprintf_P'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino Due (Programming Port).
------------

Adding line:
#include <avr/pgmspace.h>
to Marlin\src\hal\HAL_DUE\usb\usb_task.h helps to resolve those errors. Though I have to define SDSUPPORT to get rid of bunch of other errors (I will open another issue for that).

Bob-the-Kuhn, 2018-01-03:
PR #9028 also addresses this.  

I used **Arduino.h** rather than **avr/pgmspace.h**.  I have no idea if there are advantages of using one vs. the other.

Bob-the-Kuhn, 2018-01-05:
Can we close this issue?

closed 2018-02-11 by thinkyhead