MarlinFirmware/Marlin #9024: [2.0.x] Compile for Due stops in sd_mmc_spi_mem.cpp

byebyedizzy, 2018-01-03:
### Compile Error
When I compile without `SDSUPPORT` I get an error:
```
sketch\src\HAL\HAL_DUE\usb\sd_mmc_spi_mem.cpp.o: In function `sd_mmc_spi_test_unit_ready':

sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:24: undefined reference to `card'

sketch\src\HAL\HAL_DUE\usb\sd_mmc_spi_mem.cpp.o: In function `sd_mmc_spi_read_capacity':

sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:30: undefined reference to `Sd2Card::cardSize()'

sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:32: undefined reference to `card'

sketch\src\HAL\HAL_DUE\usb\sd_mmc_spi_mem.cpp.o: In function `sd_mmc_spi_removal':

sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:49: undefined reference to `card'

sketch\src\HAL\HAL_DUE\usb\sd_mmc_spi_mem.cpp.o: In function `sd_mmc_spi_usb_read_10':

sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:76: undefined reference to `Sd2Card::readStart(unsigned long)'

sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:83: undefined reference to `Sd2Card::readData(unsigned char*)'

sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:87: undefined reference to `Sd2Card::readStop()'

sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:93: undefined reference to `Sd2Card::readStop()'

sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:77: undefined reference to `card'

sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:77: undefined reference to `card'

sketch\src\HAL\HAL_DUE\usb\sd_mmc_spi_mem.cpp.o: In function `sd_mmc_spi_usb_write_10':

sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:110: undefined reference to `Sd2Card::writeStart(unsigned long, unsigned long)'

sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:118: undefined reference to `Sd2Card::writeStop()'

sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:123: undefined reference to `Sd2Card::writeData(unsigned char const*)'

sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:127: undefined reference to `Sd2Card::writeStop()'

sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:111: undefined reference to `card'

sketch\src\HAL\HAL_DUE\usb/sd_mmc_spi_mem.cpp:111: undefined reference to `card'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino Due (Programming Port).```

------------------------------------------------------------
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1599938/Configuration.zip)

PS: These errors come after resolving defines in usb_task.c (https://github.com/MarlinFirmware/Marlin/issues/9023)

Bob-the-Kuhn, 2018-01-03:
PR #9028 will fix this.

Bob-the-Kuhn, 2018-01-05:
Can we close this issue?

closed 2018-02-11 by thinkyhead