MarlinFirmware/Marlin #9026: Babystepping is not working at all + no double click to enter babystepping marlin 1.1.8

Anxles, 2018-01-03:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Bug Report
- Description: --- babystepping is not working at all. I use 2 stepper motors for Z axis, maybe this is a reason. I'm using Reprap discount Smart controller on custom build printer. Megatronics V2 motherboard.
[Uploading Configuratio

ns.rar…]()

- Expected behaviour: --- rotate the knob of babystepping during 1st layer printing should change Z position on the fly
- Actual behaviour: ---rotation of the knob (values are changing on the screen) does not actually rotate the stepper motors of Z axis. Completely no reaction, additionaly I double click the button but its not going directly do babystepping.

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

Tannoo, 2018-01-03:
Try uncomme ting this:

`  //#define DOUBLECLICK_FOR_Z_BABYSTEPPING` in `Configuration_adv.h`.

th33xitus, 2018-01-03:
I have encountered the same "issue".
You can either try and set the `DOUBLECLICK_MAX_INTERVAL 1250` to a higher value.
For me the solution is simple... i get into the babystep-ui with just keeping the knob pushed down. Even with doubleclick enabled.

Anxles, 2018-01-03:
Getting to the menu by double click is a "secondary" problem as long as baby stepping is not working at all :)

tcm0116, 2018-01-03:
It sounds like you're trying to babystep the Z axis by adjusting the Z offset via the LCD. For this to work, the printer must have been homed (G28) and leveled (G29). In the current bugfix-1.1.x branch, we've removed the requirement for the printer to have been leveled (G29).

Anxles, 2018-01-03:
I made some test and the results are: the auto bed leveling (g29) has no effect of working or not wotking the baby stepping in either activated/ disactivated feature. When Autobed leveling is enabled, performing the G29 before the print and not performin also has no effect. I found when the option "Disable Z during standstill" is enabled the babystepping IS WORKING. Unfortunately i need to have Z disable turned off because I have the Z axis driven by belt system for not important reasons right now, and if I keep motors enabled during standstill they are getting so hot that my printed brackets for the motors are getting soft and twist making z axis unusable :)

Tannoo, 2018-01-03:
Is the the current set too high for your Z steppers?

Anxles, 2018-01-03:
Yes I have set high current on my steppers on purpose. My Z axis table weights abt. 10kg and i need a lot of torque to move it up and down. Its not a problem for moving the Z axis for standard periods of time while printing or homing or even moving full length up and down at a time. But during powering while in standstill they R getting really hot.

tcm0116, 2018-01-03:
>I found when the option "Disable Z during standstill" is enabled the babystepping IS WORKING.

I'm not familiar with the "Disable Z during standstill" option, and I assume you meant to say disabled. If so, that would explain why babystepping isn't working since you can't move disabled steppers.

Anxles, 2018-01-04:
I ment "#define DISABLE_Z true"
Before layer change Z motors are also disabled and it doesn't prevent g code from changing the layer to next one. I red on one of the issues on the similar topic from one of the users that when he starts using baby stepping during Z motors being turned off, they behave (i forgot the word used: erratic, violent, or similar) on powering up: conclusion: on one of the previous versions of Marlin babystepping was starting powered off steppers when needed. I'm not a programmer but I assume there should be no problem to activate option like this again (in my case it should be no problem as motors which I use are 30:1 gear ratio and rapid power up should not affect axis position)

croisez, 2018-01-04:
I got the same issue. In fact, to enter the Babystepping panel, I must not double click, but leave the button pressed instead. The problem is that if I leave it too much time pressed, the Babystep value is validated like if I had press return key.

tcm0116, 2018-01-04:
@Anxles Marlin used to apply babysteps to disabled motors. However, this resulted in jerky movement since the motors were enabled and disabled for each babystep. Perhaps a solution would be to put in a delay before the motors are disabled, which would allow the queued babysteps to complete.

Anxles, 2018-01-04:
For me satisfying would be to make delay for turning off motors for time of the first layer. Motors won't get hot enough to make any problems, and after that, from the second layer babystepping is not needed any more. Win-win situation if You ask me.

Roxy-3D, 2018-01-04:
There are valid use cases for using BabyStepping after the first layer.   We don't want to turn that off after the first layer.   (And in fact, it would be difficult to know when the nozzle has finished the first layer.) 

> I got the same issue. In fact, to enter the Babystepping panel, I must not double click, but leave the button pressed during DOUBLECLICK_MAX_INTERVAL instead.

There is a latency issue because the Encoder Wheel is only checked periodically for the double click.   Have you tried adjusting the   #define DOUBLECLICK_MAX_INTERVAL 1250   value?

tcm0116, 2018-01-04:
Unfortunately, that's not possible as Marlin has no concept of the first layer, so it can't know when to turn off the motors. However, your slicer does, so an option could be to insert an `M18 Z` command in your layer change GCode, which will disable the Z axis motors when the layer changes. For this to work, you'd need to set `DISABLE_Z` to *false* in your firmware.

Anxles, 2018-01-04:
Ok the tcm116 comment sounds usefull, one question though the M18 command should be inserted after every layer change during print, or using it once affects whole print and will turn off the motors after each layer automatically ?

tcm0116, 2018-01-05:
@Anxles I'm not sure what slicer you're using, but Slic3r has an "After layer change G-code" script that is run after the Z height is changed. If you put the `M18 Z` command in the "After later change G-code" script, then the printer will disable the Z motors after each layer. For the first layer, the motors will be enabled for the entire layer, but will then be disabled after the Z height changes when starting the second layer. Marlin will then enable the motors when the Z height change is requested at the end of the second layer and then the `M18 Z` command will disable the motors again. I think this should achieve what you're looking to accomplish. 

If you're using a different slicer, you'll need to see where it inserts the layer change G-code relative to the change in Z position for each layer in order to get this method to work.

Anxles, 2018-01-05:
So from my point of view now its not an issue but misunderstanding. Looks like it works but only in certain conditions, the only fix should be an information before babystepping it doesnt work with turned off Z motors in the configuration file. 95% of the time I'm using Simplify 3D, which in the script section contains tab "layer change script" that suppose to resolve my problem.

tcm0116, 2018-01-05:
I agree that we need to at least disallow enabling babystepping for an axis
that is configured to automatically disable. However, putting in some sort
of delay before an axis is disabled could also be a viable solution to
allowing babystepping.

I'm not certain when S3D places the contents of the layer change script
relative to the change in Z height, so you'll want to verify that.

thinkyhead, 2018-02-11:
Marlin now enables steppers if needed for babystepping.

closed 2018-02-11 by thinkyhead

jmdearras, 2019-02-08:
Are babysteps cumulative?  That is, in a print, I try .05, not enough, try .05 again, is it now .1?

tcm0116, 2019-02-09:
@jmdearras - yes