MarlinFirmware/Marlin #9033: LED compile error

rjcuomo1, 2018-01-03:
------------------------------------------------------------
### Compile Error
When I compile with `#define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup` I get an error:

Arduino: 1.8.5 (Windows Store 1.8.10.0) (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch\neopixel.cpp: In function 'void setup_neopixel()':

neopixel.cpp:48: error: 'safe_delay' was not declared in this scope

     safe_delay(1000);

                    ^

exit status 1
'safe_delay' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Tannoo, 2018-01-04:
Well, all fixes are in the bugfix branches.

rjcuomo1, 2018-02-06:
This still happens with 1.1.8 bugfix

Tannoo, 2018-02-07:
There are only two bugfix braches: `bugfix-1.1.x` and `bugfix-2.0.x`.

rjcuomo1, 2018-02-07:
I downloaded the latest bugfix-1.1.x earlier and uploaded and tried it today. Marlin was on version 1.1.6 for the stable release when I originally made this post iirc, but I let this issue fall to the wayside while I was trying to figure out my probing issue. It worked fine on the stable 1.1.6 but it acted up with the bugfix-1.1.x and when 1.1.7 became the stable release, it still acted like this with the bugfix-1.1.x and it's still doing this with whichever bugfix-1.1.x it is as of 6 February, 2018 at 4:00 eastern standard time

Tannoo, 2018-02-07:
Then try bugfix-2.0.x

thinkyhead, 2018-02-17:
Sorry this wasn't patched sooner. The simple solution was to add an include for `utility.h` to `neopixel.cpp`.

closed 2018-02-17 by thinkyhead

pecirep, 2018-03-15:
This did not work for me. After adding #include "utility.h" I get following error:

`'safe_delay' was not declared in this scope`

which is obvious and I would fix if I knew what this "safe delay" is supposed to do.

pecirep, 2018-03-15:
This did not work for me. After adding #include "utility.h" I get following error:

´variable or field 'safe_delay' declared void´

which is obvious and I would fix if I knew what this "safe delay" is supposed to do.

thinkyhead, 2018-03-15:
@pecirep — Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

pecirep, 2018-03-15:
I already took a look at both of them and neither has 'safe_delay' declared

thinkyhead, 2018-03-16:
> neither has 'safe_delay' declared

### `bugfix-1.1.x` > `utility.h`
```cpp
void safe_delay(millis_t ms);
```
### `bugfix-1.1.x` > `utility.cpp`
```cpp
void safe_delay(millis_t ms) {
  while (ms > 50) {
    ms -= 50;
    delay(50);
    thermalManager.manage_heater();
  }
  delay(ms);
  thermalManager.manage_heater(); // This keeps us safe if too many small safe_delay() calls are made
}
```

Xanaeth, 2018-03-27:
Actually, the error in compilation as i found out was due to a missing  **#include "types.h"** in the utility.h  file.   it could not find the type declaration for "**millis_t**"    (1.1.x branch) .  After i added this i could compile without issues.   No idea if this has been resolved yet.

thinkyhead, 2018-03-28:
> No idea if this has been resolved yet.

Yes.

HaxxHaris, 2018-04-05:
I'm getting this error when I activate this line
#define NEOPIXEL_STARTUP_TEST 
on The latest release (1.1.x.zip … 1.1.8)  

Good to know if someone else has a problem, took me a while to find the error.

the post further up "#include" types.h "in the utility.h file" did not help in my case.

Have a nice day

HaxxHaris, 2018-04-09:
@thinkyhead I can confirm that the problem has been solved in the latest bugfix-1.1.x thanks. Regards Haris

GregFax, 2018-05-23:
Read and tried all of the above without success
safe_delay remains undefined
Easiest fix is to disable NEOPIXEL_STARTUP_TEST in Configuration.h (around line 1731 1.1.8)

thinkyhead, 2018-05-24:
@GregFax — The solution for the issue in 1.1.8 is to make the following changes:

### neopixel.cpp
```diff
  #include "neopixel.h"
+ #include "utility.h"
```
### utility.h
```diff
+ #include "types.h"

  void safe_delay(millis_t ms);
```

GregFax, 2018-05-25:
@thinkyhead - Thank you!  
I had the utility.h mod from further up the thread, but not it's link from neopixel.cpp.
All good now, verifies nicely, so I've turned Start_Up_Test back on.

commit fbf88192 (2018-02-17): Make safe_delay available for neopixel startup test

Fix #9033