MarlinFirmware/Marlin #9034: Printer event LEDs use wrong colors and the LEDs colors in the LED menu are incorrect Bug Report

rjcuomo1, 2018-01-04:
The colors sequence incorrectly when heating. According to the code, the sequence should be 

 _- Gradually change from blue to violet as the heated bed gets to target temp
 *  - Gradually change from violet to red as the hotend gets to temperature
 *  - Change to white to illuminate work surface_

But it goes from green to a greenish yellow as the bed gets up to temp, then from a greenish yellow to red 

Under the Lights Presets menu, when I select:
White they're white
Red they're red
Orange turns them pink
Yellow turns them a pinkish-white?
Green turns them blue
Blue turns them green
Indigo turns them cyan
Violet turns them yellow

// Support for Adafruit Neopixel LED driver
#define NEOPIXEL_LED
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE   NEO_GRB // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
  #define NEOPIXEL_PIN    30        // LED driving pin on motherboard 4 => D4 (EXP2-5 on Printrboard) / 30 => PC7 (EXP3-13 on Rumba)
  #define NEOPIXEL_PIXELS 105       // Number of LEDs in the strip
  #define NEOPIXEL_IS_SEQUENTIAL   // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
  #define NEOPIXEL_BRIGHTNESS 127  // Initial brightness (0-255)
  //#define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup
#endif

/**
 * Printer Event LEDs
 *
 * During printing, the LEDs will reflect the printer status:
 *
 *  - Gradually change from blue to violet as the heated bed gets to target temp
 *  - Gradually change from violet to red as the hotend gets to temperature
 *  - Change to white to illuminate work surface
 *  - Change to green once print has finished
 *  - Turn off after the print has finished and the user has pushed a button
 */
#if ENABLED(BLINKM) || ENABLED(RGB_LED) || ENABLED(RGBW_LED) || ENABLED(PCA9632) || ENABLED(NEOPIXEL_LED)
  #define PRINTER_EVENT_LEDS
#endif

Tannoo, 2018-01-04:
Check the orientation of your strip. RGB, GRB, or something different. Correct as needed.

rjcuomo1, 2018-01-04:
Changing it to BRG worked, thank you!

closed 2018-01-04 by rjcuomo1