MarlinFirmware/Marlin #9035: Heating failed, printer halted - Hotend Thermisor pin changed

gabriela19199, 2018-01-04:
Hello, 
 I have a problem with the pin 13. The hotend touched one of the hotbed screws and then the reading change to 191 degrees and never change. I thought that could be a resistance problem so I decided to change and use the pin 15 for the temperature medition.

I changed the pin ramps.h file
#define TEMP_0_PIN         15   // Analog Input
#define TEMP_1_PIN         13   // Analog Input
#define TEMP_BED_PIN       14   // Analog Input

and it seems to work until I try to heat the hotend and I get the following error
Heating failed, printer halted 

I would to appreciate any suggestion

revilor, 2018-01-04:
Maybe the accident pulled the thermistor away from the heater block. Or worse, cut one of it's wires. Check the position of the thermistor in the heater block. If you own a multimeter you could also measure the resistance of the thermistor. Should be close to 100kOhm at room temperature if everything is ok.

gabriela19199, 2018-01-04:
Thanks for your help. First I thought it was the thermistor so I changed it. Then I realized that the temperature was fixed in 191 degrees thats why I though the problem was in the resistance. 

Now I have a new thermisor, I have no problems to read the temperature, the hotend heats up ok, but when it reaches the maximum I get the error "Heating failed, printer halted"

revilor, 2018-01-04:
If everything works as expected, a slight increase of WATCH_TEMP_PERIOD in Configuration_adv.h might help. And if you haven't done that already I would suggest  to PID autotune the extruder.

closed 2018-02-11 by thinkyhead