MarlinFirmware/Marlin #9041: Compile error enabling SERIAL_STATS_MAX_RX_QUEUED/SERIAL_STATS_DROPPED_RX on AT90USB

paccerdk, 2018-01-04:
When I compile Marlin 1.1.8 (release) with SDSUPPORT enabled in configuration.h
and SERIAL_STATS_MAX_RX_QUEUED / SERIAL_STATS_DROPPED_RX in configuration_adv.h I get the following errors:
```
In file included from sketch\Marlin.h:46:0,

                 from sketch\Marlin_main.cpp:247:

sketch\Marlin_main.cpp: In function 'void loop()':

Marlin_main.cpp:14572: error: 'customizedSerial' was not declared in this scope

             SERIAL_ECHOLNPAIR("Dropped bytes: ", customizedSerial.dropped());

                                                  ^

sketch\serial.h:52:76: note: in definition of macro 'SERIAL_PROTOCOLPAIR'

 #define SERIAL_PROTOCOLPAIR(name, value)    (serial_echopair_P(PSTR(name),(value)))

                                                                            ^

sketch\serial.h:61:40: note: in expansion of macro 'SERIAL_PROTOCOLLNPAIR'

 #define SERIAL_ECHOLNPAIR(pre,value)   SERIAL_PROTOCOLLNPAIR(pre, value)

                                        ^

sketch\Marlin_main.cpp:14572:13: note: in expansion of macro 'SERIAL_ECHOLNPAIR'

             SERIAL_ECHOLNPAIR("Dropped bytes: ", customizedSerial.dropped());

             ^

Marlin_main.cpp:14576: error: 'customizedSerial' was not declared in this scope

             SERIAL_ECHOLNPAIR("Max RX Queue Size: ", customizedSerial.rxMaxEnqueued());

                                                      ^

sketch\serial.h:52:76: note: in definition of macro 'SERIAL_PROTOCOLPAIR'

 #define SERIAL_PROTOCOLPAIR(name, value)    (serial_echopair_P(PSTR(name),(value)))

                                                                            ^

sketch\serial.h:61:40: note: in expansion of macro 'SERIAL_PROTOCOLLNPAIR'

 #define SERIAL_ECHOLNPAIR(pre,value)   SERIAL_PROTOCOLLNPAIR(pre, value)

                                        ^

sketch\Marlin_main.cpp:14576:13: note: in expansion of macro 'SERIAL_ECHOLNPAIR'

             SERIAL_ECHOLNPAIR("Max RX Queue Size: ", customizedSerial.rxMaxEnqueued());

             ^

exit status 1
'customizedSerial' was not declared in this scope
```
Compiles fine with the defines commented
I'm guessing there's a couple of missing checks related to "customizedSerial" in Marlin_main.cpp ?

[config.zip](https://github.com/MarlinFirmware/Marlin/files/1602417/config.zip)

fiveangle, 2018-01-10:
Try now with bugfix1.1.x

paccerdk, 2018-01-10:
Looks like i am now getting a proper error during compilation:
```
In file included from sketch\MarlinConfig.h:40:0,

                 from D:\Marlin-bugfix-1.1.x\Marlin\Marlin.ino:31:

SanityCheck.h:279: error: #error "SERIAL_XON_XOFF and SERIAL_STATS_* features not supported on USB-native AVR devices."

   #error "SERIAL_XON_XOFF and SERIAL_STATS_* features not supported on USB-native AVR devices."

    ^

exit status 1
#error "SERIAL_XON_XOFF and SERIAL_STATS_* features not supported on USB-native AVR devices."
```

I guess the problem is solved, thanks!
(i was not aware that the feature is unsupported - with this error message, i would have been)

fiveangle, 2018-01-10:
close then ? :)

paccerdk, 2018-01-10:
Sure :)

PR #9127 (2018-01-10): [2.0.x] SERIAL_STATS_* features not supported on USBCON targets
Fixes #9041
closed 2018-01-10 by paccerdk