MarlinFirmware/Marlin #9046: [FR] Skew correction with no bed-levelling

TheSFReader, 2018-01-04:
Hi, I use marlin with a CNC (in addition of my 3D printer), and would like to use SKEW correction on that one since my XY axes are not square. However, the skew correction doesn't seem to have any effect.

After a fast look at the code, it seems it's only used with bed levelling ( the skew and unskew functions are only called in the apply_leveling / unapply_leveling functions), which I don't want to use on the CNC.

Could the code be updated to take the Skew correction into account outside of the bed_levelling routines when no BL is enabled ?

HekTron802, 2018-01-06:
Last I tested (a couple days before 1.7 was released) it was working without bed leveling. In fact we were having the opposite issue while testing the Skew feature! Anyways, you can always turn bed leveling on in the config but not run the leveling feature and you'll effectively have the skew correction without bed leveling.

thinkyhead, 2018-01-07:
> seems it's only used with bed levelling

It should already work without bed leveling enabled. See?

```cpp
#define PLANNER_LEVELING      (OLDSCHOOL_ABL || ENABLED(MESH_BED_LEVELING) || UBL_SEGMENTED || ENABLED(SKEW_CORRECTION))
```
```cpp
#if PLANNER_LEVELING
  /**
   * rx, ry, rz - Cartesian positions in mm
   *              Leveled XYZ on completion
   */
  void Planner::apply_leveling(float &rx, float &ry, float &rz) {

    #if ENABLED(SKEW_CORRECTION)
      skew(rx, ry, rz);
    #endif

    if (!leveling_active) return;
    . . .
}
```
Have you tested release 1.1.8 or the current `bugfix-1.1.x` branch?

TheSFReader, 2018-01-13:
I tested with the bugfix, but as far as I saw (when last I checked the code), the apply_levelling is only called when some kind of BL is applied. I may have missed a #ifdef or a define, though. 

(Would have to check a pre-compiled version of code)

thinkyhead, 2018-02-11:
> the `apply_leveling` is only called when some kind of BL is applied.

It's also called if skew is enabled, regardless of bed leveling.
A casual scan of the code confirms this.

closed 2018-02-11 by thinkyhead