MarlinFirmware/Marlin #9047: Filament sensor event not triggering M600 script

croisez, 2018-01-04:
### Bug Report
- Description:
I got a filament sensor, which is working. I am currently testing it with FW 1.1.x (latest sources from Github)

- Expected behaviour: 
When I was in RC branch (1.0.x), this feature was working, afaik. But now it seems that the detection of a filament runout does not launch the M600 script as before.

- Actual behaviour:
M600 script is working if launched from Pronterface.
M600 is not auto-launched by a filament runout detection event.

- Steps to reproduce:
Enable PIN_DEBUGGING
Test filament runout sensor, by using `M43 P4 W` command (_P4 is for Ramps1.4_).
Check filament runout event is well catched by the firmware.
Expect to see M600 procedure started.

Please, find a ZIP of my 
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/1603527/Config.zip)
`Configuration.h` and `Configuration_adv.h` in attachment of this bug.

Tannoo, 2018-01-05:
Try downloading, configuring, and experimenting with `bugfix-1.1.x`.

croisez, 2018-01-06:
I am now in bugfix-1.1.x branch, latest code.
But still same behavior.

Here is what I get in Pronterface when triggering the sensor by hand:
`>>>M43 P4 W`
`SENDING:M43 P4 W`
`Watching pins`
`PIN:   4   Port: G5        FIL_RUNOUT_PIN`
`.                          SERVO3_PIN`
`PIN:   4   Port: G5        FIL_RUNOUT_PIN`
`.                          SERVO3_PIN`

M600 script is still not executed.

I think I have found my mistake. I though I could use this runout filament sensing feature for autoloading a new filament when no filament is present. 
But the code assumes that the printer is printing, which is not right in my case.

From Marlin_main.cpp:13744:
`#if ENABLED(FILAMENT_RUNOUT_SENSOR)`
`    if ((IS_SD_PRINTING || print_job_timer.isRunning()) && (READ(FIL_RUNOUT_PIN) == FIL_RUNOUT_INVERTING))`
`      handle_filament_runout();`
`  #endif`

In fact I would like to implement a kind of filament autoloading like in Prusa MK3, when I would insert a new filament through the sensor.

closed 2018-02-11 by thinkyhead