MarlinFirmware/Marlin #9049: Error compiling 1.1.8 with ADVANCED_PAUSE_FEATURE

hpkevertje, 2018-01-04:
### Compile Error
When I compile with `ADVANCED_PAUSE_FEATURE` I get an error:
...
PAUSE_PARK_XY_FEEDRATE is now NOZZLE_PARK_XY_FEEDRATE. Please update your configuration.
...

I had to rename PAUSE_PARK_XY_FEEDRATE to NOZZLE_PARK_XY_FEEDRATE and PAUSE_PARK_Z_FEEDRATE to NOZZLE_PARK_Z_FEEDRATE in order for the compilation to succeed.

I used the settings from configuration_adv.h that was provided with the download, which still mention PAUSE_...

I'm new here, so if I'm reporting this incorrectly, let me know.

Regards,

Hans

tcm0116, 2018-01-05:
Are you sure you haven't started with configuration files from a previous version of Marlin? I just double checked 1.1.8 (and the latest bugfix-1.1.x), and `NOZZLE_PARK_XY_FEEDRATE` is defined in Configuration.h and `PAUSE_PARK_XY_FEEDRATE` is not present in Configuration_adv.h.

hpkevertje, 2018-01-05:
I just checked again. My bad. I did indeed merge these 2 lines from an old config. They are not in the downloaded file.

Sorry for this

Hans

closed 2018-01-05 by hpkevertje