MarlinFirmware/Marlin #9055: marlin define pins as mega pin #s, not as ramps pin #s

pbbob, 2018-01-04:
I am designing my own shield for the mega2650 that i will use to control my printer. however, when defining pins in the marlin firmware, the pin numbers are the pin numbers of the RAMPS, not the Arduino Mega itself. How do i define pins so that i could write `#define HEATER1PIN 5` and it will make heater1pin be on _**arduino mega pin 5**_ __--NOT--__ _**RAMPS pin 5_**

Tannoo, 2018-01-04:
[image]

Tannoo, 2018-01-04:
Pin 5 = D5 = Servo 3

Tannoo, 2018-01-05:
D66 = A12

Tannoo, 2018-01-05:
Look at the drawings. It's all there.

Bob-the-Kuhn, 2018-01-05:
Here's a couple of items I found useful.  

Curiosity - what does your shield do that RAMPS doesn't?

[image]
[arduino-mega2560-schematic.pdf](https://github.com/MarlinFirmware/Marlin/files/1605139/arduino-mega2560-schematic.pdf)

Tannoo, 2018-01-05:
ooh... I like the finger. lol