MarlinFirmware/Marlin #9058: Absolutely no Response from TMC2130 after PSU Upgrade

willtmakinen78, 2018-01-05:
### Bug Report
- Description: I use a Folgertech FT-5 printer with an MKS Gen 1.4 board. Three days ago, everything was working perfectly; I had four TMC2130s installed with StealthChop and Sensorless Homing enabled. Then, my power supply died, and after waiting a few days for a new ATX unit to arrive I installed the replacement today. Everything worked just fine; the lights turned on, the bed and extruder heated up properly, however I got zero response from any of the motors. I thought it might be a power issue, so I switched out the TMC drivers for standard Polulu ones, and everything worked fine. I then reinstalled the TMC2130s, and tried every combination of Stealthchop, Sensorless Homing, and Hybrid, all with the same result. Enabling TMC_BEBUG and MONITOR_DRIVER_STATUS yield no errors, even when disconnecting the SPI wires from one of the drivers.

- Expected behaviour: TMC2130 drivers move motors when commanded.
- Actual behaviour: Absolutely no response (whining, hissing, vibrations, etc.)
- Steps to reproduce:
  -Power Supply Replacement

It is entirely possible that the drivers were fried during the PSU upgrade, either from reverse current when moving the axis by hand or some other sort of short. If this is the case, is there a way to definitively tell if the drivers are dead? If not, what other troubleshooting steps can I take?

Here are my config files, note I changed some of the CS pins around, that is not the issue. The most concerning thing is that everything was working perfectly before changing the PSU, and afterwards the drivers ceased to work, with no firmware or wiring changes.

[TMC2130s Not Working.zip](https://github.com/MarlinFirmware/Marlin/files/1605438/TMC2130s.Not.Working.zip)

Artem-B, 2018-01-05:
Do the drivers work if you reset your board without removing power? The chip requires its power to be stable before it sees 5V on VCC_IO pin from arduino. There's a chance that 12V on your power supply stabilizes *after* the processor's power is already up and that will reset TMC2130 settings to defaults.

Another possibility is that you did get CS pins misconfigured. What does M122 show? This may need TMC_DEBUG enabled. If it shows that all warning flags are triggered and you see values like 1023 (all-1 in binary), chances are that you may have gotten CS pins wrong. It would help to verify that M43 output (enabled with PIN_DEBUG, I think) matches the actual wiring on your board.

teemuatlut, 2018-01-05:
Try the simple example from the library examples. This would rule out anything Marlin related.
Remember that the example will start driving the motor immediately and will crash an axis to its' limit. Remove the belt(s) first!

willtmakinen78, 2018-01-05:
All axis worked with the example, any thoughts on what could be wrong with Marlin?

willtmakinen78, 2018-01-05:
>  There's a chance that 12V on your power supply stabilizes after the processor's power is already up and that will reset TMC2130 settings to defaults.
Artem-B, I think that was it! What should I do to make sure the power is stabilized, or at least have the drivers not reset. Before, I had the LED PSU on an SSR, and the arduino itself was always powered on before the power supply, and this wasn't an issue. I don't get any warnings with M122.

teemuatlut, 2018-01-05:
`M80` will also init the drivers. We currently have a 100ms delay for the power to stabilize with the command but there is no delay for boot time init (note to self).

If you want to try fixing this yourself add `delay(500)` right before `tmc2130_init()` in [stepper.cpp line 956](https://github.com/MarlinFirmware/Marlin/blob/bugfix-1.1.x/Marlin/stepper.cpp#L956).

willtmakinen78, 2018-01-05:
@teemuatlut that unfortunately didn't do it for me. I increased the delay all the way up to 1000ms however no improvement.

teemuatlut, 2018-01-05:
Please post your M122 output.

willtmakinen78, 2018-01-05:
@teemuatlut When the power is connected before resetting the arduino it is as follows:
SENT: M122
READ: 		X	Y	Z	E0
READ: Enabled		false	false	false	false
READ: Set current	600	900	800	900
READ: RMS current	581	887	795	887
READ: MAX current	819	1251	1121	1251
READ: Run current	18/31	28/31	25/31	28/31
READ: Hold current	9/31	14/31	12/31	14/31
READ: CS actual		9/31	14/31	12/31	14/31
READ: PWM scale	1	1	1	1
READ: vsense		1=.18	1=.18	1=.18	1=.18
READ: stealthChop	true	true	true	true
READ: msteps		16	16	16	16
READ: tstep		1048575	1048575	1048575	1048575
READ: pwm
READ: threshold		0	0	0	0
READ: [mm/s]		-	-	-	-
READ: OT prewarn	false	false	false	false
READ: OT prewarn has
READ: been triggered	false	false	false	false
READ: off time		5	5	5	5
READ: blank time	24	24	24	24
READ: hysterisis
READ: -end		2	2	2	2
READ: -start		3	3	3	3
READ: Stallguard thrs	19	11	0	0
READ: DRVSTATUS	X	Y	Z	E0
READ: stallguard					
READ: sg_result		0	0	0	0
READ: fsactive					
READ: stst		X	X	X	X
READ: olb					
READ: ola					
READ: s2gb					
READ: s2ga					
READ: otpw					
READ: ot					
READ: Driver registers:
READ: 	X = 0x80:09:00:00
READ: 	Y = 0x80:0E:00:00
READ: 	Z = 0x80:0C:00:00
READ: 	E0 = 0x80:0E:00:00
READ: 
READ: ok

When M80 is used it looks like this:
SENT: M122
READ: 		X	Y	Z	E0
READ: Enabled		false	false	false	false
READ: Set current	600	900	800	900
READ: RMS current	1049	1602	1436	1602
READ: MAX current	1479	2259	2025	2259
READ: Run current	18/31	28/31	25/31	28/31
READ: Hold current	9/31	14/31	12/31	14/31
READ: CS actual		0/31	0/31	0/31	0/31
READ: PWM scale	0	0	0	0
READ: vsense		0=.325	0=.325	0=.325	0=.325
READ: stealthChop	false	false	false	false
READ: msteps		256	256	256	256
READ: tstep		1048575	1048575	1048575	1048575
READ: pwm
READ: threshold		0	0	0	0
READ: [mm/s]		-	-	-	-
READ: OT prewarn	false	false	false	false
READ: OT prewarn has
READ: been triggered	false	false	false	false
READ: off time		0	0	0	0
READ: blank time	16	16	16	16
READ: hysterisis
READ: -end		-3	-3	-3	-3
READ: -start		1	1	1	1
READ: Stallguard thrs	19	11	0	0
READ: DRVSTATUS	X	Y	Z	E0
READ: stallguard					
READ: sg_result		0	0	0	0
READ: fsactive					
READ: stst		X	X	X	X
READ: olb		X	X		
READ: ola		X	X		
READ: s2gb					
READ: s2ga					
READ: otpw					
READ: ot					
READ: Driver registers:
READ: 	X = 0xE0:00:00:00
READ: 	Y = 0xE0:00:00:00
READ: 	Z = 0x80:00:00:00
READ: 	E0 = 0x80:00:00:00
READ: 
READ: ok

A number of things are reset, including StealthChop.

teemuatlut, 2018-01-05:
Disable `PS_DEFAULT_OFF` and completely power cycle the machine.

Your first output looks better. Off Time and Blank Time and microsteps are correct and currents match up.
The second output not so much. It's as if the registers are read as all zeroes. But then the Driver registers are reported as having at least some hex value.

willtmakinen78, 2018-01-05:
@teemuatlut that seems to work for now, even after turning the PSU off with M81. Why is this? Shouldn't it run into the same issue as manually turning the PSU on, resetting the arduino, and then powering everything off again? Does M81 prepare the drivers to lose power/reset?

teemuatlut, 2018-01-05:
The drivers are configured on boot time.
But because if you have an ATX PSU with default off enabled, the drivers won't have power when booting and the configuration won't stick.
So when the user turns on the machine with M80 Marlin needs to resend those configurations.

willtmakinen78, 2018-01-05:
@teemuatlut interesting. Well, I'm glad I got this worked out. Thanks so much for your help!

closed 2018-01-05 by willtmakinen78