MarlinFirmware/Marlin #9059: ABL Mesh Editing Procedure

sarf2k4, 2018-01-05:
Hi,
I am using marlin 1.1.7 and my config snippet is at the bottom of this post. I decided to fix my printer that has issues with the false z-height on the edge of the bed while searching to consider other abl options, I saw the m420 and m421 to edit the mesh point.

I tried a little bit with the `m421 i2 j2 q-0.15` to adjust the offset of that particular point. The original Z value there was roughly 0.37 and if I apply the `m421 i2 j2 q-0.15`, logically it would change to 0.22. 

If my start of g-code looked like this
```
heat up
g28
g29
m421 i2 j2 q-0.15
```
Will marlin uses 0.37 or 0.22?

Another thing also I would like to add is to request a feature if this is possible, for Bilinear abl, the mesh offset adjustment can be saved in eeprom.

Here is my config snippet for marlin 1.1.7 ABL settings
```
//===========================================================================
//============================= Z Probe Options =============================
//===========================================================================
// @section probes

//
// See http://marlinfw.org/docs/configuration/probes.html
//

/**
 * Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
 *
 * Enable this option for a probe connected to the Z Min endstop pin.
 */
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

/**
 * Z_MIN_PROBE_ENDSTOP
 *
 * Enable this option for a probe connected to any pin except Z-Min.
 * (By default Marlin assumes the Z-Max endstop pin.)
 * To use a custom Z Probe pin, set Z_MIN_PROBE_PIN below.
 *
 *  - The simplest option is to use a free endstop connector.
 *  - Use 5V for powered (usually inductive) sensors.
 *
 *  - RAMPS 1.3/1.4 boards may use the 5V, GND, and Aux4->D32 pin:
 *    - For simple switches connect...
 *      - normally-closed switches to GND and D32.
 *      - normally-open switches to 5V and D32.
 *
 * WARNING: Setting the wrong pin may have unexpected and potentially
 * disastrous consequences. Use with caution and do your homework.
 *
 */
//#define Z_MIN_PROBE_ENDSTOP

/**
 * Probe Type
 *
 * Allen Key Probes, Servo Probes, Z-Sled Probes, FIX_MOUNTED_PROBE, etc.
 * Activate one of these to use Auto Bed Leveling below.
 */

/**
 * The "Manual Probe" provides a means to do "Auto" Bed Leveling without a probe.
 * Use G29 repeatedly, adjusting the Z height at each point with movement commands
 * or (with LCD_BED_LEVELING) the LCD controller.
 */
//#define PROBE_MANUALLY

/**
[... 222 lines cut ...]

#elif ENABLED(AUTO_BED_LEVELING_UBL)

  //===========================================================================
  //========================= Unified Bed Leveling ============================
  //===========================================================================

  // #define MESH_EDIT_GFX_OVERLAY   // Display a graphics overlay while editing the mesh

  #define MESH_INSET 1              // Mesh inset margin on print area
  #define GRID_MAX_POINTS_X 10      // Don't use more than 15 points per axis, implementation limited.
  #define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X

  #define UBL_PROBE_PT_1_X 39       // Probing points for 3-Point leveling of the mesh
  #define UBL_PROBE_PT_1_Y 180
  #define UBL_PROBE_PT_2_X 39
  #define UBL_PROBE_PT_2_Y 20
  #define UBL_PROBE_PT_3_X 180
  #define UBL_PROBE_PT_3_Y 20

  #define UBL_MESH_EDIT_MOVES_Z     // Sophisticated users prefer no movement of nozzle
  #define UBL_SAVE_ACTIVE_ON_M500   // Save the currently active mesh in the current slot on M500

#elif ENABLED(MESH_BED_LEVELING)
```

Roxy-3D, 2018-01-05:
The 3-Point and Grid based Auto Bed Leveling do not have a mesh.   They construct (and tilt) a plane in an attempt to minimize positioning errors.

sarf2k4, 2018-01-05:
I am actually using bilinear type

fiveangle, 2018-01-09:
> is possible, for Bilinear abl, the mesh offset adjustment can be saved in eeprom.

M500 saves mesh parameters

thinkyhead, 2018-01-09:
> Will marlin uses 0.37 or 0.22?

It will use whatever the current mesh values are, as shown by `M420 V`.

closed 2018-01-09 by thinkyhead