MarlinFirmware/Marlin #9061: Firmware for Geetech Delta printerRostok Mini Copy.

ChrisP28, 2018-01-05:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

cjsoong, 2018-01-05:
download firmware marlin 1.1.8
arduino ide V1.90 版

Roxy-3D, 2018-01-08:
> Has any one got a copy of firmware that will work with this printer.

Ah...  Yes.   If you mean the printer in this review:  3dprintboard.com/showthread.php?16024-Geeetech-G2s-Printer-Kit-Review

dembach, 2018-01-10:
on Geetech they are referring to Marlin ;-)

[https://www.geeetech.com/wiki/index.php/GT2560](url)

the Link to the Firmware archive : 
[http://www.geeetech.com/wiki/images/3/39/Marlin-Marlin_v1.zip](url)

so you only need examine the configuration and the pins to find out, if the board is anything special.

The configuration for a delta is described on the marlin.org documetation.

closed 2019-07-06 by shitcreek