MarlinFirmware/Marlin #9063: [2.0] Advanced Paused Feature Stop after pause not work properly

Nand-e, 2018-01-05:
- Description: 
   If Advanced Paused feature is enabled and you start printing from SD and pause printing after some minutes and maybe something goes wrong with your printing. So you decide to stop printing and STOP print. It works but after You would like to start another print from SD it will not work. 
The printer will try to continue the previous print.

- Expected behavior:
   After Pause and Stop. And start print from SD it should start a new printing job.

- Steps to reproduce:
  1. Enable Advanced Pause feature.
  2. Start Printing from SD.
  3. Pause Printing and wait some seconds.
  4. Stop Printing.
  5. Try to print from SD again.

Tannoo, 2018-01-06:
You did not follow the issue template...at least post your config files.

Likely, you have `SD_REPRINT_LAST_SELECTED_FILE` enabled.

Witgang, 2018-01-07:
i had the exact same bug once but i did not remember'd the steps to reproduce it so i did not post about it.
   also i had modified the stop print function to do a park nozzle comand during that time so i atributed it to that.
 just checked my config files and SD_REPRINT_LAST_SELECTED_FILE is not enabled so something else may cause this.
  i'll wait for current print to finish and try reproducing it, if i get the same results i will post my config files.

Nand-e, 2018-01-07:
Witgang

Thanks, it is very anoying problem. The another one is when you stop printing the steppers some time are not disabled.

Witgang, 2018-01-11:
i had the time to clone the main marlin repository , change the files i have changed to use with my printer and test the bug and i was able to reproduce it.
  i think it might not get triggered when park nozzle on pause is not enabled.
   anyway here is an exact copy of what i have uploaded on my printer at the moment:[https://github.com/Witgang/Anet-A6-setup](url)

Witgang, 2018-01-11:
apparently it was not what i had changed:/ don't know how but just after i clone/downloaded the marlin-1.1x that i changed on my computer  the 1.1x bugfix branch changed already before i made the clone, now when downloading the clone i made i can't compile due to some name changes that were made.. now i gota re clone the main bugfix 1.1 branch then download it, change that and test it then upload my changes... anyway github's way of managing branches/pull requets and such is sooo confusing for new ppl...

Witgang, 2018-01-11:
apparently i was mistaking the 1.1x to be the same as bugfix1.1x ....
  here's my personal fork with the changes made https://github.com/Witgang/Marlin

thinkyhead, 2018-02-11:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see if we fixed this issue. If the problem has been solved then we can close it. If you still see the bad behavior we should investigate further.

boelle, 2019-02-20:
@Nand-e Please test with the latest `bugfix-2.0.x`

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead