MarlinFirmware/Marlin #9065: [1.1.8] TMC2130 current step down is not working properly

Lyr3x, 2018-01-05:
### Bug Report
- Description: MONITOR_DRIVER_STATUS is activated. While printing i got some messages from all drivers:
E driver overtemperature warning! (850mA)
- Expected behaviour: Marlin should step down the current by 50mA (works fine in 1.1.7
- Actual behaviour: Warning will popup frequently and no current step down

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/1606480/Archive.zip)

teemuatlut, 2018-01-05:
There were no changes to the logic going from 1.1.7 to 1.1.8.
Try redownloading the repository.

Lyr3x, 2018-01-05:
No changes. Also my motors are getting quite hot quickly. May you check my configuration files? 

Running my X Motor@ 900mA (anything lower will result in missing steps)
Y Motor@800mA
Z , E0 and E1@750mA

teemuatlut, 2018-01-05:
You may want to add some cooling at those currents. It's always a battle of acceleration vs stepper current vs stepper cooling.

I re-read the step down code (it's been a while since I wrote it) and it may be that it's working as intended. What's happening is that Marlin is polling the drivers twice a second and if it sees the overtemperature prewarn condition (otpw) then it reports it to the user. However, we don't want to decrease current every time we poll and see this. Imagine we have the otpw flag triggered and it lasts for two or three seconds. During that time we would decrease the current by as much as 300mA! So what Marlin does is allow the counter to reach four (one uptick per reported otpw flag) before decreasing current. Every time the flag is not raised by the driver, Marlin decreases the counter.

If you want to test if it works like it's supposed to, send a `M906 X1100`, keep the motor moving and the current _should_ come down. As for why v1.1.7 works while v1.1.8 doesn't, I really don't know at this point.

Lyr3x, 2018-01-05:
Allright, ill test this later. Any tips for cooling the motors?

Lyr3x, 2018-01-05:
Allright, i tested some stuff today. Step down is working fine and also my Motors are doing good so far. It was a misconfiguration ( travel acceleration was too high).

closed 2018-01-05 by Lyr3x

jpscholz, 2018-03-19:
Hello, I ran into a similar problem. Help would be much appreciated.

> If you want to test if it works like it's supposed to, send a M906 X1100, keep the motor moving and the current should come down.

I set the current to 1100mA on purpose and moved the stepper until i got the overtemperature prewarn message (otpw). After a while the message repeated but the current wasn't reduced. I took a look into the current step down code in Marlin_main.cpp (1.1.8) at line ~13900 and added a bit of debug code to understand what was going on. The otpw flag was set and the counter counted up, way above 4. The only thing that kept the code from reducing the current was `if (!st.isEnabled())`. As far as I understand this prevents the current from being reduced when the motor is enabled. The problem is that while using the motor, it is always enabled so the current can't be reduced. 
@teemuatlut Is this intended or am I missing something else here? Thanks for the help!

teemuatlut, 2018-03-19:
Good find. I'll need to check if this was for some purpose or if there's been a change that switched the logic somehow.

thinkyhead, 2018-03-20:
@teemuatlut — The change was made in #8769

teemuatlut, 2018-03-20:
It certainly does look like the exclamation point is not supposed to be there and most likely a remnant of the previous implementation.

jpscholz, 2018-03-20:
I just tried the code without the exclamation mark like this:

`#if CURRENT_STEP_DOWN > 0
  // Decrease current if is_otpw is true and driver is enabled and there's been more then 4 warnings
  SERIAL_ECHO(int(otpw_cnt)); //DEBUG OUTPUT
  if(data.is_otpw){SERIAL_ECHOLN("  is_otpw");} //DEBUG OUTPUT
  if(st.isEnabled()){SERIAL_ECHOLN("  st.isEnable");} //DEBUG OUTPUT
  if(otpw_cnt > 4){SERIAL_ECHOLN("  otpw_cnt>4");} //DEBUG OUTPUT
  if (data.is_otpw && st.isEnabled() && otpw_cnt > 4) { //REMOVED !
	st.setCurrent(st.getCurrent() - CURRENT_STEP_DOWN, R_SENSE, HOLD_MULTIPLIER);
	#if ENABLED(REPORT_CURRENT_CHANGE)
	  SERIAL_ECHO(axisID);
	  SERIAL_ECHOLNPAIR(" current decreased to ", st.getCurrent());
	#endif
  }
#endif`

The output I get is this (the number is the otpw_cnt): 
`0  st.isEnable
0  st.isEnable
00:00: X driver overtemperature warning! (1100mA)
0  is_otpw
  st.isEnable
1  is_otpw
  st.isEnable
2  is_otpw
  st.isEnable
3  is_otpw
  st.isEnable
4  is_otpw
  st.isEnable
5  is_otpw
  st.isEnable
  otpw_cnt>4
X current decreased to 1050
6  is_otpw
  st.isEnable
  otpw_cnt>4
X current decreased to 1000
7  is_otpw
  st.isEnable
  otpw_cnt>4
X current decreased to 950
8  st.isEnable
  otpw_cnt>4
7  st.isEnable
  otpw_cnt>4
6  st.isEnable
  otpw_cnt>4
5  st.isEnable
  otpw_cnt>4
4  st.isEnable
3  st.isEnable
2  st.isEnable
1  st.isEnable
0  st.isEnable
0  st.isEnable`

This means the current step down function works and decreses the current until the is_otpw flag gets removed. But as long as the is_otpw flag remains true the current is stepped down further. Is this intended? I would think that it might be better to reset the otpw_cnt to 0 once the current was stepped down to give the chip the time to cool down a bit if the is_otpw flag remains true then the counter counts back up and the current gets decreased one step furter. The way it is now the current might be stepped down too far because the chip doesn't cool down quickly enough. On the other hand resetting the counter might lead to overheating because the current isn't stepped down fast enough.
But since I don't know how fast a chip takes to cool down I'm unsure what would be the best way to do this, it's just an idea from me.

teemuatlut, 2018-03-20:
#10169