MarlinFirmware/Marlin #9070: [TMC2130] Current and Threshold changes with dual Z

Lyr3x, 2018-01-06:
- Description: When changing motor currents and hybrid threshold, we cant change the values for the Z steppers with dual z setup (Z and E1). 
M906 Z <val> will change only Z and M906 E <val> will change the values for every EX. Same for M913.

- Expected behaviour: We need do configure E0, E1, ..., EX separately
- Actual behaviour: M906/M913 E changes the values for all Ex drivers

teemuatlut, 2018-01-06:
It's expected that every Z driver will be operating at the same current. Same goes for E or X or Y drivers.

The limitation mostly comes from the Marlin gcode handler which cannot differentiate same axis steppers.
Because we ignore spaces. commands like `X 210` are the same as `X210`, and we can't know if a command `X210` means `X 210` or `X2 10`.

Lyr3x, 2018-01-06:
I get that. But as mostly the dual z setup uses the E1 driver as the second Z driver and you can set the different current in the firmware for all drivers maybe the logic can be changed a bit.

I am not very familiar with the code, so maybe this wont work or it will only work with the ramps board:

When `#define Z_DUAL_STEPPER_DRIVERS` is used with one extruder, the `E1` is the second Z driver. Cant you just tell the part which changes the current, that in this case the `M906 Zx` command will change the value vor E1 as well?
I see the generalisation problem there...

I think im not the only one with dual z drivers out there, so maybe it is worth to look into it

teemuatlut, 2018-01-06:
Not really my specialty here but I believe "dual Z uses E1" only means that on compile time the E1 pins are defined as Z2 pins. But it doesn't mean that E1 _is_ Z2.
The logic is likely done purely so that users who put their Z2 driver on the E1 slot, can make the logical jump to defining the pins as such.
And I am one of those who run dual Z with a couple of TMC2208 drivers.

thinkyhead, 2018-02-11:
> Cant you just tell the part which changes the current, that in this case the M906 Zx command will change the value vor E1 as well?

That's how it works now.

closed 2018-02-11 by thinkyhead