MarlinFirmware/Marlin #9072: 1.1.8 Compile error (using the example Anet A8 config)

Mystikal91, 2018-01-06:
I guys, when I try to compile my Marlin 1.1.8 I get this error. I'll attach the configuration files too:

Arduino:1.8.0 (Windows 10), Scheda:"Anet V1.0"

In file included from sketch\Marlin_main.cpp:314:0:

endstop_interrupts.h:124: error: missing binary operator before token "("

     #if digitalPinToInterrupt(X_MIN_PIN) != NOT_AN_INTERRUPT

                              ^

endstop_interrupts.h:144: error: missing binary operator before token "("

     #if digitalPinToInterrupt(Y_MIN_PIN) != NOT_AN_INTERRUPT

                              ^

endstop_interrupts.h:164: error: missing binary operator before token "("

     #if digitalPinToInterrupt(Z_MIN_PIN) != NOT_AN_INTERRUPT

                              ^

Più di una libreria trovata per "LiquidCrystal.h"
Usata: D:\Moved\Desktop\Marlin\arduino-1.8.0\arduino-1.8.0\portable\sketchbook\libraries\LiquidCrystal
Non usata: D:\Moved\Desktop\Marlin\arduino-1.8.0\arduino-1.8.0\libraries\LiquidCrystal
exit status 1
missing binary operator before token "("

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/1608777/Config.zip)

GMagician, 2018-01-06:
I succesfully compiled with arduino 1.8.4 & Win 10 using your configuration files.
Board: Sanguino CPU: atmega1284p
Can you specify your board/cpu?

GMagician, 2018-01-06:
I also installed anet board and it continues to compile

kingb, 2018-01-07:
I'm getting the exact same error using Arduino 1.8.5 w/ ANET Board definition pulled from the ANET hardware defined from the Skynet v2.2 package. I couldn't figure out how to get the ANET board selection in Arduino 1.8.5 without that.

Maybe that's related to the error?

kingb, 2018-01-07:
Heads up, I just tried following the directions from section "Arduino IDE hardware definition for Anet main board V1.0" (from https://linuxgnublog.org/en/dealing-with-an-anet-a8-3d-printer-firmware/), and installed Sanguino ATMega1284P 16Hz definition/board and the code compiled.

@GMagician where did you get your ANET board definition?

fiveangle, 2018-01-07:
Using default [Anet8 config example](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x/Marlin/src/config/examples/Anet/A8) Marlin bugfix2.0.x also compiles without issue:

```
avr-g++ -o .pioenvs/anet10/firmware.elf -Os -mmcu=atmega1284p -Wl,--gc-sections -flto -fuse-linker-plugin .pioenvs/anet10/src/src/HAL/HAL_AVR/HAL_AVR.o .pioenvs/anet10/src/src/HAL/HAL_AVR/HAL_spi_AVR.o .pioenvs/anet10/src/src/HAL/HAL_AVR/MarlinSerial.o .pioenvs/anet10/src/src/HAL/HAL_AVR/persistent_store_impl.o .pioenvs/anet10/src/src/HAL/HAL_AVR/servo_AVR.o .pioenvs/anet10/src/src/HAL/HAL_AVR/watchdog_AVR.o .pioenvs/anet10/src/src/HAL/HAL_DUE/EepromEmulation_Due.o .pioenvs/anet10/src/src/HAL/HAL_DUE/HAL_Due.o .pioenvs/anet10/src/src/HAL/HAL_DUE/HAL_spi_Due.o .pioenvs/anet10/src/src/HAL/HAL_DUE/HAL_timers_Due.o .pioenvs/anet10/src/src/HAL/HAL_DUE/InterruptVectors_Due.o .pioenvs/anet10/src/src/HAL/HAL_DUE/MarlinSerialUSB_Due.o .pioenvs/anet10/src/src/HAL/HAL_DUE/MarlinSerial_Due.o .pioenvs/anet10/src/src/HAL/HAL_DUE/Servo_Due.o .pioenvs/anet10/src/src/HAL/HAL_DUE/persistent_store_impl.o .pioenvs/anet10/src/src/HAL/HAL_DUE/u8g_com_HAL_DUE_st7920_sw_spi.o .pioenvs/anet10/src/src/HAL/HAL_DUE/usb/ctrl_access.o .pioenvs/anet10/src/src/HAL/HAL_DUE/usb/sd_mmc_spi_mem.o .pioenvs/anet10/src/src/HAL/HAL_DUE/usb/sysclk.o .pioenvs/anet10/src/src/HAL/HAL_DUE/usb/udc.o .pioenvs/anet10/src/src/HAL/HAL_DUE/usb/udi_cdc.o .pioenvs/anet10/src/src/HAL/HAL_DUE/usb/udi_composite_desc.o .pioenvs/anet10/src/src/HAL/HAL_DUE/usb/udi_msc.o .pioenvs/anet10/src/src/HAL/HAL_DUE/usb/uotghs_device_due.o .pioenvs/anet10/src/src/HAL/HAL_DUE/usb/usb_task.o .pioenvs/anet10/src/src/HAL/HAL_DUE/watchdog_Due.o .pioenvs/anet10/src/src/HAL/HAL_LPC1768/HAL.o .pioenvs/anet10/src/src/HAL/HAL_LPC1768/HAL_LCD_I2C_routines.o .pioenvs/anet10/src/src/HAL/HAL_LPC1768/HAL_LCD_pin_routines.o .pioenvs/anet10/src/src/HAL/HAL_LPC1768/HAL_spi.o .pioenvs/anet10/src/src/HAL/HAL_LPC1768/HAL_timers.o .pioenvs/anet10/src/src/HAL/HAL_LPC1768/HardwareSerial.o .pioenvs/anet10/src/src/HAL/HAL_LPC1768/LPC1768_PWM.o .pioenvs/anet10/src/src/HAL/HAL_LPC1768/LPC1768_Servo.o .pioenvs/anet10/src/src/HAL/HAL_LPC1768/SoftwareSerial.o .pioenvs/anet10/src/src/HAL/HAL_LPC1768/WInterrupts.o .pioenvs/anet10/src/src/HAL/HAL_LPC1768/arduino.o .pioenvs/anet10/src/src/HAL/HAL_LPC1768/include/digipot_mcp4451_I2C_routines.o .pioenvs/anet10/src/src/HAL/HAL_LPC1768/main.o .pioenvs/anet10/src/src/HAL/HAL_LPC1768/persistent_store_impl.o .pioenvs/anet10/src/src/HAL/HAL_LPC1768/pinmapping.o .pioenvs/anet10/src/src/HAL/HAL_LPC1768/u8g_com_HAL_LPC1768_hw_spi.o .pioenvs/anet10/src/src/HAL/HAL_LPC1768/u8g_com_HAL_LPC1768_ssd_hw_i2c.o .pioenvs/anet10/src/src/HAL/HAL_LPC1768/u8g_com_HAL_LPC1768_st7920_hw_spi.o .pioenvs/anet10/src/src/HAL/HAL_LPC1768/u8g_com_HAL_LPC1768_st7920_sw_spi.o .pioenvs/anet10/src/src/HAL/HAL_LPC1768/u8g_com_HAL_LPC1768_sw_spi.o .pioenvs/anet10/src/src/HAL/HAL_LPC1768/watchdog.o .pioenvs/anet10/src/src/HAL/HAL_STM32F1/HAL_Servo_Stm32f1.o .pioenvs/anet10/src/src/HAL/HAL_STM32F1/HAL_Stm32f1.o .pioenvs/anet10/src/src/HAL/HAL_STM32F1/HAL_spi_Stm32f1.o .pioenvs/anet10/src/src/HAL/HAL_STM32F1/HAL_timers_Stm32f1.o .pioenvs/anet10/src/src/HAL/HAL_STM32F1/persistent_store_flash.o .pioenvs/anet10/src/src/HAL/HAL_STM32F1/persistent_store_impl.o .pioenvs/anet10/src/src/HAL/HAL_STM32F1/watchdog_Stm32f1.o .pioenvs/anet10/src/src/HAL/HAL_TEENSY35_36/HAL_Servo_Teensy.o .pioenvs/anet10/src/src/HAL/HAL_TEENSY35_36/HAL_Teensy.o .pioenvs/anet10/src/src/HAL/HAL_TEENSY35_36/HAL_spi_Teensy.o .pioenvs/anet10/src/src/HAL/HAL_TEENSY35_36/HAL_timers_Teensy.o .pioenvs/anet10/src/src/HAL/HAL_TEENSY35_36/persistent_store_impl.o .pioenvs/anet10/src/src/HAL/HAL_TEENSY35_36/watchdog_Teensy.o .pioenvs/anet10/src/src/HAL/I2cEeprom.o .pioenvs/anet10/src/src/HAL/SpiEeprom.o .pioenvs/anet10/src/src/HAL/servo.o .pioenvs/anet10/src/src/Marlin.o .pioenvs/anet10/src/src/core/serial.o .pioenvs/anet10/src/src/core/utility.o .pioenvs/anet10/src/src/feature/I2CPositionEncoder.o .pioenvs/anet10/src/src/feature/Max7219_Debug_LEDs.o .pioenvs/anet10/src/src/feature/baricuda.o .pioenvs/anet10/src/src/feature/bedlevel/abl/abl.o .pioenvs/anet10/src/src/feature/bedlevel/bedlevel.o .pioenvs/anet10/src/src/feature/bedlevel/mbl/mesh_bed_leveling.o .pioenvs/anet10/src/src/feature/bedlevel/ubl/ubl.o .pioenvs/anet10/src/src/feature/bedlevel/ubl/ubl_G29.o .pioenvs/anet10/src/src/feature/bedlevel/ubl/ubl_motion.o .pioenvs/anet10/src/src/feature/caselight.o .pioenvs/anet10/src/src/feature/controllerfan.o .pioenvs/anet10/src/src/feature/dac/dac_dac084s085.o .pioenvs/anet10/src/src/feature/dac/dac_mcp4728.o .pioenvs/anet10/src/src/feature/dac/stepper_dac.o .pioenvs/anet10/src/src/feature/digipot/digipot_mcp4018.o .pioenvs/anet10/src/src/feature/digipot/digipot_mcp4451.o .pioenvs/anet10/src/src/feature/fanmux.o .pioenvs/anet10/src/src/feature/filwidth.o .pioenvs/anet10/src/src/feature/fwretract.o .pioenvs/anet10/src/src/feature/leds/blinkm.o .pioenvs/anet10/src/src/feature/leds/leds.o .pioenvs/anet10/src/src/feature/leds/neopixel.o .pioenvs/anet10/src/src/feature/leds/pca9632.o .pioenvs/anet10/src/src/feature/leds/tempstat.o .pioenvs/anet10/src/src/feature/mixing.o .pioenvs/anet10/src/src/feature/pause.o .pioenvs/anet10/src/src/feature/runout.o .pioenvs/anet10/src/src/feature/snmm.o .pioenvs/anet10/src/src/feature/solenoid.o .pioenvs/anet10/src/src/feature/tmc_util.o .pioenvs/anet10/src/src/feature/twibus.o .pioenvs/anet10/src/src/gcode/bedlevel/G26.o . [... cut ...]
Environment megaatmega1280          	[SKIP]
Environment anet10                  	[SUCCESS]
Environment DUE                     	[SKIP]
Environment LPC1768                 	[SKIP]
Environment LPC1768_debug_and_upload	[SKIP]
Environment rambo                   	[SKIP]
Environment sanguino_atmega644p     	[SKIP]
Environment STM32F1                 	[SKIP]
Environment teensy20                	[SKIP]
Environment teensy35                	[SKIP]
========================= [SUCCESS] Took 20.23 seconds =========================
```

kingb, 2018-01-07:
UPDATE 3:

I tried uploading to my ANET A8 (ANET v1.0 board) with Sanguino ATMega1284P 16Mhz selection in Arduino IDE. I got sync errors and it failed.

I then installed the ANET Board definitions for Arduino IDE from here: https://github.com/SkyNet3D/anet-board

With those definitions I was able to upload the Marlin 1.1.8 firmware. 

@Mystikal91 Does that fix your problem too?

GMagician, 2018-01-07:
@Mystikal91  & @kingb I searched with google and download anet board files. I attach the one I've found
[hardware.zip](https://github.com/MarlinFirmware/Marlin/files/1609646/hardware.zip)
I copied these files in C:\Users\[username]\Documents\Arduino (note that zip contains 'hardware' folder)

kingb, 2018-01-13:
@Mystikal91 Did that solve your problem?

closed 2018-02-11 by thinkyhead