MarlinFirmware/Marlin #9076: Rumba with tmc2130

benny64, 2018-01-06:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```Arduino: 1.8.5 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

Build options changed, rebuilding all
stepper_indirection.cpp:139: error: 'X_CS_PIN' was not declared in this scope

     _TMC2130_DEFINE(X);

                     ^

sketch\stepper_indirection.cpp:135:104: note: in definition of macro '_TMC2130_DEFINE'

   #define _TMC2130_DEFINE(ST) TMC2130Stepper stepper##ST(ST##_ENABLE_PIN, ST##_DIR_PIN, ST##_STEP_PIN, ST##_CS_PIN)

                                                                                                        ^

stepper_indirection.cpp:145: error: 'Y_CS_PIN' was not declared in this scope

     _TMC2130_DEFINE(Y);

                     ^

sketch\stepper_indirection.cpp:135:104: note: in definition of macro '_TMC2130_DEFINE'

   #define _TMC2130_DEFINE(ST) TMC2130Stepper stepper##ST(ST##_ENABLE_PIN, ST##_DIR_PIN, ST##_STEP_PIN, ST##_CS_PIN)

                                                                                                        ^

exit status 1
'X_CS_PIN' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```
Trying to get it working but keep get errors, can't find where to put cs pin, there doesn't seem anywhere to put it in rumba.pins, 

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

teemuatlut, 2018-01-06:
Add the definition to the `pins_RUMBA.h` file. You can see `pins_RAMPS.h` for an example.

benny64, 2018-01-06:
Thank you that sorted it, tried before but copied wrong pins.

closed 2018-01-06 by benny64