MarlinFirmware/Marlin #9080: [bugfix-1.1.x] TMC current not set

ikarisan, 2018-01-06:
### Bug Report
- Description: The set current for the TMC2130 driver is not used. After a few seconds the driver runs into an "overtemperature warning".

- Expected behaviour: The configured currents should be used

- Actual behaviour: I am using a TMC2130 on my X and Y axis. In "configuration_adv.h" I have set X_CURRENT and Y_CURRENT both to 650. But when I am moving the axis the used current is much to high! This is, what M122 reports after a reset:

```
SENDING:M122
		X	Y
Enabled		true	true
Set current	0	0
RMS current	1767	1767
MAX current	2491	2491
Run current	31/31	31/31
Hold current	31/31	31/31
CS actual		31/31	31/31
PWM scale	255	255
vsense		0=.325	0=.325
```

And this is the related configuration_adv section:

```
#define R_SENSE           0.11  // R_sense resistor for SilentStepStick2130
  #define HOLD_MULTIPLIER    0.5  // Scales down the holding current from run current
  #define INTERPOLATE       true  // Interpolate X/Y/Z_MICROSTEPS to 256

  #define X_CURRENT          650  // rms current in mA. Multiply by 1.41 for peak current.
  #define X_MICROSTEPS        16  // 0..256

  #define Y_CURRENT          650
  #define Y_MICROSTEPS        16
```

After calling "M906 X650 Y650" M122 reports the correct currents and the axis are moving without any temperature warnings.

[Marlin_Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1609137/Marlin_Configuration.zip)

teemuatlut, 2018-01-06:
Send `M500` to store the current settings to EEPROM.

ikarisan, 2018-01-06:
Thanks, I'll try this. But apart from that, shouldn't the values from the configuration file be used? Or do I miss something? O.o

teemuatlut, 2018-01-06:
If the stored settings version has not changed between flashes, then Marlin will load the settings from EEPROM. `M502` will restore settings to how they were set in configuration files.

ikarisan, 2018-01-07:
@teemuatlut Thanks, will try this. But I am a little surprised that Marlin reads out values from the EEPROM that I have never set like this! For my first try with the TMC2130 drivers I did not change the default values (900mA?!).

thinkyhead, 2018-01-08:
> I am a little surprised that Marlin reads out values from the EEPROM that I have never set

Marlin writes placeholder data to EEPROM for disabled features. For TMC motor currents it just writes zeros. When loading from EEPROM it should set the current to the configured value in response to a zero, but a patch is needed:
```diff
- #define SET_CURR(N,Q) stepper##Q.setCurrent(val[N], R_SENSE, HOLD_MULTIPLIER)
+ #define SET_CURR(N,Q) stepper##Q.setCurrent(val[N] ? val[N] : Q##_CURRENT, R_SENSE, HOLD_MULTIPLIER)
```

closed 2018-01-08 by thinkyhead