MarlinFirmware/Marlin #9082: MKS GEN 1.4 + TMC2130: Printer freezes after 4 layers

mesnik, 2018-01-06:
Hopefully someone can help me!! Thanks in advance!!!!

### Bug Report
- Description: Printer locks up after printing 4 layers.

- Expected behaviour: Print the entire part. Be able to reboot and not stay locked

- Actual behaviour: After 4 layers, the printed stopped moving. Even turning the power on and off, I couldn't connect with the printer via Simplify nor Repetier. The next day, I moved the Z axis by hand and was able to get it back to life. Tried to print again, and now it is locked. The Z axis trick didn't work. Printer is locked.

- Steps to reproduce: It happened the first time and by fluke I got it unblocked. All I did was to send a new benchy to print.

[Marlin118.zip](https://github.com/MarlinFirmware/Marlin/files/1609190/Marlin118.zip)

mesnik, 2018-01-06:
Extra info:

Just took all the TMCs wiring out, and redid all of them.  Printer reconnected with Simplify, I was able to move the X axis for a couple of seconds, and all locked up again.  I have the wring hook up per Tom's youtube video for the stepper drivers into aux2 and aux3.  I have no LCD yet (couldn't get that to work yet).

lemmingDev, 2018-01-06:
Are they the fysetc or watterott ones?

I'm having problems with the fysetc ones...

mesnik, 2018-01-06:
I bought them at Digikey, in the US... not sure how to tell the difference...

mesnik, 2018-01-06:
...looking closer, I see the www.watternott.com printed on them

mesnik, 2018-01-06:
FINAL RESULTS and ISSUE UPDATED on first post

closed 2018-01-07 by thinkyhead

mesnik, 2018-01-07:
Sorry, why was this closed? Thanks!!