MarlinFirmware/Marlin #9083: NO_VOLUMETRICS COMPILE ERROR

Rigid3D, 2018-01-06:
### Compile Error
When I compile with `NO_VOLUMETRICS` I get an error:

When NO_VOLUMETRICS is defined in configuration_adv.h MARLIN does not compile, with the error message listed below.

NO_VOLUMETRICS option is not properly handled in ultralcd for menu items relating to volumetric extrusion.

ERROR MESSAGE:

Arduino: 1.8.5 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch\ultralcd.cpp: In function 'void lcd_control_filament_menu()':

ultralcd.cpp:3638: error: 'class Planner' has no member named 'filament_size'

     MENU_MULTIPLIER_ITEM_EDIT_CALLBACK(float43, MSG_FILAMENT_DIAM, &planner.filament_size[0], 1.5, 3.25, planner.calculate_volumetric_multipliers);

                                                                             ^
sketch\ultralcd.cpp:290:30: note: in definition of macro '_MENU_ITEM_PART_2'

     menu_action_ ## TYPE(__VA_ARGS__); \

                          ^
sketch\ultralcd.cpp:345:66: note: in expansion of macro 'MENU_MULTIPLIER_ITEM'

 #define MENU_MULTIPLIER_ITEM_EDIT_CALLBACK(type, label, ...) MENU_MULTIPLIER_ITEM(setting_edit_callback_ ## type, label, PSTR(label), ## __VA_ARGS__)

                                                              ^
sketch\ultralcd.cpp:3638:9: note: in expansion of macro 'MENU_MULTIPLIER_ITEM_EDIT_CALLBACK'

     MENU_MULTIPLIER_ITEM_EDIT_CALLBACK(float43, MSG_FILAMENT_DIAM, &planner.filament_size[0], 1.5, 3.25, planner.calculate_volumetric_multipliers);

     ^
ultralcd.cpp:3638: error: 'class Planner' has no member named 'filament_size'

     MENU_MULTIPLIER_ITEM_EDIT_CALLBACK(float43, MSG_FILAMENT_DIAM, &planner.filament_size[0], 1.5, 3.25, planner.calculate_volumetric_multipliers);

                                                                             ^
sketch\ultralcd.cpp:294:102: note: in definition of macro '_MENU_ITEM_PART_2'

     lcd_implementation_drawmenu_ ## TYPE(encoderLine == _thisItemNr, _lcdLineNr, PSTR(LABEL), ## __VA_ARGS__); \

                                                                                                  ^
sketch\ultralcd.cpp:345:66: note: in expansion of macro 'MENU_MULTIPLIER_ITEM'

 #define MENU_MULTIPLIER_ITEM_EDIT_CALLBACK(type, label, ...) MENU_MULTIPLIER_ITEM(setting_edit_callback_ ## type, label, PSTR(label), ## __VA_ARGS__)

                                                              ^
sketch\ultralcd.cpp:3638:9: note: in expansion of macro 'MENU_MULTIPLIER_ITEM_EDIT_CALLBACK'

     MENU_MULTIPLIER_ITEM_EDIT_CALLBACK(float43, MSG_FILAMENT_DIAM, &planner.filament_size[0], 1.5, 3.25, planner.calculate_volumetric_multipliers);

     ^
exit status 1
'class Planner' has no member named 'filament_size'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.```
Paste_the_error_text_here
```

closed 2018-02-11 by thinkyhead

Semogj, 2018-02-19:
Hello!

I am having exactly the same issue. Do you have a solution or a commit that solves this problem?

Thanks!

thinkyhead, 2018-02-20:
I closed it because it was patched. If you enable `FILAMENT_WIDTH_SENSOR` with `NO_VOLUMETRICS` the sanity-check catches it.
```
Marlin/SanityCheck.h:950:6: error: #error "FILAMENT_WIDTH_SENSOR requires NO_VOLUMETRICS to be disabled."
#error "FILAMENT_WIDTH_SENSOR requires NO_VOLUMETRICS to be disabled."
```
And if that doesn't do it then there's this:
```cpp
  #if DISABLED(NO_VOLUMETRICS) || ENABLED(ADVANCED_PAUSE_FEATURE)
    /**
     *
     * "Control" > "Filament" submenu
     *
     */
    void lcd_control_filament_menu() {
      . . .
    }
  #endif // !NO_VOLUMETRICS || ADVANCED_PAUSE_FEATURE
```

thinkyhead, 2018-02-20:
I closed it because it was patched. If you enable `FILAMENT_WIDTH_SENSOR` with `NO_VOLUMETRICS` the sanity-check catches it.
```
Marlin/SanityCheck.h:950:6: error: #error "FILAMENT_WIDTH_SENSOR requires NO_VOLUMETRICS to be disabled."
#error "FILAMENT_WIDTH_SENSOR requires NO_VOLUMETRICS to be disabled."
```
And if that doesn't do it then there's this:
```cpp
  #if DISABLED(NO_VOLUMETRICS) || ENABLED(ADVANCED_PAUSE_FEATURE)
    /**
     *
     * "Control" > "Filament" submenu
     *
     */
    void lcd_control_filament_menu() {
      . . .
      #if DISABLED(NO_VOLUMETRICS)
        . . .
      #endif
      . . .
      #if ENABLED(ADVANCED_PAUSE_FEATURE)
        . . .
      #endif
    }
  #endif // !NO_VOLUMETRICS || ADVANCED_PAUSE_FEATURE
```

mrah, 2018-05-03:
Today i download 1.1.8 from marlinfw.org but i had the same error. FILAMENT_WITH_SENSOR is disabled. (No issue on Marlin 2.0)

**My config files.**
[Marlin-1.1.x.zip](https://github.com/MarlinFirmware/Marlin/files/1971711/Marlin-1.1.x.zip)

`/var/folders/q2/p9jfwcmn2n3crj7yhfvfc3080000gn/T/arduino_build_77120/sketch/ultralcd.cpp: In function 'void lcd_control_filament_menu()':
ultralcd.cpp:3638: error: 'class Planner' has no member named 'filament_size'
         MENU_MULTIPLIER_ITEM_EDIT_CALLBACK(float43, MSG_FILAMENT_DIAM, &planner.filament_size[0], 1.5, 3.25, planner.calculate_volumetric_multipliers);
                                                                                 ^
/var/folders/q2/p9jfwcmn2n3crj7yhfvfc3080000gn/T/arduino_build_77120/sketch/ultralcd.cpp:290:30: note: in definition of macro '_MENU_ITEM_PART_2'
         menu_action_ ## TYPE(__VA_ARGS__); \
                              ^
/var/folders/q2/p9jfwcmn2n3crj7yhfvfc3080000gn/T/arduino_build_77120/sketch/ultralcd.cpp:345:66: note: in expansion of macro 'MENU_MULTIPLIER_ITEM'
     #define MENU_MULTIPLIER_ITEM_EDIT_CALLBACK(type, label, ...) MENU_MULTIPLIER_ITEM(setting_edit_callback_ ## type, label, PSTR(label), ## __VA_ARGS__)
                                                                  ^
/var/folders/q2/p9jfwcmn2n3crj7yhfvfc3080000gn/T/arduino_build_77120/sketch/ultralcd.cpp:3638:9: note: in expansion of macro 'MENU_MULTIPLIER_ITEM_EDIT_CALLBACK'
         MENU_MULTIPLIER_ITEM_EDIT_CALLBACK(float43, MSG_FILAMENT_DIAM, &planner.filament_size[0], 1.5, 3.25, planner.calculate_volumetric_multipliers);
         ^
ultralcd.cpp:3638: error: 'class Planner' has no member named 'filament_size'
         MENU_MULTIPLIER_ITEM_EDIT_CALLBACK(float43, MSG_FILAMENT_DIAM, &planner.filament_size[0], 1.5, 3.25, planner.calculate_volumetric_multipliers);
                                                                                 ^
/var/folders/q2/p9jfwcmn2n3crj7yhfvfc3080000gn/T/arduino_build_77120/sketch/ultralcd.cpp:294:102: note: in definition of macro '_MENU_ITEM_PART_2'
         lcd_implementation_drawmenu_ ## TYPE(encoderLine == _thisItemNr, _lcdLineNr, PSTR(LABEL), ## __VA_ARGS__); \
                                                                                                      ^
/var/folders/q2/p9jfwcmn2n3crj7yhfvfc3080000gn/T/arduino_build_77120/sketch/ultralcd.cpp:345:66: note: in expansion of macro 'MENU_MULTIPLIER_ITEM'
     #define MENU_MULTIPLIER_ITEM_EDIT_CALLBACK(type, label, ...) MENU_MULTIPLIER_ITEM(setting_edit_callback_ ## type, label, PSTR(label), ## __VA_ARGS__)
                                                                  ^
/var/folders/q2/p9jfwcmn2n3crj7yhfvfc3080000gn/T/arduino_build_77120/sketch/ultralcd.cpp:3638:9: note: in expansion of macro 'MENU_MULTIPLIER_ITEM_EDIT_CALLBACK'
         MENU_MULTIPLIER_ITEM_EDIT_CALLBACK(float43, MSG_FILAMENT_DIAM, &planner.filament_size[0], 1.5, 3.25, planner.calculate_volumetric_multipliers);
         ^
Using library U8glib at version 1.18 in folder: /Users/Emrah/Documents/Arduino/libraries/U8glib 
exit status 1
'class Planner' has no member named 'filament_size'`

Bob-the-Kuhn, 2018-05-03:
That's definietly a bug in 1.1.8.

Please use this file:
[ultralcd.cpp..zip](https://github.com/MarlinFirmware/Marlin/files/1972811/ultralcd.cpp.zip)

bogaskilop, 2018-07-18:
rc bufgix downloaded just now has same error 

exit status 1
'class Planner' has no member named 'filament_size'

when no volumetric selected  :(