MarlinFirmware/Marlin #9085: [BR][2.0.x] Cannot enable levelling with MBL after manual restore of mesh data

fiveangle, 2018-01-07:
After flash of latest FW as of 082d161a75288252507d5dae4c4f5dff39a94e36 manually edit all mesh points, M500, M420 Z5.0 then when issue M503, get:

```
Send: M503
Recv: echo:  G21    ; Units in mm
Recv: echo:  M149 C ; Units in Celsius
Recv: 
Recv: echo:Steps per unit:
Recv: echo:  M92 X20.00 Y20.00 Z507.00 E96.00
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X250.00 Y250.00 Z8.00 E80.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X1500 Y1500 Z200 E10000
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P1000.00 R3000.00 T1200.00
Recv: echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 S0.00 T0.00 B20000 X8.00 Y8.00 Z0.30 E5.00
Recv: echo:Mesh Bed Leveling:
Recv: echo:  M420 S0 Z5.01
Recv: echo:  G29 S3 X1 Y1 Z-0.02600
Recv: echo:  G29 S3 X2 Y1 Z-0.04800
Recv: echo:  G29 S3 X3 Y1 Z-0.08400
Recv: echo:  G29 S3 X4 Y1 Z-0.11300
Recv: echo:  G29 S3 X5 Y1 Z-0.11600
Recv: echo:  G29 S3 X6 Y1 Z-0.13000
Recv: echo:  G29 S3 X7 Y1 Z0.00000
Recv: echo:  G29 S3 X1 Y2 Z-0.13200
Recv: echo:  G29 S3 X2 Y2 Z-0.13900
Recv: echo:  G29 S3 X3 Y2 Z-0.17300
Recv: echo:  G29 S3 X4 Y2 Z-0.22000
Recv: echo:  G29 S3 X5 Y2 Z-0.23300
Recv: echo:  G29 S3 X6 Y2 Z-0.22600
Recv: echo:  G29 S3 X7 Y2 Z-0.22100
Recv: echo:  G29 S3 X1 Y3 Z-0.16300
Recv: echo:  G29 S3 X2 Y3 Z-0.19600
Recv: echo:  G29 S3 X3 Y3 Z-0.22800
Recv: echo:  G29 S3 X4 Y3 Z-0.27900
Recv: echo:  G29 S3 X5 Y3 Z-0.28200
Recv: echo:  G29 S3 X6 Y3 Z-0.26500
Recv: echo:  G29 S3 X7 Y3 Z-0.26900
Recv: echo:  G29 S3 X1 Y4 Z-0.17900
Recv: echo:  G29 S3 X2 Y4 Z-0.19500
Recv: echo:  G29 S3 X3 Y4 Z-0.20800
Recv: echo:  G29 S3 X4 Y4 Z-0.24200
Recv: echo:  G29 S3 X5 Y4 Z-0.27000
Recv: echo:  G29 S3 X6 Y4 Z-0.27000
Recv: echo:  G29 S3 X7 Y4 Z-0.28100
Recv: echo:  G29 S3 X1 Y5 Z-0.16200
Recv: echo:  G29 S3 X2 Y5 Z-0.14600
Recv: echo:  G29 S3 X3 Y5 Z-0.17300
Recv: echo:  G29 S3 X4 Y5 Z-0.20600
Recv: echo:  G29 S3 X5 Y5 Z-0.22500
Recv: echo:  G29 S3 X6 Y5 Z-0.24000
Recv: echo:  G29 S3 X7 Y5 Z-0.26700
Recv: echo:  G29 S3 X1 Y6 Z-0.12600
Recv: echo:  G29 S3 X2 Y6 Z-0.10800
Recv: echo:  G29 S3 X3 Y6 Z-0.12900
Recv: echo:  G29 S3 X4 Y6 Z-0.16200
Recv: echo:  G29 S3 X5 Y6 Z-0.18700
Recv: echo:  G29 S3 X6 Y6 Z-0.21100
Recv: echo:  G29 S3 X7 Y6 Z-0.24900
Recv: echo:  G29 S3 X1 Y7 Z-0.14000
Recv: echo:  G29 S3 X2 Y7 Z-0.13700
Recv: echo:  G29 S3 X3 Y7 Z-0.16000
Recv: echo:  G29 S3 X4 Y7 Z-0.19800
Recv: echo:  G29 S3 X5 Y7 Z-0.22000
Recv: echo:  G29 S3 X6 Y7 Z-0.22400
Recv: echo:  G29 S3 X7 Y7 Z-0.22900
Recv: echo:Material heatup parameters:
Recv: echo:  M145 S0 H180 B70 F0
Recv: echo:  M145 S1 H240 B110 F0
Recv: echo:PID settings:
Recv: echo:  M301 P30.00 I5.00 D50.00
Recv: echo:  M304 P270.00 I30.00 D550.00
Recv: echo:Linear Advance:
Recv: echo:  M900 K0.00 R0.00
Recv: echo:Filament load/unload lengths:
Recv: echo:  M603 L0.00 U100.00
Recv: ok P15 B3
[...]
```

But levelling cannot be enabled:

```
Send: M420 S1
Recv: Error:Failed to enable Bed Leveling
Recv: echo:Bed Leveling Off
Recv: echo:Fade Height 5.00
Recv: ok P15 B3
[...]

```
as printer reports it has no levelling data:

```
Send: G29 S0
Recv: Mesh bed leveling has no data.
Recv: X:0.00 Y:0.00 Z:0.00 E:-4.30 Count X:0 Y:0 Z:0
Recv: ok P15 B3

```

If I'm doing something wrong, please fill me in.  I was absolutely sure this used to work but it's been a while now so perhaps my memory is failing :)

Thanks,

-=dave

[Configuration.h](https://github.com/fiveangle/Marlin/blob/bf2-xmas_regression_edition/Marlin/Configuration.h), [Configuration_adv.h](https://github.com/fiveangle/Marlin/blob/bf2-xmas_regression_edition/Marlin/Configuration_adv.h)

thinkyhead, 2018-01-07:
Mesh Bed Leveling has a flag, `has_mesh`, which is only set when you do a `G29`. If you've never done a `G29` then this flag will never have been set. In my view the flag is superfluous, but for now you will need to do a `G29`, then you can modify your mesh points, and then you can do `M500`. After that it will work.

closed 2018-01-07 by thinkyhead

fiveangle, 2018-01-07:
So am I the only person who feels that running "G29 S2" 49 times for no reason is a reasonable way that this should work ?   G29 of course by itself will not allow it (tried already).

thinkyhead, 2018-01-07:
> So am I the only person who feels that running "G29 S2" 49 times for no reason is a reasonable way that this should work?

Leveling without an LCD?

fiveangle, 2018-01-07:
I can't fit LCD_LEVELLING - lol.  I can of course do it manually… but it just seems nutty.

Doesn't it make more sense to set `has_mess` the moment anyone issues `G29 S3 XX` ?

thinkyhead, 2018-01-07:
The `has_mesh` value is just an indicator that a `G29` has been completed. I consider it obsolete, since we can just look at the mesh and if it's all zero then consider that as not having a mesh.

fiveangle, 2018-01-07:
I see what you mean that smarter logic one could infer the mess from the data, but a simpler (and possibly smaller footprint) solution is to just enable it on manual point edit.

#9086 for your consideration.

bmxorcist, 2019-12-22:
I had this same issue. Could not enable any type of ABL, and printer would report "Z probe off bed". I started with a fresh download of Marlin 2.0 and it started working again. In other threads it seems this is caused by some incompatibility of manual home position or probe inset settings.