MarlinFirmware/Marlin #9089: BLTouch Smart always triggered with MK Gen 1.4

mistsoul, 2018-01-07:
Hello,

I have a genuine bltouch smart that I am trying to set up. I bought it from one of the ressellers listed on the ANTCLabs website and the unit has the inscriptions on the back and the blue led. 

The unit starts in an error state so it is always show as triggered using M119.  I read several similar issues on this forum and online but I have not figured out the issue yet. The only thread #5705 i could find with the same issue dealing with an MKS Gen 1.4 board like me was several months ago and the user had a clone. Needless to say i did also try to operate it as a clone with the necessary firmware modifications but it did not work as it triggered open all the time. 

-  I can deploy it and stow it, and the leds are working properly. 
- So all the functions S10, S60, S90, S160 are working
- M119 returns triggered whether deployed or stowed 
- setting zmin endstop inverting false does not change this
- the reset function works every time as expected
- I am also able to run the test without an issue
- i double checked all the cables and made sure they were plugged to the correct pins
- the zmin pin is set correctly to 18 and it worked before on my previous firmware with the mechanical 

My configuration attached is for the genuine bltouch. 

does anybody know what I need to do to get it to work? Thank you in advance.

PS: The only error i got when compiling was on old configuration, so i commented it as I did not know where the conflict was.

[Marlin-1.1.x - MT2.0.zip](https://github.com/MarlinFirmware/Marlin/files/1609579/Marlin-1.1.x.-.MT2.0.zip)

Roxy-3D, 2018-01-09:
After homing...    Can you use PronterFace to lower the nozzle (very carefully) until the nozzle just touches the bed.    What does the printer think the Z-Height is?    (Look at the LCD and/or issue a M114 command)

mistsoul, 2018-01-09:
thanks for your help @Roxy-3D 

with a paper sliding under the nozzle it is about -31.5 to lower it to the paper. 

To be able to do this i had to temporarily undefine MIN_SOFTWARE_ENDSTOPS or set my zmin to a high negative in the firmware.

With G28 the probing works and it goes to the middle of the bed, then m114 returns x,y,z at 0,0,0 eventhough z is well above the bed. This is the case even if i set a -z offset. Then G1 Z0 does not do anything since it thinks it is already at 0.

G29 does not bring the probe back to x,y 0,0, it skips to the right then tries to probe but fails. If i trick it by placing something under the probe before it fails, it continues moving to the x axis but rams the xmax extrusion since it thought the middle was point 0. In any case G29 either fails to reach the bed or reaches it but does not retract, both case sends it into an error.

Trying to home z individually sends it to the back right corner which would theoretically correspond to the middle if was trully starting at x,y, 0,0. Then the probing there goes into an error too.

Roxy-3D, 2018-01-09:
@thinkyhead  is the Bi-Linear expert....    Let's wait and see what he says...

mistsoul, 2018-01-09:
Thanks, i read somewhere that having the max endstops activated may cause issues. I had them commented as i thought they were being used by the filament sensors that I was going to setup next. However it looks like the sensors will need their pins defined separately. I will undefine these max endstops later and see if that helps.

mistsoul, 2018-01-11:
Apparently the firmware I was using had additional files from the printer manufacturer which had some modifications made by them on how G28 behaves, maybe to accommodate their laser toolhead. I am not sure if this is the cause of my issue, but I have now migrated to the latest Marlin version 010107 and updated/compiled it without the extra files. I will give it a try later and see if i still have the issues above.

mistsoul, 2018-01-13:
Problem fixed.

closed 2018-01-13 by mistsoul

tramage, 2020-06-12:
I'm having the same issue.  I have spent numerous hours following several tutorials on YouTube for Lowering the nozzle to the correct distance from the bed that should be Z0 including the instructions listed in this thread.  I used the long list of commands on Pronterface including the M211 S0, slowly lowering my nozzle then trying to set that for my new Z0 but when I home again it goes right back to being nearly 3 or more mm higher.  
The one video I watched the most that seemed to help at first was:
https://www.youtube.com/watch?v=y_1Kg45APko&t=6s
But I still cannot get the nozzle at the correct height.
Any suggestions?

RFBomb, 2020-06-13:
@tramage  
The order should be:
M211 S0
M851 Z0
G28
G0 X Y [Middle of bed]    (so current position - probe offset for X and Y respecitvely)
Manually lower Z until nozzle touches bed. 
M851 [Current Z value]  (SO if you are at -2.1, then that is your new Z offset)

Repeat the process. Your nozzle should be at z0 touching the bed.

mistsoul, 2020-06-13:
@tramage

I am not sure if your issue has to do with how you edited your firmware, if so the Marlin team may need to see a log of your changes.

Otherwise the video you posted is very good, and I derived the following steps from it not long ago, and this is how I level my bed (some of the steps are not absolutely needed but it works for me):

Home
M851 Z0
M500 [save]
M501 [active]
M503 [read]
G28 Z
G1 F50 Z0
M211 S0 

Slide paper underneath and lower nozzle as indicated in the video
When you reach the right Z height, don't forget to account for the paper thickness (0.1)
So if you obtain an offset of -1.3 your final Z offset is -1.3-0.1=-1.4 which is # per Z# below

M851 Z#
M211 S1 
M500 save
M501 active
M503 read