MarlinFirmware/Marlin #9090: ABL / UBL for K-Factor Pattern Generator

Sineos, 2018-01-07:
I want to improve the ABL / UBL part in the K-Factor Pattern Generator. As I do not use ABL myself, I want to ask for some input, which g-code combinations are best used to have a maximum compatibility for the users.
If needed I can do some select to choose different BL flavors.

@Roxy-3D some input would be appreciated.

Roxy-3D, 2018-01-07:
I don't understand what exactly you are trying to accomplish.   But one idea jumps out...   It may be possible to modify the G26 Mesh Validation Pattern generation to facilitate tuning of the K-Factor.

I need a couple of paragraphs of text to help me understand what you are thinking....

Sineos, 2018-01-07:
The K-Factor test pattern generator http://marlinfw.org/tools/lin_advance/k-factor.html has an option to use ABL.
All it does as of today is to send a `G29`, which does not make to much sense imo. I would like to extend it to be able for the users to chose what kind of leveling they want to use / have configured.
I would need some support to know what are the proper / most universal g-code command to send as I do not use any ABL.

fiveangle, 2018-01-07:
With the multitude of leveling schemes is it wise to try and automate this ? If the user is truly so much a noob that they cannot master enabling BL for thier prints, what would make their answers on the generator JS page any better ?

Seems ripe for being a support mess for those who think they can jump right into LA + BL without any knowledge. But if you think you can do it, go for it !

At a minimum would be to ensure complete print abort if bed leveling cannot be activated and the user enabled it in the JS page.  Now that I think about it, I’m even more concerned,  because gcode doesn’t give you the ability to test for specific responses as far as I know. So how do you ensure the user selected the right boxes ?

Roxy-3D, 2018-01-07:
Can you describe what the generated GCode is trying to do to assist calibration of the K-Factor?
I'm wondering if it should just be built into Marlin like the G26 Mesh Validation Pattern is.   Certainly, the G26 pattern could be done externally and printed via GCode.   But the problem with doing that is getting everything to like up correctly.   There is real value in having the line intersections positioned exactly over the mesh points.   Trying to coordinate that with a Slicer to generate the GCode is problematic.

Sineos, 2018-01-08:
I guess we might be in misunderstanding.
The k-pattern calibration generator just uses G29 to activate ABL. The script assumes that the user has configured any of the available leveling options in the firmware and EEPROM. 
My question is if this one command is enough to properly use (the already configured) bed leveling or if additional commands are needed.

Roxy-3D, 2018-01-08:
> My question is if this one command is enough to properly use (the already configured) bed leveling or if additional commands are needed.

For any of the mesh based systems...   The mesh needs to be setup.   And activated.    That takes a number of commands.    For UBL, to bring up the system this 'Cheat Sheet' of commands will usually do a good job:   https://github.com/MarlinFirmware/MarlinDocumentation/blob/master/_features/unified_bed_leveling.md

fiveangle, 2018-01-08:
For MBL for example, G29 does nothing useful, except perhaps starting the laborious leveling process if `has_mesh `isn't yet set.  `M420 S1` is required to enable leveling on MBL if `has_mesh `is set.

Again, each leveling system is different (why is a mystery because it certainly wasn't required to be like this, but here we are today...) and so you would need to take into consideration every possible method (and the user would also have to choose the leveling scheme they use, and if they make a mistake, it will not get expected results).

As activating bed leveling is such a fundamental operation that a user needs to be able to understand (and typically will already have setup in their slicer already), I'll re-affirm my concerned over trying to automate this.  The benefit seems not great enough, given the level of development/testing required to ensure it works as expected.

Roxy-3D, 2018-01-08:
> As activating bed leveling is such a fundamental operation that a user needs to be able to understand (and typically will already have setup in their slicer already), I'll re-affirm my concerned [...]

Yes.   I share similar feelings.   My belief is if this should be done, it can be done most completely and easiest for the user in the firmware....

fiveangle, 2018-01-08:
> My belief is if this should be done, it can be done most completely and easiest for the user in the firmware....

I agree that pushing this into an LCD menu entry to automatically be done (like UBL G26) would be ideal.   But for now, getting the feature nailed down in the much-more-managable JS page seems wise.  Once it's vetted and working well and people are used to it, pushing into FW might make sense.  Babysteps... :)

Roxy-3D, 2018-01-09:
I'm still waiting to hear exactly what is important to be able to calibrate this number.   How do different values affect the appearance of the line that printed?   What is the user looking for in order to know how much and which direction to shift the magical constant.

Using G26 as an example...    The user is looking to see if the filament has a small amount of 'squish' to it and it is stuck to the bed everywhere.    What is the user looking for in this case????

fiveangle, 2018-01-09:
> I'm still waiting to hear exactly...

Wish in one hand... 😉 

I suggest you try the LIN_ADV calibration procedure yourself manually before you poo-poo its potential meri, as well as the requirement to get the K factor close for the filament in use in order for it to add any value.  Now that I remember, it was the near-miracle improvement in print quality at-speed of my Printrbot Metal after implementing Sebastian's test fork that spurred me to kick down some beer money to the Marlin project in the first place -- it made that much of an improvement.  What a difference a year makes ! 😄

Roxy-3D, 2018-01-10:
>> I'm still waiting to hear exactly...

> Wish in one hand... :wink:

> I suggest you try the LIN_ADV calibration procedure yourself manually before you poo-poo its potential merit, as well as the requirement to get the K factor close for the filament in use in order for it to add any value.

I'n not poo-poo-ing anything.   I was hoping to get somebody that knows about the topic to help me understand things quickly.   And maybe it will be possible to re-purpose (or double purpose) some of the G26 code to help people easily calibrate the Linear Advance settings.

fiveangle, 2018-01-10:
https://mattshub.com/2017/10/02/linear-advance/

Definitely a much simpler pattern than G26

closed 2018-02-11 by thinkyhead