MarlinFirmware/Marlin #9104: No Documentation for Compiling for MKS SBASE

Zakm0n, 2018-01-09:
So, I'm decent enough with arduino ide to compile and flash standard arduino boards, but I've been searching for hours on how to compile a working firmware.bin for my mks sbase and I've come up empty handed. Is it possible to maybe include some instructions on how to do so?

Spawn32, 2018-01-09:
Install platformio, i just use the simple cmd line version of it.
After that , just type:  platformio run -e LPC1768

Use the config files from the example folder...

thinkyhead, 2018-01-09:
Also available as an add-on for the free Atom and Visual Studio Code project editors. http://platformio.org/platformio-ide

closed 2018-01-09 by thinkyhead

MS1987, 2018-02-11:
@Spawn32 After compile, I got the "firmware.bin" in the path "Marlin-bugfix-2.0.x\.pioenvs\LPC1768". 
How can I update the sbase? Just copy the "firmware.bin" to the tf card and reboot? How about the primary "config.txt"?

AnHardt, 2018-02-11:
In Marin we don't have a "config.txt" file like in Smothy.
The configuration is built in into the "firmware.bin".
If your board/processor has no EEPROM a 'M500' writes to an emulate EEPROM, either emulated in the program memory (DUE) or a file "eeprom.dat" (LPC). However this file is a EEPROM image - so not intended to be human readable/editable. You can get about the same functionality with a autostart file "auto0.g" where you could paste in the output of a 'M503' and edit a bit.

Spawn32, 2018-02-11:
@MS1987 
Yes, just copy the file to you sd and reboot :)

MS1987, 2018-02-12:
@Spawn32 After I copy the firmware.bin to the tf card and reboot, the usb port cannot comminucate with pc. As it just display the "Marlin usb device", however, when I use smoothie firmware, the driver is ok.
But it can recognize it as Udisk to read the files inside sdcard.
And I have confirmed I have config "#define SERIAL_PORT -1".

p3p, 2018-02-12:
On Windows versions < 10 (the only OS to require this for what has been a standard peripheral device for 20 years) you need a driver, We can't just use someone else's VID and PID numbers in the Marlin LPC1768 usb stack so the Smoothieware driver will not automatically work (although I'm relatively sure you can manually install it for the Marlin device) we supply one [here](https://raw.githubusercontent.com/MarlinFirmware/Marlin/bugfix-2.0.x/frameworks/CMSIS/LPC1768/lib/usb/Re-ARM_usb_driver.inf) (it is unsigned though).

MS1987, 2018-02-12:
@p3p You are right! It works after I install it manually.Thanks a lot.

projectoblio, 2019-09-30:
Is there any way to flash this firmware from USB? What would the settings be in Arduino IDE?