MarlinFirmware/Marlin #9105: MKS SBASE EEPROM not working. "Error:EEPROM datasize error"

Zakm0n, 2018-01-09:
Ok, got my SBASE flashed, moving and shaking, but the M500 command returns Error:EEPROM datasize error. I have "#define EEPROM_SETTINGS" uncommented in my config. What gives?

ikarisan, 2018-01-09:
Same here!

```
Connecting...
start
Printer is now online.
echo:Marlin bugfix-1.1.x
echo: Last Updated: 2017-10-24 12:00 | Author: Shinji Ikari
echo:Compiled: Jan  8 2018
echo: Free Memory: 3600  PlannerBufferBytes: 1344
Error:EEPROM datasize error.
[ERROR] Error:EEPROM datasize error.
```

And all settings stored with "M500" are lost on a restart!

ikarisan, 2018-01-09:
After cleaning the EEPROM with the Arduino example sketch I get this:

```
Connecting...
start
Printer is now online.
echo:Marlin bugfix-1.1.x
echo: Last Updated: 2017-10-24 12:00 | Author: Shinji Ikari
echo:Compiled: Jan  9 2018
echo: Free Memory: 3600  PlannerBufferBytes: 1344
echo:EEPROM version mismatch (EEPROM= Marlin=V49)
echo:Hardcoded Default Settings Loaded
```

Zakm0n, 2018-01-09:
@ikarisan I am on the 32bit mks sbase smoothie based board. If you are running 1.1 you're on an arduino based mks base, right?

ikarisan, 2018-01-09:
Yes, I am on an "MKS Gen V1.4".

ikarisan, 2018-01-09:
@Zakm0n : Do you use TMC2XXX drivers?

thinkyhead, 2018-01-09:
Does `M500`, `M502` clear up the issue?

thinkyhead, 2018-01-09:
Please post your `Configuration.h` and `Configuration_adv.h`. You can zip them up and drop them on your next reply.

thinkyhead, 2018-01-10:
I've added extra debugging to the EEPROM code in #9122 ([branch here](https://github.com/thinkyhead/Marlin/tree/bf1_eeprom_debug)). Download that code, add this line to your `Configuration.h`…
```cpp
#define DEBUG_EEPROM_READWRITE
```
…and then we'll see what the EEPROM commands report. It should give us a better idea where the error is originating.

thinkyhead, 2018-01-10:
I've merged those PRs into `bugfix-1.1.x` and `bugfix-2.0.x` so you can just download the latest.

thinkyhead, 2018-01-10:
I've merged new EEPROM debugging code into `bugfix-1.1.x` and `bugfix-2.0.x` so please download the latest code, add this line to your `Configuration.h`…
```cpp
#define DEBUG_EEPROM_READWRITE
```
…and then we'll see what the EEPROM commands report. It should give us a better idea where the error is originating.

Zakm0n, 2018-01-10:
[CONFIGS.zip](https://github.com/MarlinFirmware/Marlin/files/1617294/CONFIGS.zip)

This contains my configs as well as my lightly modified pins file. I'll try the new code and report back.

Zakm0n, 2018-01-10:
I don't know what you did, but the new code fixed the errors. M500 and M501 is working fine now

Zakm0n, 2018-01-10:
Aaand now it's back. "SENT: M500
READ: Error:Field home_offset mismatch."

Zakm0n, 2018-01-11:
I tried re-flashing and resetting and I'm still getting errors

bogaskilop, 2018-01-11:
Got the same problem  when flashed today  bugfix-1.1.x mks board sbase 1.4 from tornado 

[Marlin_z_configuratora.zip](https://github.com/MarlinFirmware/Marlin/files/1623574/Marlin_z_configuratora.zip)

READ: echo: Free Memory: 2671  PlannerBufferBytes: 1312
Free Memory: 2671  PlannerBufferBytes: 1312
READ: Error:EEPROM datasize error.
READ: echo:Index: 594 Size: 665
Index: 594 Size: 665

bogaskilop, 2018-01-12:
**thinkyhead** after adding  this option to configuration.h:

Attempting connection at \\.\COM3...
READ: start
  Connected to machine!
READ: echo:Marlin bugfix-1.1.x
Marlin bugfix-1.1.x
READ: 
READ: echo: Last Updated: 2017-10-24 12:00 | Author: (kilop, Tornado BL+Feeder)
Last Updated: 2017-10-24 12:00 | Author: (kilop, Tornado BL+Feeder)
READ: echo:Compiled: Jan 12 2018
Compiled: Jan 12 2018
READ: echo: Free Memory: 2671  PlannerBufferBytes: 1312
Free Memory: 2671  PlannerBufferBytes: 1312
READ: Error:EEPROM datasize error.
READ: echo:Index: 594 Size: 665
Index: 594 Size: 665
READ: echo:Hardcoded Default Settings Loaded
Hardcoded Default Settings Loaded
SENT: T0
SENT: M105
SENT: M105
SENT: M105
SENT: M105

What kind of info you will need to find error ?

ikarisan, 2018-01-12:
@thinkyhead Yes, but I thought that there might be a common source of error. ;)
For the moment I am back to 1.1.6, because I had too many problems since upgrading to 1.1.>=7 :(

bogaskilop, 2018-01-13:
yea me to when i moved to 1.1.7  or  1.1.8 i got problems even with homing  some speeds changes  for so slow or  in lcd missing  e1 and e2  unload   in 1.1.8 there is only unload filament  but there is no option to change to second extruder in rc-bugfix it was. 1.1.6 and 1.1.7  can not find this ehh 
my setup is  one hotend and tl-feeder ( Y option )

bogaskilop, 2018-01-17:
Is there any update something was found what causing that error ?

ikarisan, 2018-01-18:
@bogaskilop 
After using 1.1.6 for a week I flashed the newest bugfix-1.1.x (2017-01-16) version. At the first start Marlin complained about the version of the configuration found in the EEPROM. After calling M502 followed by M500, both executed without any errors, the bug (my version) seems to be fixed.

teddybee, 2018-01-18:
I have the same issue with the latest 2.0.0. When I send M500, MKS Sbase says:

Send: M500
Recv: echo:Settings Stored (631 bytes; crc 58192)
Recv: Error:EEPROM datasize error.
Changing monitoring state from 'Operational' to 'Error: EEPROM datasize error.
'
Recv: ok

p3p, 2018-01-18:
datasize errors should have been fixed by #9161 (merged about 4 days ago), how new is latest?

teddybee, 2018-01-18:
@p3p: I pulled down and compiled 7 hours ago.

p3p, 2018-01-18:
can you delete eeprom.dat off the sdcard and see if that helps and make sure the new version flashed properly

teddybee, 2018-01-18:
@p3p I have tried it again: Compile -> delete the two files from the root of sd card -> copy fw -> M502 -> M500     and  Error:EEPROM datasize error
I am sure I used the latest version: 'Correct mesh size calculations' was the last commit in the log.

p3p, 2018-01-18:
can you supply your config

bogaskilop, 2018-01-18:
ok no error so tomorrow testing :)

teddybee, 2018-01-19:
@p3p This is my current config.
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1646480/configuration.zip)

tcm0116, 2018-01-19:
Can you see if #9240 fixes this issue for you?

teddybee, 2018-01-19:
@tcm0116 I still have the datasize issue.

tcm0116, 2018-01-19:
@teddybee I just pushed an updated to #9240, which fixes your issue. It originally fixed the issue for writing the EEPROM, but not for reading, so it should be good to go now.

teddybee, 2018-01-19:
@tcm0116 M502, M500 and M501 is fixed, no error

bogaskilop, 2018-01-21:
so  flasheed tis and super  proper menu for second extruder load and unload  wich is missine in all preverious version of marlin  but now when it start  prints   it missing first few lines at the beggining of print LoL   lets say   we start printign  a test cube so after outlines printed ( also not starting from beggining  when it should )  it goes to make a first layer of bottom and at thje beginign about 5-6 lines are not printed  i tested it few times LOL 
wahy each time on new version some thing what works good is messed up ?

thinkyhead, 2018-01-22:
> why each time on new version something that works good is messed up ?

It can only be attributable to human error. This sort of thing has cropped up before, and it has always been due to human error.

thinkyhead, 2018-01-22:
> now when it start prints it missing first few lines at the beginning of print LoL lets say we start printing a test cube so after outlines printed ( also not starting from beginning when it should ) it goes to make a first layer of bottom and at the beginning about 5-6 lines are not printed i tested it few times

Interesting. Does this affect all files? Can you discern any pattern?

AnHardt, 2018-01-22:
>  lets say we start printing a test cube so after outlines printed ( also not starting from beginning when it should ) it goes to make a first layer of bottom and at the beginning about 5-6 lines are not printed i tested it few times 

Does the nozzle not do the moves or is just no material coming out of the nozzle?
Did you 'prime' the nozzle?

closed 2018-02-14 by thinkyhead