MarlinFirmware/Marlin #9106: ###Bug Report Actual Z probe Offset changes when enabling/disabling UBL

HalfShaft, 2018-01-09:
- Description: --- Setting up a new compile of Marlin 1.1.8 I cannot get UBL to work in the manner in which I would expect it to function. 
- Expected behaviour: Doing a G28 after first compile with z probe offset set to 0 I enter M502, M500, M501 and home to z0. Disabling the software endstops using M211 S0 I slowly lower the z height until the gap feels good on a piece of folded paper. I recompiled with this new offset (M502, M500, M501) and G28 after reboot puts me at the same spot with a g1 z0. I start the g29 p1 process (Tried both using GCode and the LCD Menu with the same results.) G29 P3 to fill in the rest of the mesh to make sure everything works. G29 T shows a full mesh map, so I hit G29 S, G29 A and then M500, M501. Doing a G28 at this point leaves my nozzle a considerable distance from my bed.  If I disable the UBL using G29 D the g28 followed by g1 z0 puts the nozzle back on the bed softly. G29 A I'm floating in the air after homing at G28. I can change my z offset using m851 or the LCD and saving to eeprom and reloading. G28 now puts my nozzle on the bed with a radically different offset than before. Starting a print the code calls out a g28 to set the home position and it starts printing with the nozzle actually burried into the bed far enough to cut a pretty line in the tape the outline of my part. Turn off UBL using g29 D and the nozzle homes to z0 barely touching the bed. If I reboot the printer and check G29 T the mesh shows values in the 2-4mm range, while doing a G29 L1 followed by a G29 T shows values in the <.3mm range. 

Another interesting tidbit, the UBL setup does not handle a Z probe located behind the extruder well, (I know it's not ideal but I can't make a new one till I get it up and running again.) The max limits calculated are taken as a min value between the bed and the bed + y offset. Since the probe is a positive offset to the nozzle this results in the printer trying to probe all the way up to the bed edge with the sensor hanging off of the back when doing a G29 P1. I had to change the bed - y offset in the config_adv to get the item to work properly for me.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1614068/Marlin.zip)

Recv: echo:  G21    ; Units in mm
Recv: echo:  M149 C ; Units in Celsius
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo:  M92 X100.00 Y100.00 Z400.00 E100.00
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X400.00 Y400.00 Z8.00 E50.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X2000 Y2000 Z100 E10000
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P400.00 R1000.00 T1000.00
Recv: echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.30 E5.00
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Unified Bed Leveling:
Recv: echo:  M420 S1 Z10.00
Recv: 
Recv: Unified Bed Leveling System v1.01 active.
Recv: 
Recv: Active Mesh Slot: 1
Recv: EEPROM can hold 8 meshes.
Recv: 
Recv: echo:Material heatup parameters:
Recv: echo:  M145 S0 H180 B70 F0
Recv: echo:  M145 S1 H240 B110 F0
Recv: echo:PID settings:
Recv: echo:  M301 P28.34 I4.07 D49.32
Recv: echo:Z-Probe Offset (mm):
Recv: echo:  M851 Z-0.45

Roxy-3D, 2018-01-09:
You have:
```
#define Z_CLEARANCE_DEPLOY_PROBE   10 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES  5 // Z Clearance between probe points
```
After the G28, the nozzle should be up above the bed...   And the fact you can do a G1 Z0 just touches the nozzle to the bed suggests everything is OK.

HalfShaft, 2018-01-09:
I understand that it should be above the bed after a G28. That much makes sense, but the amount it is above the bed changes whether G29 is set to A or D. I have to have a different offset depending on whether the UBL is enabled. This is in manual, when it goes to print it goes into the bed even further and digs through the painters tape to the aluminum.

Roxy-3D, 2018-01-09:
How much different is the height?   Can you post a G29 T and a G29 W please.

HalfShaft, 2018-01-09:
From printer power on should I do a g29 l1 first as the mesh maps looks different before and after I do that? Do I need to do a g28 or anything else? I know with my older Marlin build I ran a Z offset of -.35 but after the first mishap with the UBL running off of the bed I now use a -.7 offset. (I think the nozzle might have gotten written about as it scraped across the bed.) it will be a couple of hours till I get off of work and can try it.

Roxy-3D, 2018-01-09:
Can you print a small object with no bed leveling built into the firmware?   It is important that you be able to do that prior to turning on bed leveling.

If you can print a small object at the center of the bed with no bed leveling active, have you tried following these steps to bring up the UBL system?   

https://github.com/MarlinFirmware/MarlinDocumentation/blob/master/_features/unified_bed_leveling.md

HalfShaft, 2018-01-09:
I printed the standard tree frog from thingiverse last night after restarting and not enabling G29 and resetting the offset on z again. It didn't print the bottom 2 layers due to sitting too low but the print finished without  issue. I will print something else tonight after completely removing the bed leveling from the configuration. Also, I followed that document verbatim to get to the point I am at.

Roxy-3D, 2018-01-09:
OK!   Thank you for confirming that!    If you can get the home height a little bit closer to 'correct' so the first few layers print right....      I think following those directions will get everything working the way you want it.

HalfShaft, 2018-01-10:
I tried a few things tonight, and am here to report the results. Turning on the printer from a completely powered off state I issued a G29 T command. (Recv is from using my Octopi window as a terminal instead of dragging out my surface pro.)

> G29 T
> 
> Recv: echo:Home XYZ first
> Recv: 
> Recv: Bed Topography Report:
> Recv: 
> Recv: (0,9)                                                                   (9,9)
> Recv: (6,214)                                                                (214,214)
> Recv:  -0.879   -0.471   -0.193   -0.240   -0.087   -0.229   -0.331   -0.413   -0.495   -0.517
> Recv: 
> Recv:  -0.898   -0.520   -0.262   -0.279   -0.161   -0.263   -0.340   -0.422   -0.504   -0.526
> Recv: 
> Recv:  -0.917   -0.569   -0.331   -0.318   -0.235   -0.297   -0.349   -0.431   -0.513   -0.535
> Recv: 
> Recv:  -0.936   -0.618   -0.400   -0.357   -0.309   -0.331   -0.358   -0.440   -0.522   -0.544
> Recv: 
> Recv:  -0.955   -0.667   -0.469   -0.396   -0.383   -0.365   -0.317   -0.364   -0.416   -0.428
> Recv: 
> Recv:  -0.989   -0.696   -0.548   -0.430   -0.377   -0.349   -0.306   -0.343   -0.345   -0.297
> Recv: 
> Recv:  -1.033   -0.725   -0.577   -0.509   -0.471   -0.413   -0.350   -0.342   -0.329   -0.301
> Recv: 
> Recv:  -1.082   -0.799   -0.641   -0.503   -0.510   -0.442   -0.364   -0.346   -0.293   -0.241
> Recv: 
> Recv:  -1.121   -0.878   -0.755   -0.607   -0.544   -0.451   -0.448   -0.350   -0.327   -0.210
> Recv: 
> Recv: [-1.065]  -0.772   -0.714   -0.621   -0.533   -0.440   -0.377   -0.339   -0.206   -0.084
> Recv: (6,40)                                                                    (214,40)
> Recv: (0,0)                                                                     (9,0)
> Recv: ok

followed by a G29 W

>  Send:
> 
>  G29 W
> Recv: echo:Home XYZ first
> Recv: Unified Bed Leveling System v1.01 active.
> Recv: Mesh 1 Loaded.
> Recv: UBL object count: 1
> Recv: planner.z_fade_height : 10.0000
> Recv: # of samples: 100
> Recv: Mean Mesh Height: -0.478979
> Recv: Standard Deviation: 0.224329
> Recv: zprobe_zoffset: -0.4500000
> Recv: MESH_MIN_X  ((((((220) / 2) - (220) / 2) + 6)>(-5 + 0)?((((220) / 2) - (220) / 2) + 6):(-5 + 0)))=6
> Recv: MESH_MIN_Y  ((((((220) / 2) - (220) / 2) + 6)>(-10 + 50)?((((220) / 2) - (220) / 2) + 6):(-10 + 50)))=40
> Recv: MESH_MAX_X  ((((((220) / 2) + (220) / 2) - (6))<(220 + 0)?((((220) / 2) + (220) / 2) - (6)):(220 + 0)))=214
> Recv: MESH_MAX_Y  ((((((220) / 2) + (220) / 2) - (6))<(220 + 50)?((((220) / 2) + (220) / 2) - (6)):(220 + 50)))=214
> Recv: GRID_MAX_POINTS_X  10
> Recv: GRID_MAX_POINTS_Y  10
> Recv: MESH_X_DIST  23.11
> Recv: MESH_Y_DIST  19.33
> Recv: X-Axis Mesh Points at: 6.000  29.111  52.222  75.333  98.444  121.556  144.667  167.778  190.889  214.000
> Recv: Y-Axis Mesh Points at: 40.000  59.333  78.667  98.000  117.333  136.667  156.000  175.333  194.667  214.000
> Recv: 
> Recv: Unified Bed Leveling sanity checks passed.
> Recv: ok

I then disabled UBL by typing G29 D and homing with G28. My -.45 z-offset has the nozzle touching a piece of paper at Z0. (G28 probes at x110 y60 due to sensor offset, moved the carriage to y110 so the same spot is used.) I then typed G29 A and redid the G28 command. My nozzle now touches with what I would call the same force at Z0.4 instead of Z0 with the same process. I punched in a G29 D and redid the G28 and everything again touches at Z0 and I am currently printing a 20x20x20 cube with UBL disabled without a single issue. The first layer is clean and flat and seems to be sticking without an issue. (I have not as of yet reflashed the firmware without the UBL as I thought this might be a valuable step.)

When I tried doing a G29 J to adjust for bed tilt and reran the G29 T and G29 W commands I got a much different result.

> Send: G29 T
> Recv: 
> Recv: Bed Topography Report:
> Recv: 
> Recv: (0,9)                                                                   (9,9)
[... 24 lines cut ...]
> Send: G29 W
> Recv: Unified Bed Leveling System v1.01 active.
> Recv: Mesh 1 Loaded.
> Recv: UBL object count: 1
> Recv: planner.z_fade_height : 10.0000
> Recv: # of samples: 100
> Recv: Mean Mesh Height: -2.481641
> Recv: Standard Deviation: 0.360384
> Recv: zprobe_zoffset: -0.4500000
> Recv: MESH_MIN_X  ((((((220) / 2) - (220) / 2) + 6)>(-5 + 0)?((((220) / 2) - (220) / 2) + 6):(-5 + 0)))=6
> Recv: MESH_MIN_Y  ((((((220) / 2) - (220) / 2) + 6)>(-10 + 50)?((((220) / 2) - (220) / 2) + 6):(-10 + 50)))=40
> Recv: MESH_MAX_X  ((((((220) / 2) + (220) / 2) - (6))<(220 + 0)?((((220) / 2) + (220) / 2) - (6)):(220 + 0)))=214
> Recv: MESH_MAX_Y  ((((((220) / 2) + (220) / 2) - (6))<(220 + 50)?((((220) / 2) + (220) / 2) - (6)):(220 + 50)))=214
> Recv: GRID_MAX_POINTS_X  10
> Recv: GRID_MAX_POINTS_Y  10
> Recv: MESH_X_DIST  23.11
> Recv: MESH_Y_DIST  19.33
> Recv: X-Axis Mesh Points at: 6.000  29.111  52.222  75.333  98.444  121.556  144.667  167.778  190.889  214.000
> Recv: Y-Axis Mesh Points at: 40.000  59.333  78.667  98.000  117.333  136.667  156.000  175.333  194.667  214.000
> Recv: 
> Recv: Unified Bed Leveling sanity checks passed.
> Recv: ok
> 

Hopefully, this might give some insight into what I am seeing. Let me know if there is something else you would like me to try or something you need to ask.

boelle, 2019-02-20:
@HalfShaft Please test with the latest `bugfix-2.0.x`

boelle, 2019-03-06:
@thinkyhead i think we can close this one

boelle, 2019-06-22:
https://github.com/MarlinFirmware/Marlin/issues/9048

closed 2019-07-06 by shitcreek