MarlinFirmware/Marlin #9108: [Question] - Help to tune buffers

MasterPIC, 2018-01-09:
I read about quality issues when controlling the printer by Raspberry by Octoprint.
It seems the issues, if existing, can be solved by tweaking buffers size.

- I think that by enabling **ADVANCED_OK** the printer will output not simply "ok" but "ok" followed by spaces left in the planner and in the command buffer. So I think I should tune **BLOCK_BUFFER_SIZE** accordingly, am I wrong?
- As for serial communication issues should I set/tune **RX_BUFFER_SIZE**?
- Can the LCD be updated less frequently?

P.s. I'm using Marlin 2.0 on a stock Anet Board controlled by Raspberry Pi Zero W (Octoprint). I disabled SD support but I'm using AUTO_BED_LEVELING_BILINEAR.

gavatron3000, 2018-01-21:
switching to a pi 3 might solve your problem. 0.5gb more ram, it has 3 more cores for processing etc etc

thinkyhead, 2018-02-14:
> Can the LCD be updated less frequently?

Currently it updates once per second. Graphical has been well optimized and will automatically throttle itself if the planner looks like it might starve.

closed 2019-02-20 by MasterPIC