MarlinFirmware/Marlin #9117: ### Bug Report

marcubea1, 2018-01-09:
### Bug Report
- Description: The probe that i am using is a ANTCLABS BLTouch. I am having issues with the probe positions, when it goes to probe it tries to probe outside of the bed on both Auto bed levelling bilinear and delta autocalibration. I know this may not be a bug but would just like some help with it.
Thanks
Marcus
- Expected behaviour: Probe inside the bed area
- Actual behaviour: Probe reaches outside the bed area

marcubea1, 2018-01-09:
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/1616715/Config.zip)

thinkyhead, 2018-01-10:
How do you reconcile this?
```cpp
#define X_PROBE_OFFSET_FROM_EXTRUDER +35     // X offset: -left  +right  [of the nozzle]
. . .
  #define DELTA_PROBEABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 25)
```

Seems like it would try to probe 10mm off the right side with those settings.

marcubea1, 2018-01-10:
Why doesn't marlin account for these offsets when probing by default. Surely it could be made so that when it probes it changes the 0,0 to the probe position instead of the nozzle so that the probe is in the centre of the bed rather than the nozzle. Kind of like a tool change on a dual extrusion printer

marcubea1, 2018-01-10:
I'm not being malicious here if thats what it sounds like i love your guys work its amazing what you have achieved

Roxy-3D, 2018-01-10:
Marlin supports a huge amount of different hardware types.  And it supports an almost exponential combination of different hardware types.

At some point...    The firmware has to trust the user put in settings that accurately describe their machine.      There are extensive Sanity Checks.   And perhaps this type of error is worth checking for.   But...   at some point the firmware needs accurate numbers.

marcubea1, 2018-01-10:
Okay thanks for your time and help
Much appriciated

closed 2018-02-11 by thinkyhead