MarlinFirmware/Marlin #9134: [1.1.8] After each print I need to add ~-0.2 babystepping value to print on bed

Fleckz, 2018-01-10:
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1614795/Configuration.txt)
So, I turn on printer, start printing first part, I babystep to around -0.200, stop print, start new print, now I have to babystep to around -0.400, stop, start, next print is -0.600 etc, Now I am at -1.600 :D Ideas?

v1ru3, 2018-01-13:
reinitialize eeprom, worked for me a few times.

Fleckz, 2018-01-13:
Thanks, I will try that!

th33xitus, 2018-01-15:
Anyone got a solution for it? @Fleckz did it work for you?
I encountered the same issue but i don't even have EEPROM enabled.

Fleckz, 2018-01-15:
I have EEPROM enabled, but I didn't have time to test it, I hope to try later today!

Fleckz, 2018-01-15:
Nope, it's not helping here also!
1st print: ~+0.2 babystep
2nd print: -0.1 babystep
3rd print: was way off, -1. and still was far from bed surface...
did not finish, just stoped print, did autohome, bed level again and printing...

tcm0116, 2018-01-18:
Can you also post your Configuration_adv.h?

Fleckz, 2018-01-18:
sure, there you go!
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/1643129/Configuration_adv.txt)

jovo0911, 2018-01-21:
same issue here, Board also "MKS_BASE"!

thinkyhead, 2018-02-14:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see if we fixed this issue. If the problem has been solved then we can close it. If you still see the bad behavior we should investigate further.

jovo0911, 2018-02-14:
thx, for answer. will do, takes few days ….

jovo0911, 2018-02-15:
Hello Scott, 

I made a test with the latest bugfix 1.1.x  from 14 feb. 2018. Printed a few things (means up to 10 small items) without power off or reset.
„NO“ change in z-offset, stable - could print without correct the z-offset. Look good so far. Will check the next day again  

Great Job thank you to the team.

regards
Joachim Vorrath

Fleckz, 2018-02-15:
hmm, I don't see any changes, just configured/uploaded bugfix-1.1.x, Z is all over the place!
First print, Z offset was set to -4, nozzle was too low, had to babystep to +2.3, so I changed Z offset to -1.7
started print - had to babystep to -0.45, so I changed Z offset to -2.2, started printing - had to babystep top -0.848... so, it's broken here!

jovo0911, 2018-02-15:
Hello guys,
as I said yesterday - today still some tests.
I can not confirm what "Fleckz" writes, after 4 runs z-offset is still stable. At start -1.26 slight correction to -1.25. no changes were necessary for the following runs. After the 4th pass still -1.25.
Yesterday, as suggested by 1.1.8 stable, I switched to the bugfix 1.1.x. Configs merged - and that's the result.
It looks very good to me.

Reliability of the sensor:

Z-Probe Repeatability Test

Send: M48 P10 X110 Y110 V4
Recv: M48 Z-Probe Repeatability Test
Recv: Positioning the sample ...
........
Recv: Finished!
Recv: Mean: 0.048900 Min: 0.048 Max: 0.051 Range: 0.004
Recv: Standard Deviation: 0.001300

greetings
Joachim Vorrath

thinkyhead, 2018-02-23:
@Fleckz

Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-1.1.x` or `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

You can use `M114 D` as you move around the bed to see lots of details about the current position. If you see strange things, also copy them into your text file.

Since you're using `BILINEAR` leveling, you can also use `G29 Q` at any time to see details about leveling at the current position.

We look forward to your next report. Hopefully we can narrow this down to a single cause and fix it.

boelle, 2019-02-20:
@Fleckz still having issues when you use latest bugfix 2.0?

Fleckz, 2019-02-20:
Hey, thanks for the reply, I have not used my printer for some time, someday will try to upgrade and let you know! Thanks again!

boelle, 2019-02-20:
then maybe close this one and open new issue if you have problems?

Fleckz, 2019-02-20:
I kinda lean towards my printer problem, not Marlin, meh, don't know, maybe I should just close the issue and see later...

boelle, 2019-02-20:
yep, you can always open new issue if you have problems still

boelle, 2019-02-20:
it may be 6 months or more before you are back and a lot of things can happen in that time